MarlinFirmware/Marlin #13793: [1.1.x BUG] Jerky behavior at start/end of a movement

timkostka, 2019-04-23:
### Description

Jerky behavior when transitioning from stopped to moving on any extruder.

### Steps to Reproduce

1. Install latest [v1.1.9] or [v1.1.x bugfix] Marlin on Arduino Mega2560 board, without any modifications.
2. Set jerk limit to 0mm/s.
3. Look at STEP/DIR signal outputs while executing g-code to move from 0mm to 1mm and back on an axis.
4. Postprocess results to examine position vs time and velocity vs time curves.

**Expected behavior:**

Velocity vs time should resemble a trapezoid, with a period of constant acceleration, then a period of constant velocity, then a period of constant deceleration.

**Actual behavior:**

Velocity vs time has a large jump at the beginning and end of a movement.  It outputs 1 pulse on STEP, then pauses for a while, then jumps up to 9.8mm/s, then proceeds smoothly along the expected trapezoid curve.  When it hits the end of a movement the same thing happens as it slows down.

#### Additional Information

The attached image shows results from the following g-code:
```
G21 ; set units to mm
G90 ; set mode to absolute
M92 X80 Y80 Z400 E93 ; set steps/mm
M203 X250 Y250 Z50 E250 ; set max feed rate (mm/s)
M204 P1000 R1000 T1000 ; set acceleration (mm/s^2)
M205 X0 Y0 Z0 T0 ; set jerk in (mm/s)
M121 ; disable endstops
G92 E0 X0 Y0 Z0 ; set position to origin
G0 X1
G0 X0
G0 Y1
G0 Y0
G0 Z1
G0 Z0
M400
```

[image]

Note the large jumps in velocity at the beginning and start of a movement.  Are these expected?  A jump from ~0mm/s to 9mm/s with standard settings (80 steps/mm for X and Y, 400 steps/mm for Z) seems like poor performance.  The dwell near 0mm is unexpected as well.  For reference, the max velocity reached here is 25mm/s for X and Y (despite the settings being 250mm/s, not sure why on that either).

This happens with both v1.1.x and v1.1.x bugfix.

I can provide more information or testing if there is consensus that this is in fact an issue and there is a desire to find a solution.

p3p, 2019-04-23:
This looks like standard jerk behaviour, although I'm unsure why it happens when you seem to have zeroed jerk, it's like its still enabled.

Marlin 1.1.x no longer gets fundamental changes only mission critical fixes, so it's unlikely this will be fixed unless the problem is the same in 2.0.x.

timkostka, 2019-04-23:
The behavior in 2.0.x is identical, same steps to reproduce.

p3p, 2019-04-23:
Does the jerk act correctly for other values of jerk than zero?

timkostka, 2019-04-23:
If I bump it up to around 10mm/s, it acts correctly.  But between 0 and 10mm/s, it acts like a combination somewhere between the two.  I'll produce plots showing that in a few hours.

timkostka, 2019-04-23:
Here are plots of X position and velocity at various jerk settings.  Note that the 5mm/s and 10 mm/s are basically perfect, but the 0mm/s and 2mm/s have odd behavior.

I also did the same measurements on a Duet board (Reprap firmware) at 0mm/s jerk.  This is the expected behaviour.  The jerkiness here is only due to the limited resolution of the steps.

[image]

shitcreek, 2019-04-23:
This is the expected behavior as stated in the documentation:  http://marlinfw.org/docs/configuration/configuration.html#acceleration - it's just below this section, I've submitted a commit to make Jerk an anchor link

I'm guessing between 0 and 4mm/s is too slow / indistinguishable for non 32 bit boards

timkostka, 2019-04-24:
@shitcreek I don't see where in the jerk section it would explain this is the expected behavior.  I wouldn't have expected jerk values below ~5mm/s to behave this poorly.

I had thought it may be due to the speed being too slow, but if I slow the movements down, Marlin is perfectly happy chugging along smoothly at well under 1 mm/s.

If I slow the acceleration down on this channel, the jump in velocity diminishes proportionally as well.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
will close this one as it work as expected, we can always reopen if its not the case

closed 2019-09-24 by boelle