MarlinFirmware/Marlin #13795: [Question] Mks Gen L 1.0 with TMC2130

CriNti, 2019-04-23:
Hello all. 

I have Bigtreetech TMC2130 with a 3Dtouch (Bltouch clone) connected to a MKS Gen L 1.0 for two weeks now.. After a lot of tries and research online, i found out that one of the drivers which i didn't have SPI communication at all, was damaged. So i order a replacement and all my drivers seem to work OK before starting the print:

https://imgur.com/iBL32Q2

Recv: 		X	Y	Z	E0
Recv: Enabled		false	false	false	false
Recv: Set current	760	760	760	900
Recv: RMS current	734	734	734	887
Recv: MAX current	1035	1035	1035	1251
Recv: Run current	23/31	23/31	23/31	28/31
Recv: Hold current	11/31	11/31	11/31	14/31
Recv: CS actual	11/31	11/31	11/31	14/31
Recv: PWM scale	1	1	1	1
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true
Recv: msteps		16	16	16	16
Recv: tstep		1048575	1048575	1048575	1048575
Recv: pwm
Recv: threshold	98	98	65	52
Recv: [mm/s]		100.85	100.85	3.04	30.41
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		5	5	5	5
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2
Recv: -start		3	3	3	3
Recv: Stallguard thrs	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E0
Recv: stallguard
Recv: sg_result		0	0	0	0
Recv: fsactive
Recv: stst		X	X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x80:0B:00:00
Recv: 	Y = 0x80:0B:00:00
Recv: 	Z = 0x80:0B:00:00
Recv: 	E0 = 0x80:0E:00:00

After i start the print the printer stops with an E0 error message without saying if it's overheating or Coil Short etc.

https://imgur.com/UIxzuPA

Recv: E0 driver error detected:
Recv: 		 X	Y	Z	E0
Recv: Enabled		true	true	true	true
Recv: Set current	760	760	760	900
Recv: RMS current	734	734	734	887
Recv: MAX current	1035	1035	1035	1251
Recv: Run current	23/31	23/31	23/31	28/31
Recv: Hold current	11/31	11/31	11/31	14/31
Recv: CS actual		23/31	23/31	23/31	28/31
Recv: PWM scale		91	81	77	0
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true
Recv: msteps		16	16	16	16
Recv: tstep		328	1048575	21861	1309
Recv: pwm
Recv: threshold		98	98	65	52
Recv: [mm/s]		100.85	100.85	3.04	30.41
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		5	5	5	5
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2
Recv: -start		3	3	3	3
Recv: Stallguard thrs	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E0
Recv: stallguard
Recv: sg_result		0	0	0	0
Recv: fsactive
Recv: stst			X
Recv: olb					X
Recv: ola					X
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x00:17:00:00
Recv: 	Y = 0x80:17:00:00
Recv: 	Z = 0x00:17:00:00
Recv: 	E0 = 0x60:1C:00:00
Recv: 
Recv: 
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Printing" to "Error: Printer halted. kill() called!"
Changing monitoring state from "Error: Printer halted. kill() called!" to "Offline (Error: Printer halted. kill() called!)"

I checked my wiring a million times and three different 2130 drivers... I'm going crazy.... What am i missing here? 

Thank you very much.

teemuatlut, 2019-04-24:
The print is killed because a driver reported an error state.
Marlin has to go by what the response from the driver was. So you either have bad connection somewhere, or interference from the LCD or SD or something.
You can try the newed bugfix-2.0.x branch as it has updated logic behind the monitoring or if you're certain these are false positives, you can disable just the stop on error or the whole monitoring altogether.

CriNti, 2019-04-24:
Teemuatlut at first i want to thank you for taking the time to respond in my problem. It really means a lot. 

I have tried printing from Octoprint and from SD card too and the problem remains the same. Also i have tried disabling stop on error and what i see is that there is no Extrusion going. 

Just to be specific, when the printing starts, the nozzle is goind at the left of the bed (as usual) in order to leave any unwanted traces of filament. Everything is going ok after the back and forth movement while doing the necessary extrusion. Going to the next point is where the printer stops. 

So... i'll try the bugfix-2.0.x with fingers crossed...

Again, thank you very much.

CriNti, 2019-04-26:
Ok, so with the bugfix-2.0.x and the exact same configuration seems to be working ok! My issue now is the clone of the Bltouch which is very inconsistent but that's another sorry. 

Anyway thank you very much teemualut for the support.

boelle, 2019-07-21:
closing since this seems more like a support question and that should be asked elsewhere
on top of that the issue seems fixed

closed 2019-07-21 by boelle