MarlinFirmware/Marlin #13796: [BUG] G26 Mesh Validation does strange behavior

7effrey89, 2019-04-23:
### Description
When sending G26 the extruder and heatbed will start warming up. After the temperature has reached their target, the nozzle starts dropping with very fast pace (i think it's like 4x-8x times my normal homing speed) towards the heatbed, but will mid-way during drop reset Marlin. See video here: https://www.youtube.com/watch?v=SYE-lW3slPU

Precondition: 10x10 UBL Mesh is already stored in eeprom

1. Send G26 via Pronterface

**Expected behavior:** Was expecting it to print a grid on the heatbed. I'm following Chris's tutorial to every detail at: https://www.youtube.com/watch?v=ONpKxkil16Q

**Actual behavior:**  T

he nozzle starts dropping with very fast pace (i think it's like 4x-8x times my normal homing speed) towards the heatbed, but will mid-way during drop reset Marlin

#### Additional Information
My configuration:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3108982/Marlin.zip)
Demonstration of issue: https://www.youtube.com/watch?v=SYE-lW3slPU
* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

**Log** 
This is the log captured during the video:
Setting bed temperature to 60.000000 degrees Celsius.
Setting hotend temperature to 0.000000 degrees Celsius.
Setting hotend temperature to 200.000000 degrees Celsius.
))) G26 p10
SENDING:G26 P10
G26 command started. Waiting for heater(s).
echo:busy: processing
------It seems like Marlin restarted here ---- The LCD restarted at least..
start
echo:Marlin bugfix-1.1.x
echo: Last Updated: 2018-07-31 | Author: Jeffrey
echo:Compiled: Apr 21 2019
echo: Free Memory: 2221  PlannerBufferBytes: 1488
echo:V55 stored settings retrieved (655 bytes; crc 6361)
Unified Bed Leveling System v1.01 active.
Unified Bed Leveling initialized.
Mesh loaded from slot 1
Mesh 1 loaded from storage.
echo:  G21    ; (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X1280.00 Y1280.00 Z3200.00 E288.23
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z80.00 E50.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P(print_accel) R(retract_accel) T(travel_accel)
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: Q(min_segment_time_us) S(min_feedrate) T(min_travel_feedrate) X(max_x_jerk) Y(max_y_jerk) Z(max_z_jerk) E(max_e_jerk)
echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E2.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Unified Bed Leveling:
echo:  M420 S1 Z10.00
Unified Bed Leveling System v1.01 active.
Active Mesh Slot: 1
EEPROM can hold 7 meshes.
echo:Material heatup parameters:
echo:  M145 S0 H200 B60 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:  M304 P312.01 I16.21 D1501.63
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.00
echo:Stepper driver current:
echo:  M906 X800 Y750 Z800
echo:Hybrid Threshold:
echo:  M913 X100 Y100 Z3
echo:SD card ok

**This is my mesh:**
>>> G29 T
SENDING:G29 T
Bed Topography Report:
    ( 35,115)                                                              (159,115)
        0       1       2       3       4       5       6       7       8       9
 9 | -1.247  -1.243  -1.208  -1.169  -1.046  -0.963  -0.770  -0.619  -0.424  -0.271
   |
 8 | -1.311  -1.288  -1.269  -1.208  -1.133  -1.020  -0.863  -0.658  -0.462  -0.283
   |
 7 | -1.337  -1.344  -1.304  -1.251  -1.134  -1.038  -0.867  -0.674  -0.468  -0.274
   |
 6 | -1.351  -1.368  -1.321  -1.264  -1.157  -1.037  -0.875  -0.690  -0.479  -0.263
   |
 5 | -1.362  -1.375  -1.348  -1.250  -1.149  -1.027  -0.881  -0.707  -0.487  -0.268
   |
 4 | -1.357  -1.384  -1.350  -1.261  -1.155  -1.024  -0.887  -0.713  -0.506  -0.271
   |
 3 | -1.367  -1.379  -1.359  -1.255  -1.150  -1.026  -0.894  -0.719  -0.488  -0.271
   |
 2 | -1.357  -1.366  -1.340  -1.277  -1.151  -1.052  -0.897  -0.726  -0.488  -0.265
   |
 1 | -1.369  -1.371  -1.346  -1.275  -1.184  -1.057  -0.900  -0.732  -0.503  -0.268
   |
 0 | -1.346  -1.354  -1.326  -1.260  -1.177  -1.056  -0.894  -0.718  -0.493  -0.255
        0       1       2       3       4       5       6       7       8       9
    ( 35, 35)                                                              (159, 35)

Roxy-3D, 2019-04-25:
I'm surprised nobody has jumped in with a suggestion.    What ever problem you are seeing...   It isn't being caused by this.   But you should raise the right side of your bed by a milli-meter and re-generate your mesh.      And using those numbers (from the G29 T) and comparing it to the G29 T posted up above...   Do one more iteration of correction.

And if you are still having the original problem...   It is going to be a lot easier to figure out what is going wrong.

7effrey89, 2019-04-28:
I'm afraid i can't lift the left side of the heat bed, because my mk52 heat bed is directly installed to the bed frame. In my early days i had some homing accidents that bent it permanently :(

But i've tried to print stuff, and it seems to work fine in the center area. Haven't tried really small things yet.

Roxy-3D, 2019-04-28:
OK...   I understand.

> But i've tried to print stuff, and it seems to work fine in the center area. Haven't tried really small things yet.

The only reason for 'small' is that has the best chance of succeeding.    Larger, might run into issues that can not be fixed without the bed leveling active.     If you can print things in the center of the bed, that helps a lot in the debugging process because we are then only debugging the bed leveling system.   We are not debugging the basic printer operation.

Can you turn on #define DEBUG_LEVELING_FEATURE in Configuration.h ?   And re-build and load the firmware with that enabled?      Once the machine reboots...    If you give the printer a M111 S255 command before you do anything, it will give fairly verbose information about what it is doing at each step of the process.     

If you cut and paste that output here....   somebody will be able to figure out what is going wrong with your printer.    (Things happen fast...   There is a lot of output.    But if you can show us where things are OK and where the problem actually happens in the output, that will help a lot too.)

7effrey89, 2019-04-28:
Here is my log with  M111 S255 command while trying to perform G26. I've put some of my comments in '//' and bold to highlight what i'm doing and what's happening:

Attempting connection at \\.\COM3...
READ: start
  Connected to machine!
READ: echo:Marlin bugfix-1.1.x
Marlin bugfix-1.1.x
READ: 
READ: echo: Last Updated: 2018-07-31 | Author: Jeffrey Lai - 4/21/19
Last Updated: 2018-07-31 | Author: Jeffrey  - 4/21/19
SENT: T0
READ: echo:Compiled: Apr 28 2019
Compiled: Apr 28 2019
READ: echo: Free Memory: 2191  PlannerBufferBytes: 1488
Free Memory: 2191  PlannerBufferBytes: 1488
READ: echo:V55 stored settings retrieved (655 bytes; crc 6361)
V55 stored settings retrieved (655 bytes; crc 6361)
READ: Unified Bed Leveling System v1.01 active.
READ: 
READ: Unified Bed Leveling initialized.
READ: 
READ: Mesh loaded from slot 1
READ: Mesh 1 loaded from storage.
READ: echo:  G21    ; (mm)
G21    ; (mm)
READ: echo:  M149 C ; Units in Celsius
M149 C ; Units in Celsius
READ: 
READ: echo:Filament settings: Disabled
Filament settings: Disabled
READ: echo:  M200 D1.75
M200 D1.75
READ: echo:  M200 D0
M200 D0
READ: echo:Steps per unit:
Steps per unit:
READ: echo:  M92 X1280.00 Y1280.00 Z3200.00 E288.23
M92 X1280.00 Y1280.00 Z3200.00 E288.23
READ: echo:Maximum feedrates (units/s):
Maximum feedrates (units/s):
READ: echo:  M203 X300.00 Y300.00 Z80.00 E50.00
M203 X300.00 Y300.00 Z80.00 E50.00
READ: echo:Maximum Acceleration (units/s2):
Maximum Acceleration (units/s2):
READ: echo:  M201 X3000 Y3000 Z100 E10000
M201 X3000 Y3000 Z100 E10000
READ: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
READ: echo:  M204 P3000.00 R3000.00 T3000.00
M204 P3000.00 R3000.00 T3000.00
READ: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
READ: echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E2.00
M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E2.00
READ: echo:Home offset:
Home offset:
READ: echo:  M206 X0.00 Y0.00 Z0.00
M206 X0.00 Y0.00 Z0.00
READ: echo:Unified Bed Leveling:
Unified Bed Leveling:
READ: echo:  M420 S1 Z10.00
M420 S1 Z10.00
READ: 
READ: Unified Bed Leveling System v1.01 active.
READ: 
READ: Active Mesh Slot: 1
READ: EEPROM can hold 7 meshes.
READ: 
READ: echo:Material heatup parameters:
Material heatup parameters:
[... 514 lines cut ...]
READ: echo:Material heatup parameters:
Material heatup parameters:
READ: echo:  M145 S0 H200 B60 F0
M145 S0 H200 B60 F0
READ: echo:  M145 S1 H240 B110 F0
M145 S1 H240 B110 F0
READ: echo:PID settings:
PID settings:
READ: echo:  M301 P22.20 I1.08 D114.00
M301 P22.20 I1.08 D114.00
READ: echo:  M304 P312.01 I16.21 D1501.63
M304 P312.01 I16.21 D1501.63
READ: echo:Z-Probe Offset (mm):
Z-Probe Offset (mm):
READ: echo:  M851 Z-1.00
M851 Z-1.00
READ: echo:Stepper driver current:
Stepper driver current:
READ: echo:  M906 X800 Y750 Z800
M906 X800 Y750 Z800
READ: 
READ: echo:Hybrid Threshold:
Hybrid Threshold:
READ: echo:  M913 X100 Y100 Z3
M913 X100 Y100 Z3

Roxy-3D, 2019-04-28:
Did you start the G26 at front left corner of the bed?      It looks like it tried to print a semi-circle before it went crazy.    Can you start the G26 at the center of the bed and post the same log here?

7effrey89, 2019-04-29:
yes, i started at the corner. But now i moved to the center of my bed after homing, and the result is the same:

Log from pronterface:
Connecting...
start
Printer is now online.
echo:Marlin bugfix-1.1.x
echo: Last Updated: 2018-07-31 | Author: Jeffrey  - 4/21/19
echo:Compiled: Apr 29 2019
echo: Free Memory: 2191  PlannerBufferBytes: 1488
echo:V55 stored settings retrieved (655 bytes; crc 6361)
Unified Bed Leveling System v1.01 active.
Unified Bed Leveling initialized.
Mesh loaded from slot 1
Mesh 1 loaded from storage.
echo:  G21    ; (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X1280.00 Y1280.00 Z3200.00 E288.23
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z80.00 E50.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E2.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Unified Bed Leveling:
echo:  M420 S1 Z10.00
Unified Bed Leveling System v1.01 active.
Active Mesh Slot: 1
EEPROM can hold 7 meshes.
echo:Material heatup parameters:
echo:  M145 S0 H200 B60 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:  M304 P312.01 I16.21 D1501.63
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.00
echo:Stepper driver current:
echo:  M906 X800 Y750 Z800
echo:Hybrid Threshold:
echo:  M913 X100 Y100 Z3
Setting hotend temperature to 200.000000 degrees Celsius.
Setting bed temperature to 60.000000 degrees Celsius.
echo:SD card ok
SENDING:G0 X100.0 Y100.0 F3000
echo:endstops hit:  Z:-1.37
echo:busy: processing
echo:busy: processing
echo:busy: processing
SENDING:G0 X100.0 Y100.0 F3000
echo:busy: processing
>>> g26
SENDING:G26
G26 command started. Waiting for heater(s).
echo:busy: processing
echo:busy: processing
echo:busy: processing
start
echo:Marlin bugfix-1.1.x
echo: Last Updated: 2018-07-31 | Author: Jeffrey  - 4/21/19
echo:Compiled: Apr 29 2019
echo: Free Memory: 2191  PlannerBufferBytes: 1488
echo:V55 stored settings retrieved (655 bytes; crc 6361)
Unified Bed Leveling System v1.01 active.
Unified Bed Leveling initialized.
Mesh loaded from slot 1
Mesh 1 loaded from storage.
echo:  G21    ; (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X1280.00 Y1280.00 Z3200.00 E288.23
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z80.00 E50.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E2.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Unified Bed Leveling:
echo:  M420 S1 Z10.00
Unified Bed Leveling System v1.01 active.
Active Mesh Slot: 1
EEPROM can hold 7 meshes.
echo:Material heatup parameters:
echo:  M145 S0 H200 B60 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:  M304 P312.01 I16.21 D1501.63
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.00
echo:Stepper driver current:
echo:  M906 X800 Y750 Z800
echo:Hybrid Threshold:
echo:  M913 X100 Y100 Z3
echo:SD card ok

7effrey89, 2019-04-29:
I found the solution: My Max Z Feed Rate was too high: 80mm/s. When reducing it to 5mm/s it worked fine. Will be trying other speeds as well, but i guess it may be very printer specific for what is considered "too high" speed..

closed 2019-04-29 by 7effrey89

Roxy-3D, 2019-04-29:
During actual printing...    You don't need very fast Z movement.   5mm per second is plenty fast for the bed leveling to do its stuff as the nozzle moves in the X and Y axis.