MarlinFirmware/Marlin #13798: [BUG] Z2 current independent from Z1 with TMC2208

sn4k3, 2019-04-23:
### Description

After install second Z on E1 with TMC2208 UART, it works, but current is independent from Z1.
My test is: 

1. M906 Z10 ; Set to 10mAh
2. Move Z by 50mm
3. Touch or light grab Z1 skips step / stop motor but heavy grab Z2 shaft keeps rotating and skip my fingers, also can't rotate it when is holding while Z1 i can easy rotate just by touch with such current.
I tried to send `M906 I1 Z10` but nothing change.
And i tried multiple configurations

My connections are like image below, except UART for E on that picture goes to Z2 driver:
[image]

Board: MKS GEN L
Drivers:
```
#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208
//#define X2_DRIVER_TYPE TMC2208
//#define Y2_DRIVER_TYPE TMC2208
#define Z2_DRIVER_TYPE TMC2208
#define E0_DRIVER_TYPE LV8729
//#define E1_DRIVER_TYPE TMC2208
//#define E2_DRIVER_TYPE LV8729
//#define E3_DRIVER_TYPE LV8729
//#define E4_DRIVER_TYPE LV8729

```

Pins:
```
#define Z_SERIAL_TX_PIN    42
#define Z_SERIAL_RX_PIN    65
#define Z2_SERIAL_TX_PIN   44
#define Z2_SERIAL_RX_PIN   66

#define E0_SERIAL_TX_PIN   44
#define E0_SERIAL_RX_PIN   66
#define E1_SERIAL_TX_PIN   44
#define E1_SERIAL_RX_PIN   66
```

What i'm doing wrong?

teemuatlut, 2019-04-24:
Have you verified you have successful communication with the drivers?

Paste the `M906` output after sending a command.

sn4k3, 2019-04-24:
> Send: M906
> Recv: X driver current: 600
> Recv: Y driver current: 600
> Recv: Z driver current: 800
> Recv: ok

> Send: M906 Z10
> Recv: ok
> Send: M906
> Recv: X driver current: 600
> Recv: Y driver current: 600
> Recv: Z driver current: 10
> Recv: ok

> Send: M122
> Recv: 		X	Y	Z
> Recv: Enabled		false	false	false
> Recv: Set current	600	600	10
> Recv: RMS current	1049	1049	55
> Recv: MAX current	1479	1479	78
> Recv: Run current	18/31	18/31	0/31
> Recv: Hold current	9/31	9/31	0/31
> Recv: CS actual		0/31	0/31	0/31
> Recv: PWM scale		0	0	0
> Recv: vsense		0=.325	0=.325	0=.325
> Recv: stealthChop	false	false	false
> Recv: msteps		256	256	256
> Recv: tstep		0	0	0
> Recv: pwm
> Recv: threshold		0	0	0
> Recv: [mm/s]		-	-	-
> Recv: OT prewarn	false	false	false
> Recv: OT prewarn has
> Recv: been triggered	false	false	false
> Recv: off time		0	0	0
> Recv: blank time	16	16	16
> Recv: hysteresis
> Recv: -end		-3	-3	-3
> Recv: -start		1	1	1
> Recv: Stallguard thrs
> Recv: DRVSTATUS	X	Y	Z
> Recv: stst
> Recv: olb
> Recv: ola
> Recv: s2gb
> Recv: s2ga
> Recv: otpw
> Recv: ot
> Recv: 157C
> Recv: 150C
> Recv: 143C
> Recv: 120C
> Recv: s2vsa
> Recv: s2vsb
> Recv: Driver registers:	X = 0x00:00:00:00
> Recv: 	Y = 0x00:00:00:00
> Recv: 	Z = 0x00:00:00:00

teemuatlut, 2019-04-24:
You have a connection issue that likely prevents your settings from applying.
See the TMC troubleshooting section
http://marlinfw.org/docs/hardware/tmc_drivers.html#troubleshooting

sn4k3, 2019-04-24:
> You have a connection issue that likely prevents your settings from applying.
> See the TMC troubleshooting section
> http://marlinfw.org/docs/hardware/tmc_drivers.html#troubleshooting

Settings apply to all axis, except Z2.
I will take a look at the link

teemuatlut, 2019-04-24:
Have you enabled `Z_DUAL_STEPPER_DRIVERS`?

teemuatlut, 2019-04-24:
> Yes, i have attached my config files if anyone want take a look.

I'm checking now.

> As i said second Z motor works, but current not set equal like primary z

TMC2208 has build in defaults that allow it to "work" even if you can't connect to it.

sn4k3, 2019-04-24:
I tried now to exhange cable, I move the jumpers 42/65 from Z1 to 44/66 pins, and send M906 Z10, it still apply in debug but motor current not set, so not a cable problem, it look likes it's ignoring something... As i understand when you use Dual Motor when you send X, Y or Z marlin apply to both Z1 and Z2 motor without the need of set current for 1 and 2 right?

> I assume you're using 1.1.9? Try either one of the bugfix branches.

Yes' im using lastest release. I will look at branches and search for bugfixes or related notes

wesk-ms, 2019-04-27:
@teemuatlut if it helps, I can repro this issue with TMC5160s connected over SPI.  I'm using bugfix-2.0.x that I downloaded and built yesterday.  M122 output shows a connection to all Trinamic drivers with no unexpected register values.

M906 Z2500 Z22500 results in only Z being set to 2.5A.

M906 Z2500 results in only Z being set to 2.5A.

The printer prints fine and both Z motors work correctly, just can't set the current on Z2 from the console.

teemuatlut, 2019-04-27:
See the syntax for controlling secondary drivers on an axis.
http://marlinfw.org/docs/gcode/M906.html

sn4k3, 2019-04-28:
> @teemuatlut if it helps, I can repro this issue with TMC5160s connected over SPI. I'm using bugfix-2.0.x that I downloaded and built yesterday. M122 output shows a connection to all Trinamic drivers [...]

Can you try with M906 I1 Z10 ?
Also does Z2 shows on console?

sn4k3, 2019-05-17:
After use bugfix2.0 it start work as expected

teemuatlut, 2019-05-17:
Please always remember to close the issue when a successful solution has been found =)

closed 2019-05-17 by sn4k3