MarlinFirmware/Marlin #13801: [BUG] invalid python 2 vs 3 print()

Andrei-Pozolotin, 2019-04-23:
when trying to migrate python 2 -> 3

[upload_extra_script.py](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_LPC1768/upload_extra_script.py) breaks build

https://sebastianraschka.com/Articles/2014_python_2_3_key_diff.html#the-print-function

p3p, 2019-04-24:
Last time I checked PlatformIO didn't support Python 3 so this wasn't a problem, I see it still doesn't officially (according to the install guide) but they are working towards it, anyway fixing that error isn't a problem just by using `from __future__ import print_function` and moving to Python 3 print syntax,

Andrei-Pozolotin, 2019-04-24:
pio 4.0.0 works fine on py3 when not using pio-ide
https://github.com/platformio/platformio-core/issues/895

p3p, 2019-05-04:
Should be fixed now, tested on linux in a python 3 virtual env.

Andrei-Pozolotin, 2019-05-05:
1. works, thank you
2. perhaps should keep this open for the rest of `*.py` migration

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
closing this one as #13931 did fix the issue, can always reopen if its not the case

closed 2019-09-24 by boelle

Koby88, 2020-02-09:
Hi i hawe this problem like Andrei:
File "/home/work/source/git/Marlin/Marlin/src/HAL/HAL_LPC1768/upload_extra_script.py", line 16
print '\nUnable to find destination disk (' + e + ')\n' \
^
SyntaxError: invalid syntax

I m not a programmer or something like that im just regular guy truing to upload firmware to 3d printer, can you pls tell me what is sollution ?

thinkyhead, 2020-02-09:
I barely know any Python, but according to Google either this…
```py
    print('\nUnable to find destination disk (', e, ')\n' \
          'Please select it in platformio.ini using the upload_port keyword ' \
          '(https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) ' \
          'or copy the firmware (.pio/build/', env.get('PIOENV'), '/firmware.bin) manually to the appropriate disk\n')
```
…or maybe this…
```py
def print_error(e):
    print('\nUnable to find destination disk (%s)\n' \
          'Please select it in platformio.ini using the upload_port keyword ' \
          '(https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) ' \
          'or copy the firmware (.pio/build/%s/firmware.bin) manually to the appropriate disk\n' \
          % (e, env.get('PIOENV')))
```

Koby88, 2020-02-09:
I tried that but with no luck...

thinkyhead, 2020-02-10:
‪Several resources are at your disposal.‬

‪- http://marlinfw.org‬
‪- http://forums.reprap.org/list.php?415‬
‪- https://discuss.toms3d.org/‬
‪- https://www.facebook.com/groups/1049718498464482/‬
‪- https://www.facebook.com/groups/3Dtechtalk/‬
‪- https://www.youtube.com/results?search_query=marlin+configuration‬
‪- https://discord.gg/n5NJ59y‬

PR #13931 (2019-05-08): Convert py scripts for py2 and py3 compatibility
convert all py scirpts from repo to py2 and py3 compatibility

Should fix #13801