MarlinFirmware/Marlin #13807: [BUG 1.1.9] Z_Min is always triggeted (potential bug & not what you think)

gazit6, 2019-04-24:
Hi, I have several printers I built myself and I have a repeating problem with Z homing I cannot resolve.  

Problem description:
When trying to home, the bed is only moving down and not up. M119 is reporting Z_Min is always triggered.
When: it happens suddenly on a machine that worked before with no problem.
It happened in more than one machine already and it is very frustrating as it happens suddenly.
I could solve it once by just loading a new firmware but it did not help on the other machines.
Not sure why the loading of the new firmware helped.

Configuration:
MKS GEN 1.4
BL Touch
Base firmware 1.1.9 bug fix
The controller is running on 24V
UBL is active

Testing I have made:
Probing: tried both BL Touch and inductive sensor, replacing sensor did not help.
Marlin version: tried 1.1.8, 1.1.9 and the latest bug fix with no help.
Changed wires, connected the BL touch directly to board with no help.
Tried disconnecting all other drivers/wires for checking if there is an e-noise in the system with no help.
Tried changing ENDSTOP_INVERTING with no help
Tried using Z_Max instead by changing pins, no help
Cleanup of EEPROM with M502 (factory default) and EEPROM clear from the Arduino IDE did not help

What worked: again, there was one time only that uploading a new firmware somehow solved the problem.
It was actually the standard 1.1.9 Bugfix I am using with standard configuration (BL touch) so I am suspecting
that it is something that is related to memory cleanup or other Marlin bug. Since replacing the firmware once solved it, the hardware is ok.
Also, the fact it happened suddenly on several machines at different times, may suggest a potential bug.

Any ideas?
Thanks!

reloxx13, 2019-04-24:
config missing

gazit6, 2019-04-25:
I was sure I uploaded it. 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3115219/Configuration.zip)

dosono, 2019-04-25:
Strange, I cannot see any problem in your settings.
Is it a board that worked before correctly for some time?
MKS Gen has a pullup resistor on the board itself, maybe it is defected?
Did you try another pin, a pin that is not either the Z_min or Z_Max?

Ketchu13, 2019-04-25:
Sorry if I'm wrong, but don't you have to set the same value for
Z_MIN_PROBE_ENDSTOP_INVERTING and Z_MIN_ENDSTOP_INVERTING since Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is enable ?

marcio-ao, 2019-04-26:
You may want to try adjusting ENDSTOP_NOISE_THRESHOLD or repositioning the wires. A lot of mysterious probing/homing issues can be caused by electrical noise on the wires.

gazit6, 2019-04-28:
I have tried to connect the inductive sensor\bltouch to Zmax instead of Zmin but with no success. 
@marcio-ao I can't seem to find this line of code: ENDSTOP_NOISE_THRESHOLD
Where can I find it?

Aurawin, 2019-05-02:
Try editing Conditions_LCD.h 

// Always disable probe pin inverting for BLTouch
  #undef Z_MIN_PROBE_ENDSTOP_INVERTING
  #define Z_MIN_PROBE_ENDSTOP_INVERTING true
  #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
    #undef Z_MIN_ENDSTOP_INVERTING
    #define Z_MIN_ENDSTOP_INVERTING true
  #endif

This worked for me

gazit6, 2019-05-02:
Thank you for the suggestion. 
I tried to edit what you said but what happened is after pressing 'Home Z' the bed won't go down, bltouch pin is out and then in and then stops.

Aurawin, 2019-05-04:
Ok, so you have to make sure the signal wire is connected to the signal pin on the board.  There is a comment that suggests that you actually switch the wires as you would with a NO/NC limit switch

So you need to check both the wires are installed as instructed by OEM as well as these two changes

Configuration.h
Z_MIN_PROBE_ENDSTOP_INVERTING true
 Z_MIN_ENDSTOP_INVERTING true

Conditionals_LCD.h  remove:

// Always disable probe pin inverting for BLTouch
#undef Z_MIN_PROBE_ENDSTOP_INVERTING
#define Z_MIN_PROBE_ENDSTOP_INVERTING true
#if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
#undef Z_MIN_ENDSTOP_INVERTING
#define Z_MIN_ENDSTOP_INVERTING true
#endif

Configuration_adv.h

#define PINS_DEBUGGING

So when you reboot run a test on the BLTouch 

 M43 S  yields:
		19:24:31.129 : Servo probe test
		19:24:31.129 : .  using index:  0
		19:24:31.129 : .  deploy angle: 10
		19:24:31.130 : .  stow angle:   90
		19:24:31.130 : . probe uses Z_MIN pin: 33
		19:24:31.132 : . uses Z_MIN_ENDSTOP_INVERTING (ignores Z_MIN_PROBE_ENDSTOP_INVERTING)
		19:24:31.133 : . Z_MIN_ENDSTOP_INVERTING: true
		19:24:31.133 : . deploy & stow 4 times
		19:24:36.141 : please trigger probe
		19:24:36.880 : BLTouch compatible probe detected - pulse width (+/- 4mS): 12

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle