MarlinFirmware/Marlin #13808: [BUG?] CR10 S5 BLTouch

RasAlgehti, 2019-04-24:
Hi, I wanted to know if there are still bugs in the Firmware regarding the BLtouch sensor,
I upgraded my Creality CR10 5s with a BLTouch for obvious reasons,
I did so following the guide in the link below,

http://3dmakerita.it/2018/03/21/bltouch-on-cr10s-english-version-i-hope/

I downloaded the latest Marlin firmware for the CR10 5S off of the Creality website,
At first I was able to let the printer run after making alterations to the firmware as described in the guide (see Link),

Yet the Arduino IDE did give an error/warning that I was running out of memory on the board and stability issues could occur, this however seemed like a fairly arbitrary error/warning and I did not really do anything with it.

Next I calibrated the Z-offset through Repetier Host as described in the guide with the M500,
When I first connected the printer to Repetier Host I got an error message describing something that the BLTouch didn;t connect something with an M999 command (Sorr for the vague description, I can't recall the exact warning and I don't have a screen capture unfortunately)
I ignored this warning, as you do, and the printer seemed to function and the G30 & G29 commands did their thing.

Yet trouble arose when I started a print, the printer would  go through its regular start cycle, but then abruptly stop after the G29/G30 command, not proceeding to actually print.

I used the following start G-codes;

CURA DEFAULT START G-CODE:
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0

Did work fine before the M500 Z-offset thing, haven;t ran it after yet.

G-CODE MENTIONED IN THE GUIDE;
G28 ;Home XYZ
G92 E0 ;Home Extruder
G30 Z0.325 ;Autolevel Function change back to command 29 when done
M420 S1 ;Reset Probe

Has never actually worked,

G-CODE FOR MY TEVO LITTLE MONSTER w/ BLTOUCH
G28 ;Home
G92 E0
M280 PS3
G28 ; home all axes
G1 Z50 F25000 ; lift noizzle
G30 Z0.325 ; my noizzle hight - je mehr umso näher zum boden
M280 S7 ; Lift BL Sensor

Did work before the M500 Z-offset thing, but does not work anymore after.
I read that the Z-touch sensor was bugged? But 

TL;DR
Printer works with stock firmware,
Works fine with installed BLTouch and altered firmware
Does not work fine at all, in fact works the polar opposite of 'fine at all' when the Z-offset is determined in Repetier host and saved using the M500 command.
Printer also has trouble communicating with repetier after this.
Don;t know if this is still a bug yes or no, will head for the RepRap forum for help.

1. [First Step] Run the firmware without modifications; printer runs as expected but with a defect in the Z axis offset (expected)
2. [Second Step] Calibrate the Z-offset in Repetier Host, resend and save to EEPROM with M500; printer won't proceed to print after carrying out a G30/G29 command (unexpected)
3. [Third step] Scream in frustration and burn it all down,
[BLTouch CR10S5.zip](https://github.com/MarlinFirmware/Marlin/files/3112610/BLTouch.CR10S5.zip)

FanDjango, 2019-04-24:
Too much too quick. Better step by step.
Do the config stuff, flash the firmware as instructed.
Getting close to the storage limit? Not so very good, depends how close. Save some space somewhere else, for the moment at least - like turn off SD card for just the following testing.
Before you run any serious GCODE, test M48 commands a couple of times, test G29 commands a couple of time (all manually).
etc.
What kind of BLTouch is it? Look at it and see if it is a V3? Go to the ANTClabs website and check if it is a pre April 5 or after April 5 V3?
That should give you some pointers where to start. Don't be frustrated and berfore you burn it tell me where I can pour the water :-)

GMagician, 2019-04-24:
Error you report seems the one raised when input is not correctly working. Have you tryed with M199? 'Triggered' on bltouch input is not a good thing!

Are you sure SERVO0_PIN 11 you have is correct?

CRCinAU, 2019-04-26:
As another note, your procedure should be this via the terminal or directly to the printer.
G28 ; home all axis.
G29 ; probe the bed.
M500 ; save to EEPROM.

Then use M420 S1 in the gcode start. You should use:
G28 ; home all axis
M420 S1 ; enable z map

RasAlgehti, 2019-05-07:
Thanks for the replies, I haven;t been in the office for 2-3 weeks so I only just saw the replies,

I'm not sure if the BLTOUCH triggers it, at least the signal doesn't seem to,
The thing is that now it won't proceed to print after the G29/G30 command and when I try to connect to Repetier I get an error describing that it sent an XXXXXXXX line number but but recieved a different number or something in that gist, I don;t remember exactly what the error described.

Trouble started when I saved the new Z position to EEPROM using Repetier,
Everything is wired correctly and the Z-probe worked before I saved the the new Z_Offset to EEPROM,
Before I did that the printer ran without problem, only that it was printing about 4~mm above the bed.

The printer now won;t proceed to print at all, regardless if there's a G29/G30 command in the start G-code.

I think I'll just re-flash the stock firmware and try again...

Thanks anyway

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

RasAlgehti, 2019-07-21:
Hi Bo,

Thank you for the notice i completely forgot about the matter! 

I haven’t been able to solve the issue. I just reflashed the standard firmware and disabled the auto-level sensor after which the printer ran again. I will attempt to do it again, but that probably wont be until december. I’ll be too busy with my other work at the university until then.

Kind regards,
Ritchie

boelle, 2019-07-21:
no worries, we can always reopen when you get back to it