MarlinFirmware/Marlin #13810: Thermister failure immediately stops hot end fan

Dan018, 2019-04-24:
Using Marlin 2.0 on a Mega 2560 / RAMPS 1.4.

I've mapped D9 as E0_AUTO_FAN_PIN as I exclusively print in ABS, so have no need for a part cooling fan. 

The fan works fine, auto starting and stopping when the temperature threshold is reached.

Recently while doing a hot end PID tune the hot end thermister connector came away from the RAMPS board. The printer identified this and stopped heating as the temperature was showing -14 degrees.

However when the printer halted, the fan was also turned off, presumably because -14 degrees is below the threshold of 50. This also happened when I used a different pin.

I'm using a Titan extruder which has a plastic holder for the fan clipped directly to the hot end so needs the fan to be running while it's hot.

I don't think this is a bug necessarily, but I can't see a setting along the line of "enable all fans if thermal runaway detected". Would it be possible to add such a setting ?

psavva, 2019-04-25:
The idea behind a thermal runaway is to shut down the system.  You do but wasn't to Street l start turning on fans if a real fire broke out.

Dan018, 2019-04-25:
That's a valid point, running a fan with a fire is not what you want. Its a bit counterintuative but not running the fan, even for a couple of minutes, is more likely to cause a fire than stopping it immediately.

7eggert, 2019-04-26:
In that case the fan should be wired to VCC? On my printer, there was one fan continuously cooling the extruder and one cooling the printed object on demand.

Dan018, 2019-04-26:
Yes, wiring the fan to VCC would leave it running, but would also continue to run if thermal runaway shut the system down as per psavva's comment above. I've put a quick and dirty code edit in to the watch loop that triggers thermal runaway to spin the fan for 2 minutes which seems to work ok.

closed 2019-04-26 by Dan018