MarlinFirmware/Marlin #13812: SD Interferes With Extruder Temperature

linuxjunkie999, 2019-04-24:
Hardware
     Creality Ender 3 Pro
             ATMEGA1284P
          

Development Platform
Arduino IDE 1.8.9 With Mighty Core 2.0.1
Arduino IDE 1.9 Beta With Mighty Core 2.0.1
Platform IO using Microsoft Visual Code and Mighty Core 2.0.1

Software Affected
Marlin 1.1.x
Marlin 1.1.x BugFix
Marlin 2.0 Bugfix

Configuration of Marlin
I have tried every conceivable configuration of settings related to temperature and SD card.

SD Init and SD Report Change the Extruder Temp

SD Init in lcd menu sets extruder temperature to -14
SD List in ping debugging temporarily shows extruder temp on the lcd a random positive number depending how many times M20 is sent

### Steps to Reproduce SD INIT BUG

WHAT I EXPECT TO HAPPEN:The SD card to be detected and the extruder temperature to NOT BE SET TO -14 if temperature is in 2 digits or loose digit if temperature is in 3 digits OMG.

WHAT REALLY HAPPENS: The SD card is detected but the extruder temperature is somehow set to -14 or looses a digit.

1. turn on the printer
2. make note of the extruder temperature
3. navigate to the SD Init option in the LCD menu
4. click it
5. navigate back to the main menu
6. note that the temperature is now -14

### Steps to Reproduce SD List Temporarily Sets Extruder Temperature to a random positive number that is affected by how many M20 commands are sent in a row.

WHAT I EXPECT M20 TO DO: list the contents of the SD card in the serial console output and NOTHING ELSE.

WHAT REALLY HAPPENS WHEN M20 IS SENT: the SD contents are listed but then also the extruder temperature flashes to a different temperature and back to whatever it was before and if M20 is spammed the temperature flashed increases proportional to how many times M20 is spammed.

1. turn on the printer
2. open arduino serial console
3. spam M20.
4. watch the extruder temperature go from whatever it is now to other temperatures that seem to be proportional in increase to how many times you spammed M20.

#### Additional Information

[CONFIGFILES.zip](https://github.com/MarlinFirmware/Marlin/files/3114117/CONFIGFILES.zip)

linuxjunkie999, 2019-04-25:
1 day has passed no response to even let me know the bug report has been read.

MikeRich88, 2019-04-26:
I can confirm the second issue also on my Ender 3 Pro

gloomyandy, 2019-04-26:
I think you need to provide some more details about this problem. At the moment this sounds very much like the glitches and screen problems that you see when using a display that has a shared SPI with the SD card. So....
What has changed? Did you have a previous setup that worked better than things do now? If so what is different? Have you made any changes to your hardware? Marlin version are you now using a newer version of Marlin(if so what version was working OK)?

Supplying a link to a short video clip of the problem will probably make it easier for someone to say if it is the known problem with some displays or something new. At the moment I suspect that most devs will just assume this is the known glitch problem and not do anything about it.

linuxjunkie999, 2019-04-26:
Q:Previous configuration that worked?
A: As I stated the stock firmware works but does not have thermal protection. The stock firmware is a heavily modified version of Marlin 1.1.6.

Q: which versions work?
A: the only version that works is the stock firmware included (modified version of Marlin 1.1.6)

Q: video
A: I will make a video clip for both problems I described and upload.

linuxjunkie999, 2019-04-26:
[

](url)

linuxjunkie999, 2019-04-26:
The current config was a last desperate attempt that enabled pidbedtemp and too wide a variance of temperature tolerance, I will change those back and send the second video of the extruder and bed heating successfully.

linuxjunkie999, 2019-04-26:
This video shows the init sd problem

https://www.dropbox.com/s/owvm92m5l1nqs2e/IMG_7295.MOV?dl=0

linuxjunkie999, 2019-04-26:
[AnotherConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/3123200/AnotherConfigFiles.zip)
The corrected config files

linuxjunkie999, 2019-04-26:
https://www.dropbox.com/s/a32tlmamy9ywudi/IMG_7297.MOV?dl=0

Showing the extruder and bed heating

linuxjunkie999, 2019-04-26:
https://www.dropbox.com/s/fd2xcy9fal2mcg8/IMG_7298.MOV?dl=0

The loose a digit if temperature is in 3 digits.

linuxjunkie999, 2019-04-26:
https://www.dropbox.com/s/fi7vjja08jug65j/IMG_7302.MOV?dl=0

The spam M20 and get flashes of different temperature that the temperature shown is proportional in increase to how many m20 commands where spammed. The sd init Big has to be activated first before this phenomenon exhibits.

linuxjunkie999, 2019-04-26:
> I think you need to provide some more details about this problem. At the moment this sounds very much like the glitches and screen problems that you see when using a display that has a shared SPI [...]

Thank You for paying attention to my problem and responding . Hopefully what I have provided will help in figuring out the problem.

gloomyandy, 2019-04-27:
Have you made any changes to any other files (or have you downloaded any of them from somewhere else), like for instance have you made changes to the pins file? If so you need to post them here. 

You should probably run an M43 and post the output here. 

You may also get more luck posting questions on a general 3D printer forum. Your printer is very popular and I'm sure that many, many people have your configuration working happily with Marlin. I'm pretty sure that your problem will turn out to be a configuration problem of some kind.

Ketchu13, 2019-04-27:
@linuxjunkie999 The Ender 3 Pro original LCD is a RRD Full Graphics ?
In the config file you provide, you have enabled the `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`
instead of `CR10_STOCKDISPLAY`
https://github.com/MarlinFirmware/Marlin/blob/5d96a6d983c065403216ed52a8c92ed22dc268ba/Marlin/example_configurations/Creality/Ender-3/Configuration.h#L1743-L1745

linuxjunkie999, 2019-04-28:
Q: Is the LCD a reprap?
A: I read and translated over exactly what was in the configuration.h and configuration_adv.h of the source code for the ender 3 pro provided by creality. So yes the LCD is a reprap discount. 

Q: did I change any other files? 
A: no I did not.

Q: configuration issue?
A: doubtful as I translated directly from the source code provided by creality.

gloomyandy, 2019-04-28:
As above please post the output from running M43. Also please post the memory usage summary produced when you compile your firmware using the Arduino IDE.

Oh and I'm not sure that it makes sense to "translated directly from the source code provided by creality" as they may have made other changes to Marlin that do not match the standard setup. You may be better off starting from the Marlin example files for your printer, or obtaining a working configuration for standard Marlin from another user of the same printer.

linuxjunkie999, 2019-04-29:
What I meant was I went line by line and compared the configuration.h and configuration_adv.h from the source code provided by Creality with the default config and made the appropriate changes to things, I opted to not have a custom bootscreen image so i did not activate any of that or copy those files over.

(The source code from Creality)
[ENDER3 PRO SOURCE CODE.zip](https://github.com/MarlinFirmware/Marlin/files/3129035/ENDER3.PRO.SOURCE.CODE.zip)

(Free Memory)
14:59:26.072 -> echo: Free Memory: 11706  PlannerBufferBytes: 1232
14:59:26.105 -> echo:V55 stored settings retrieved (655 bytes; crc 28777)

("M43 I"output)
[M43 I Output.txt](https://github.com/MarlinFirmware/Marlin/files/3129022/M43.I.Output.txt)

linuxjunkie999, 2019-04-29:
(Compile memory and storage usage)
Sketch uses 127494 bytes (98%) of program storage space. Maximum is 130048 bytes.
Global variables use 4632 bytes (28%) of dynamic memory, leaving 11752 bytes for local variables. Maximum is 16384 bytes.

linuxjunkie999, 2019-04-29:
The entire output from the compile. Notice there are a bunch of "-wunused parameter" warnings for U8glib.

[EntireCompileOutput.txt](https://github.com/MarlinFirmware/Marlin/files/3129073/EntireCompileOutput.txt)

linuxjunkie999, 2019-04-29:
Is there anymore information you need?

You have mentioned more than once that I should find a working configuration.h and configuration_adv.h, I have looked and have not found such.

The built in profile for the Ender 3 is not for the Ender 3 Pro. ( there are some hardware differences like the LCD)

I am willing to work with one of you in real time via a phone call or a remote desktop session, please let me know.

gloomyandy, 2019-04-29:
Looking at the output from M43 you have a clash with TEMP_0_PIN (Analog pin 7) and the LCD_SDSS (pin 31). I'm no expert on the Arduino but it looks to me like this may be because of the way you are compiling. It seems that when using Mighty Core there are (at least) three different pinout mappings available. It looks like you are using "standard" which maps A7 to pin 31, but Marlin is setup expecting the "sanguino" pinout which maps A7 to pin 24. I think you can select the pinout in the tools menu of the Arduino IDE. For more details see: https://elementztechblog.wordpress.com/2016/10/28/mightycore-an-arduino-core-for-the-atmega16-atmega32-atmega324-and-more/ and https://github.com/MCUdude/MightyCore#pinout

So basically I think you need to select a different pinout and recompile.

linuxjunkie999, 2019-05-01:
I will try compiling using the sanguino pinout and let you know the results.

linuxjunkie999, 2019-05-01:
Problem seems resolved. Thankyou gloomyandy.

closed 2019-05-01 by linuxjunkie999