MarlinFirmware/Marlin #13813: [FR] Endstop for extruder and EMAX and EMIN (for syringe-style dispensers)

marcio-ao, 2019-04-24:
We are developing a bio-printer that uses a syringe instead of a continuous feedstock on the extrusion axis. Because of this, it will be necessary to add an endstop to the E axis and to have limits on that axis.

How difficult would it be to add this type of functionality to Marlin?

psavva, 2019-04-25:
Would it not be easier to start with a full syringe and knowing the travel required to empty it from a linear motion perspective?

How would the endstops on the extruder axis help here?  You would always start with an empty syringe then somehow fill it automatically?

robbycandra, 2019-04-25:
@psavva i think to start with full syringe, we need E_MIN_STOP at least.

marcio-ao, 2019-04-25:
Honestly, I haven't put that much thought into it. It was more of a general question to gauge how difficult it would be to implement something like this. At the minimum, if not hardware endstops, I would need to be a button that the user could press when the syringe was "full", that would issue "G92 E0"; then Marlin would need software endstops that would keep the extruder from bottoming out the syringe.

However, if Marlin had hardware endstops, it would be possible to wire them up to the stallguard output of Trinamic drivers, which would allow the printer to know when the syringe had bottomed out or became clogged.

psavva, 2019-04-26:
I think if you had to treat the endstop as a filament runout sensor, it would probably be the best...

IE: when hitting the endstop trigger the filament runout pin, and in this way, your extrusion will stop when you run out of material...

marcio-ao, 2019-04-26:
@psavva: This is an interesting idea. I'll ask our hardware people whether that could be done. Of course, there would still be a problem is the user retracted the other way. At some point the plunger will pop out of the syringe and will reach a travel limit.

psavva, 2019-04-26:
I guess the same could be said for filament.  If you retracted it enough, it would pop out of the extruder.   Using the same logic, I would expect that retraction on the syringe would be minimal and should be zero chance of it popping out given the retraction is set to some logical value.

marcio-ao, 2019-04-26:
@psavva: I am not referring to retraction, the limits would be useful to keep the user from causing the linear axis from bottoming out. The reason for E_MAX and E_MIN would be for idiot-proofing.

DerAndere1, 2019-06-08:
@marcio-ao You can have a look at my [e_homing](https://github.com/DerAndere1/Marlin/tree/e_homing) branch but I recommend also to have a look at the [Marlin2ForPipetBot] (https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot/Marlin) branch of my marlin fork. The latter is configured for my liquid handling robot pipetBot-A8. You pick my commits individually, it will be quite easy for you merge it with current Marlin2.0 code:  https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot/

Using wx.python+mecode I also developped [GGCGen](https://gitlab.com/RobotsByDerAndere/CartesianRobotsByDerAndere/GGCGen), a GUI for creating gcode-scripts for lab robots that run Marlin. 
More recources on open source lab robots can be found on the project homepage https://it-by-derandere.blogspot.com/p/pipetbot-a8.html

robbycandra, 2019-06-09:
Interesting, you use marlin for lab robot.

Grogyan, 2019-06-09:
Curious question.

Marlin currently supports a filament runout sensor, could this not be utilised to detect when the syringe has bottomed out?

Not ideal, but maybe a work around till something could be implemented?

marcio-ao, 2019-06-10:
@DerAndere1: That's great! I'll have to take a look at your branch and see if there is anything we can use!

DerAndere1, 2019-06-10:
There is also an Old discussion at https://github.com/MarlinFirmware/Marlin/issues/8519 and the customized marlin1 from the evobot Software https://bitbucket.org/afaina/evobliss-software/src/master/

EhsanNri, 2023-07-25:
Hi @marcio-ao and @DerAndere1 
I am working on a chocolate 3d printer and the same as your project. I have a cylinder and a piston to extrude the chocolate instead of a syringe, but the mechanisms are the same. I need my piston reach to the min position by touching a micro switch same as the x y z axis home option and limited travel to about 5 cm. Please let me know how to implement it!!

DerAndere1, 2023-07-25:
@EhsanNri : One way is to have a look at the last 4 commits in my 2.0.x_E_homing branch: https://github.com/DerAndere1/Marlin/commits/2.0.x_E_homing . Another way is to configure a linear axis instead of an extruder for the syringe. Have a look at options  `Ì_DRIVER_TYPE`, `AXIS4_NAME`, `I_MIN_POS` and `I_MAX_POS` in Configuration.h and define `I_MIN_PIN` or `I_STOP_PIN` in your pins_YOUR_MOTHERBOARD.h file

EhsanNri, 2023-07-26:
@DerAndere1 Thank you for your reply!
I couldn't find the solution and the Marlin file you mentioned doesn't work. Please let me know how to implement these changes in marlin. Thanks a lot!!