MarlinFirmware/Marlin #13817: [FR] [2.0.x-bugfix] Moving Z-Offset right after Toolchange

cad435, 2019-04-25:
Related to https://github.com/MarlinFirmware/Marlin/issues/13695
@InsanityAutomation 

Immediately After a Tool-Change, Z-Axis should move the amount of OFFSET_Z

> Z-offset should be corrected as soon as the Tool-change is requested. In most gcode is no "G1 Z" command after Toolchange, so the Printer knows now it is offset but won't correct the offset till the next Layer-change, as there is a absolute "G1 Z" command. Same thing applies to all axis if your gcode contains incremental movements (but personally I don't know anyone who does this, nor do I know if theres any benefits)
In Slic3r I can put a 'G1 Z[layer_z]' command after my Tool-Change G-code, wich will move the Z-Axis and therefore actually correct the Z-Offset. I don't know if other slicers have such a "Z-Layerheight" Variable.

cad435

UticaTechClub, 2019-04-26:
Hi there,
This is something I have recently worked on and succeeded in my multi-tool configuration. I am on latest `bugfix-2.0.x`

How do you currently set your `T1...T<n>` offsets?
Why have you derived such a conclusion that Z offset will only be applied after `G1 Zx` command?

InsanityAutomation, 2019-04-26:
@UticaTechClub Only relevant for IDEX in Full Control Mode. Other dual extrusion or IDEX modes are not affected.

BTW, is that Utica, MI?

cad435, 2019-04-26:
> Only relevant for IDEX

corrected title to be more specific...

UticaTechClub, 2019-04-26:
> @UticaTechClub Only relevant for IDEX in Full Control Mode. Other dual extrusion or IDEX modes are not affected.
> 
> BTW, is that Utica, MI?

Upstate New York

InsanityAutomation, 2019-05-25:
Can you test with #14117

cad435, 2019-05-27:
Right now I get the following Compilation Error:

```
In file included from Marlin\src\module/stepper.h:226:0,
from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper_indirection.h:210:24: error: 'X2_AXIS' was not declared in this scope
extern TMC_CLASS(X2) stepperX2;
^
Marlin\src\module/stepper_indirection.h:65:78: note: in definition of macro '____TMC_CLASS'
#define ____TMC_CLASS(MODEL, A, I, E) TMCMarlin<TMC##MODEL##Stepper, A, I, E>
^
Marlin\src\module/stepper_indirection.h:67:39: note: in expansion of macro '___TMC_CLASS'
#define __TMC_CLASS(MODEL, A, I, E) ___TMC_CLASS(_##MODEL, A, I, E)
^
Marlin\src\module/stepper_indirection.h:68:35: note: in expansion of macro '__TMC_CLASS'
#define _TMC_CLASS(MODEL, L, E) __TMC_CLASS(MODEL, L, E)
^
Marlin\src\module/stepper_indirection.h:69:25: note: in expansion of macro '_TMC_CLASS'
#define TMC_CLASS(ST) _TMC_CLASS(ST##_DRIVER_TYPE, TMC_##ST##_LABEL, ST##_AXIS)
^
Marlin\src\module/stepper_indirection.h:210:14: note: in expansion of macro 'TMC_CLASS'
extern TMC_CLASS(X2) stepperX2;
^
Marlin\src\module/stepper_indirection.h:65:79: error: template argument 4 is invalid
#define ____TMC_CLASS(MODEL, A, I, E) TMCMarlin<TMC##MODEL##Stepper, A, I, E>
^
Marlin\src\module/stepper_indirection.h:66:40: note: in expansion of macro '____TMC_CLASS'
#define ___TMC_CLASS(MODEL, A, I, E) ____TMC_CLASS(MODEL, A, I, E)
^
Marlin\src\module/stepper_indirection.h:67:39: note: in expansion of macro '___TMC_CLASS'
#define __TMC_CLASS(MODEL, A, I, E) ___TMC_CLASS(_##MODEL, A, I, E)
^
Marlin\src\module/stepper_indirection.h:68:35: note: in expansion of macro '__TMC_CLASS'
#define _TMC_CLASS(MODEL, L, E) __TMC_CLASS(MODEL, L, E)
^
Marlin\src\module/stepper_indirection.h:69:25: note: in expansion of macro '_TMC_CLASS'
#define TMC_CLASS(ST) _TMC_CLASS(ST##_DRIVER_TYPE, TMC_##ST##_LABEL, ST##_AXIS)
^
Marlin\src\module/stepper_indirection.h:210:14: note: in expansion of macro 'TMC_CLASS'
extern TMC_CLASS(X2) stepperX2;
^
Compiling .pioenvs\megaatmega2560\src\src\feature\bedlevel\mbl\mesh_bed_leveling.cpp.o
*** [.pioenvs\megaatmega2560\src\src\Marlin.cpp.o] Error 1

In file included from Marlin\src\feature\../module/stepper.h:226:0,
from Marlin\src\feature\babystep.cpp:30:
Marlin\src\feature\../module/stepper_indirection.h:210:24: error: 'X2_AXIS' was not declared in this scope
extern TMC_CLASS(X2) stepperX2;
^
Marlin\src\feature\../module/stepper_indirection.h:65:78: note: in definition of macro '____TMC_CLASS'
#define ____TMC_CLASS(MODEL, A, I, E) TMCMarlin<TMC##MODEL##Stepper, A, I, E>
^
Marlin\src\feature\../module/stepper_indirection.h:67:39: note: in expansion of macro '___TMC_CLASS'
#define __TMC_CLASS(MODEL, A, I, E) ___TMC_CLASS(_##MODEL, A, I, E)
^
Marlin\src\feature\../module/stepper_indirection.h:68:35: note: in expansion of macro '__TMC_CLASS'
#define _TMC_CLASS(MODEL, L, E) __TMC_CLASS(MODEL, L, E)
^
Marlin\src\feature\../module/stepper_indirection.h:69:25: note: in expansion of macro '_TMC_CLASS'
#define TMC_CLASS(ST) _TMC_CLASS(ST##_DRIVER_TYPE, TMC_##ST##_LABEL, ST##_AXIS)
^
Marlin\src\feature\../module/stepper_indirection.h:210:14: note: in expansion of macro 'TMC_CLASS'
extern TMC_CLASS(X2) stepperX2;
^
Marlin\src\feature\../module/stepper_indirection.h:65:79: error: template argument 4 is invalid
#define ____TMC_CLASS(MODEL, A, I, E) TMCMarlin<TMC##MODEL##Stepper, A, I, E>
^
Marlin\src\feature\../module/stepper_indirection.h:66:40: note: in expansion of macro '____TMC_CLASS'
#define ___TMC_CLASS(MODEL, A, I, E) ____TMC_CLASS(MODEL, A, I, E)
^
Marlin\src\feature\../module/stepper_indirection.h:67:39: note: in expansion of macro '___TMC_CLASS'
#define __TMC_CLASS(MODEL, A, I, E) ___TMC_CLASS(_##MODEL, A, I, E)
^
Marlin\src\feature\../module/stepper_indirection.h:68:35: note: in expansion of macro '__TMC_CLASS'
#define _TMC_CLASS(MODEL, L, E) __TMC_CLASS(MODEL, L, E)
^
Marlin\src\feature\../module/stepper_indirection.h:69:25: note: in expansion of macro '_TMC_CLASS'
#define TMC_CLASS(ST) _TMC_CLASS(ST##_DRIVER_TYPE, TMC_##ST##_LABEL, ST##_AXIS)
^
Marlin\src\feature\../module/stepper_indirection.h:210:14: note: in expansion of macro 'TMC_CLASS'
extern TMC_CLASS(X2) stepperX2;
^
*** [.pioenvs\megaatmega2560\src\src\feature\babystep.cpp.o] Error 1
```

InsanityAutomation, 2019-05-27:
Someone else posted a similar error with multi-axis using TMC drivers. That's unrelated to this bug fix but needs to be resolved regardless. I'll try to take a look at this one soon as well if one of the guys on the TMC side doesn't get it taken care of.

InsanityAutomation, 2019-10-15:
@boelle @shitcreek this one is long gone as well

closed 2019-10-15 by boelle