MarlinFirmware/Marlin #13821: [BUG]SD Card kind of shows up but doesn't wrok

doggyfan, 2019-04-25:
I have the SKR 1.3 controller board, I thought I'd disable the SD card on the display board (REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER) and use the SD card on the SKR board for the GCode files.

I have Configuration_adv.h set too ..

`
// LPC-based boards have on-board SD Card options. Override here or defaults apply.
  #ifdef TARGET_LPC1768
    //#define LPC_SD_LCD          // Use the SD drive in the external LCD controller.
    #define LPC_SD_ONBOARD      // Use the SD drive on the control board. (No SD_DETECT_PIN. M21 to init.)
    //#define LPC_SD_CUSTOM_CABLE // Use a custom cable to access the SD (as defined in a pins file).
    #define USB_SD_DISABLED     // Disable SD Card access over USB (for security).
    #if ENABLED(LPC_SD_ONBOARD)
      //#define USB_SD_ONBOARD    // Provide the onboard SD card to the host as a USB mass storage device.
    #endif
  #endif
`

The gcode files show up OK over the USB serial connection (which is good).

But, I get an inaccessible disk drive show up on my windows 7 laptop as ..

  "Removeable Disk (D:)"

Would be nice if it worked as per usual USB mass storage drive if it's going to show up like that, otherwise I'd have though it shouldn't be showing up at all being as access is disabled ?

I'm unable to configure Marlin (bugfix 2.0) to use the SKR 1.3 on board SD card for GCode files AND have a working accessible USB mass storage drive :(

ManuelMcLure, 2019-04-25:
Marlin and the USB port can't access the card at the same time. You need to use the ``M21`` (Init SD card) and ``M22`` (Release SD card) gcodes to switch what interface is using the card. Try an ``M22`` and see if the card is now accessible from Windows. Then you'll need to do an ``M21`` so that Marlin can see the card again.

doggyfan, 2019-04-26:
I get an 'ok' back from Marlin when I send M22 to it, but no the M22 command doesn't allow Mass storage SD card access ?    there is no change to how the SD card shows up in Windows 7.

I've changed the **Configuration_adv_h** options to try and allow the M22 command to work (presumably) too ..

  // LPC-based boards have on-board SD Card options. Override here or defaults apply.
  #ifdef TARGET_LPC1768
    //#define LPC_SD_LCD          // Use the SD drive in the external LCD controller.
    #define LPC_SD_ONBOARD      // Use the SD drive on the control board. (No SD_DETECT_PIN. M21 to init.)
    //#define LPC_SD_CUSTOM_CABLE // Use a custom cable to access the SD (as defined in a pins file).
    //#define USB_SD_DISABLED     // Disable SD Card access over USB (for security).
    #if ENABLED(LPC_SD_ONBOARD)
      #define USB_SD_ONBOARD    // Provide the onboard SD card to the host as a USB mass storage device.
    #endif
  #endif

The only way it seems to get the SD card to show up properly in Windows is to move Marlin back over to using it's display SD card for the GCode files. Then the Mass storage on board controller SD card shows up properly (with it's proper SD card name) and is fully accessible from Windows.

ManuelMcLure, 2019-04-26:
Hmmm... that works for me on Windows 10 and with Linux on my Raspberry Pi running Octoprint connecting to my Re-ARM which uses the same LPC1768 as your SKR. Right after connecting the USB cord (and thus booting the board) I get an error message from Windows saying there’s a problem with the drive, but a few seconds later it appears fine. Maybe W7 doesn’t handle things quite the same.

doggyfan, 2019-04-26:
Just tried it on a Windows 10 Pro machine. We're getting the same result, the Mass storage SD card is not working.

Need one of the dev team to try this on the SKR 1.3 board.

gloomyandy, 2019-04-26:
Try using a different SD card (yes I know that might sound odd), I found when testing this code on the SKR V1.1 (which for this side of things is pretty much the same as the V1.3) that with the "free" card that comes with the board (which is very small and I think very slow), that USB access is very hit and miss. I have both V1.1 and V1.3 boards working happily with both Marlin and PC access to the onboard SD reader (with both Linux and Windows 10). Note though that as above we only allow either USB or Marlin access to the card at a time, so you need to switch access using M21/M22. I think that with a display at boot then USB access is provided (but there has been a whole different discussion on that!).

Oh and if you don't define anything in the config files then I think the default is to use the onboard reader for both Marlin and USB access. 

But having said that if you have USB_SD_DISABLED and do not have USB_SD_ONBOARD defined then it should not show up at all. I will have to investigate that at some point.

gloomyandy, 2019-04-26:
Sorry ignore the comment about the default config, I think that is the case for the V1.1 board.

doggyfan, 2019-04-26:
Thank you gloomyandy.

I was indeed using the little 128MB card that came with the board. Just tried a 4GB card and up it pops in Windows 7 shortly after sending the M22 command to the board.

I did say I might well be missing the obvious ;)

I'll close this then.

Thank you again.

closed 2019-04-26 by doggyfan

gloomyandy, 2019-04-26:
Thanks for the update!

I suspect that the "free" SD is a relatively slow card. Our SD card code runs the SPI bus at a fixed speed (after the initial connect) and makes no attempt to adjust the bus speed based on the card capabilities. I suspect that is the problem. I suppose that we could fix it, but given that pretty much all currently available cards operate at way faster speeds than we use, it may not be worth it.

doggyfan, 2019-04-26:
Right, Okey doke, thank you for the info.

What I don't yet get is why the little free 128MB SD card that comes with the SKR 1.3 board works 100% of the time without fail if Marlin is set to use the display SD card for it's GCode files.

I guess Marlins init' of the card might be different in it's two modes of config ? .. no idea

Anyway, I'll no doubt look into it at some point to see what shows up, but I first need to familiarize myself with the code (takes time for that).

gloomyandy, 2019-04-26:
The thing to remember is that there are two totally separate SD card modules here. One built into Marlin and used when Marlin is doing stuff with the card and one built into the LPC176x framework and used for the USB access. They both meet way down at the hardware SPI level. The problem seems to be when the card is initially accessed via the USB code and then by Marlin and then again by the USB code, something in that process seems to lock that card up. I suspect it is something to do with the switch from the basic initialisation into "full speed" mode. But since it only seems to happen with that small/slow card and not with any of the others I've used (and I've tried about a dozen different size and speed cards) and the only reports of problems have been with the cards supplied with those SKR boards, I've not looked any further into it. There are more important things to be fixed in Marlin!

doggyfan, 2019-04-26:
Thank you again gloomyandy.

It's that kind of sharing of code insight that helps me a great deal. Shame their wasn't flow charts and/or block diagrams of the code.

Yes, totally with you on their being more important things to fix. Leave you in peace now.

Thank you again.

robbycandra, 2019-05-26:
I got same problem here.

gloomyandy, 2019-05-26:
@robbycandra What problem do you have? As above have you tried using an SD card that is not the one that comes free with the SKR boards? If you have and you still have a problem then you should open a new issue and supply full details of whatever problem it is you have and your configuration files.

robbycandra, 2019-05-27:
Thanks, @gloomyandy,  Problem solved after change Free 128MB SD Card with new SDCard.