MarlinFirmware/Marlin #13826: [BUG] (short description) Unable to compile for TMC2208

jimmyeao, 2019-04-26:
### Description

### Steps to Reproduce

Arduino 1.8.9, TMCStepper library installed (0.3.1) TMCStepper2208 (0.2.5)

Error on compile
Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from c:\users\jimmy\appdata\local\temp\arduino_build_338611\sketch\configuration_adv.h:2:0,

                 from c:\users\jimmy\appdata\local\temp\arduino_build_338611\sketch\src\inc\MarlinConfigPre.h:38,

                 from c:\users\jimmy\appdata\local\temp\arduino_build_338611\sketch\src\inc\marlinconfig.h:28,

                 from sketch\src\feature\tmc_util.cpp:23:

D:\OneDrive\Documents\Arduino\libraries\TMC2208Stepper\src/TMC2208Stepper_REGDEFS.h:6:24: error: expected unqualified-id before numeric constant

 #define TMC2208_SYNC   0x05

                        ^

D:\OneDrive\Documents\Arduino\libraries\TMCStepper\src/TMCStepper.h:899:29: note: in expansion of macro 'TMC2208_SYNC'

   static constexpr uint8_t  TMC2208_SYNC = 0x05,

                             ^

sketch\src\feature\tmc_util.cpp: In function 'TMC_driver_data get_driver_data(TMC2208Stepper&)':

D:\OneDrive\Documents\Arduino\libraries\TMC2208Stepper\src/TMC2208Stepper_REGDEFS.h:182:20: error: expected unqualified-id before numeric constant

 #define OTPW_bp    0

                    ^

sketch\src\feature\tmc_util.cpp:144:25: note: in expansion of macro 'OTPW_bp'

       constexpr uint8_t OTPW_bp = 0, OT_bp = 1;

                         ^

D:\OneDrive\Documents\Arduino\libraries\TMC2208Stepper\src/TMC2208Stepper_REGDEFS.h:209:26: error: expected unqualified-id before numeric constant

 #define CS_ACTUAL_bm     0x1F0000ul

                          ^

sketch\src\feature\tmc_util.cpp:147:26: note: in expansion of macro 'CS_ACTUAL_bm'

       constexpr uint32_t CS_ACTUAL_bm = 0x1F0000; // 16:20

                          ^

D:\OneDrive\Documents\Arduino\libraries\TMC2208Stepper\src/TMC2208Stepper_REGDEFS.h:195:24: error: expected unqualified-id before numeric constant

 #define STEALTH_bp     30

                        ^

sketch\src\feature\tmc_util.cpp:148:25: note: in expansion of macro 'STEALTH_bp'

       constexpr uint8_t STEALTH_bp = 30, STST_bp = 31;

                         ^

exit status 1
In file included from c:\users\jimmy\appdata\local\temp\arduino_build_338611\sketch\configuration_adv.h:2:0,

                 from sketch\src\inc/MarlinConfigPre.h:38,

                 from sketch\src\inc/MarlinConfig.h:28,

                 from sketch\src\Marlin.h:24,

                 from sketch\src\Marlin.cpp:31:

D:\OneDrive\Documents\Arduino\libraries\TMC2208Stepper\src/TMC2208Stepper_REGDEFS.h:6:24: error: expected unqualified-id before numeric constant

 #define TMC2208_SYNC   0x05

                        ^

D:\OneDrive\Documents\Arduino\libraries\TMCStepper\src/TMCStepper.h:899:29: note: in expansion of macro 'TMC2208_SYNC'

   static constexpr uint8_t  TMC2208_SYNC = 0x05,

                             ^

Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

**Expected behaviour:** [What you expect to happen]
Should compile
**Actual behaviour:** [What actually happens]
Fails
#### Additional Information

No issues compiling 1.1.9 bugfix

teemuatlut, 2019-04-26:
Second line in your `Configuration_adv.h`:
```cpp
#include <TMC2208Stepper_REGDEFS.h>
```
Remove it.

jimmyeao, 2019-04-26:
Awesome, thank you

closed 2019-04-26 by jimmyeao