MarlinFirmware/Marlin #13827: [Question] How to properly configure print bed location

UticaTechClub, 2019-04-26:
The view from the top of the printer:
 
I have `COREXY `configuration with five tools mounted along X axis on the top. 
X and Y limit switches are mounted at the bottom and left. 

The motion range is 296 for X and 321 for Y. 
```
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 296
#define Y_MAX_POS 321
#define Z_MAX_POS 180
```
The bed size is 200 for X and 194 for Y.
```
#define X_BED_SIZE 200
#define Y_BED_SIZE 194
```
Print bed and all tools are all located within motion range.

Question:
What is the best way to configure the print bed location so that homing (`G28`) and bed leveling (`G29`) could be executed correctly (**over the print bed area only**) while tools can be picked up and dropped off outside of the print bed area?

I am using the latest `bugfix-2.0.x` branch, but I don't think it matters for this question.

For homing Z (`G28`), it probes somewhere on the print bed because it moves to the center of the motion range.
I got it to work but only using `AUTO_BED_LEVELING_3POINT` (for `G28`) and specifying the location of every probing point. 
```
#if EITHER(AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_UBL)
  #define PROBE_PT_1_X 76
  #define PROBE_PT_1_Y 34
  #define PROBE_PT_2_X 190
  #define PROBE_PT_2_Y 34
  #define PROBE_PT_3_X 138
  #define PROBE_PT_3_Y 184
#endif
```
The desire is to use **`MESH_BED_LEVELING`**.

Thank you so much for your help.

ManuelMcLure, 2019-04-26:
The front left corner (or back left corner) of the print surface should be defined to be 0, 0. This means that ``X_MIN_POS`` and/or ``Y_MIN_POS`` may end up negative. Marlin handles this fine. For example, on my AM8 I have ``X_MIN_POS=-36`` and ``Y_MIN_POS=-6``. 
Once you have that set, ``Z_SAFE_HOMING`` should work fine because it will move to the center of the bed (``(X_BED_SIZE / 2), (Y_BED_SIZE / 2)``).

ManuelMcLure, 2019-04-26:
You can provide negative values for ``X`` and ``Y`` in your various ``G0/G1`` calls to go pick up any tools that are in front of/to the left of the bed. For example, my startup gcode includes a ``G0 X-10, Y40`` to park my nozzle off the bed surface while the hotend comes to temperature so I can avoid any ooze on the bed.

UticaTechClub, 2019-04-26:
> You can provide negative values for `X` and `Y` in your various `G0/G1` calls to go pick up any tools that are in front of/to the left of the bed. For example, my startup gcode includes a `G0 X-10, [...]

@ManuelMcLure Thank you so much.
In your second post, you have also answered my pending question: Will it be OK to travel to negative location to pickup my tools?

So if my bed is at `0, 0` and limit switches are at say `-50, -40` then the config will change to 
```
#define X_MIN_POS -50
#define Y_MIN_POS -40
#define X_MAX_POS 246
#define Y_MAX_POS 281
```

And my tool X positions will change from:
`#define SWITCHING_TOOLHEAD_X_POS          { 4.5, 76, 152 } `
to:
`#define SWITCHING_TOOLHEAD_X_POS          { -45.5, 26, 102 } `

I will try it.
Thank you again.

UticaTechClub, 2019-04-29:
I worked perfectly.
Thank you!

closed 2019-04-29 by UticaTechClub