MarlinFirmware/Marlin #13828: [BUG] BigtreeTech SKR 1.3 3DTouch not responding

Strahler85, 2019-04-26:
### Description

I have a Ender 3 and upgraded it with the BigtreeTeck SKR 1.3 with 4 TMC2206 controlled over UARD. Additional I added a 3DTouch sensor from Triangle. I defined the BLTOUCH in the configuration and Servo Pin is P2_00. At startup the probe deploy and stow. But than no reaction to gcode or LCD selection.
M43 comes up with this result

m43
SENDING:M43
PIN: 0.00        E1_DIR_PIN                             Output = 1
PIN: 0.01        E1_STEP_PIN                            Input  = 1
PIN: 0.02 (A 7)  <unused/unknown>   Analog in =     0   Input  = 1
PIN: 0.03 (A 6)  <unused/unknown>   Analog in =     0   Input  = 1
PIN: 0.04        <unused/unknown>                       Input  = 1
PIN: 0.05        <unused/unknown>                       Input  = 1
PIN: 0.06        SDSS                                   protected
.                SS_PIN                                 protected
PIN: 0.07        DOGLCD_SCK                             protected
.                SCK_PIN                                protected
PIN: 0.08        MISO_PIN                               protected
PIN: 0.09        DOGLCD_MOSI                            protected
.                MOSI_PIN                               protected
PIN: 0.10        E1_ENABLE_PIN                          Output = 1
PIN: 0.11        E0_DIR_PIN                             protected
PIN: 0.15        <unused/unknown>                       Input  = 1
PIN: 0.16        <unused/unknown>                       Input  = 1
PIN: 0.17        <unused/unknown>                       Input  = 1
PIN: 0.18        <unused/unknown>                       Input  = 1
PIN: 0.19        Y_STEP_PIN                             protected
PIN: 0.20        Y_DIR_PIN                              protected
PIN: 0.21        Z_ENABLE_PIN                           protected
PIN: 0.22        Z_STEP_PIN                             protected
PIN: 0.23 (A 0)  <unused/unknown>   Analog in =   983   Input  = 0
PIN: 0.24 (A 1)  <unused/unknown>   Analog in =   982   Input  = 0
PIN: 0.25 (A 2)  <unused/unknown>   Analog in =     0   Input  = 1
PIN: 0.26 (A 3)  <unused/unknown>   Analog in =     0   Input  = 1
PIN: 0.27        <unused/unknown>                       Input  = 0
PIN: 0.28        BTN_ENC                                Input  = 1
PIN: 0.29        <unused/unknown>                       Input  = 1
PIN: 0.30        <unused/unknown>                       Input  = 0
PIN: 1.00        <unused/unknown>                       Input  = 1
PIN: 1.01        E1_CS_PIN                              Input  = 1
.                Z2_SERIAL_RX_PIN                       Input  = 1
.                E1_SERIAL_RX_PIN                       Input  = 1
PIN: 1.04        Z2_SERIAL_TX_PIN                       Input  = 1
.                E1_SERIAL_TX_PIN                       Input  = 1
PIN: 1.08        E0_CS_PIN                              Input  = 1
.                E0_SERIAL_RX_PIN                       Input  = 1
PIN: 1.09        E0_SERIAL_TX_PIN                       Output = 1
PIN: 1.10        Z_CS_PIN                               Input  = 1
.                Z_SERIAL_RX_PIN                        Input  = 1
PIN: 1.14        Z_SERIAL_TX_PIN                        Output = 1
PIN: 1.15        Y_CS_PIN                               Input  = 1
.                Y_SERIAL_RX_PIN                        Input  = 1
PIN: 1.16        Y_SERIAL_TX_PIN                        Output = 1
PIN: 1.17        X_CS_PIN                               Input  = 1
.                X_SERIAL_RX_PIN                        Input  = 1
PIN: 1.18        BTN_EN1                                Input  = 1
PIN: 1.19        <unused/unknown>                       Output = 1
PIN: 1.20        BTN_EN2                                Input  = 1
PIN: 1.21        LCD_PINS_D4                            Output = 0
PIN: 1.22        LCD_PINS_RS                            Output = 0
PIN: 1.23        LCD_PINS_ENABLE                        Output = 0
PIN: 1.24        <unused/unknown>                       Input  = 1
PIN: 1.25        Z_MIN_PIN                              protected
PIN: 1.26        <unused/unknown>                       Input  = 1
PIN: 1.27        Y_MIN_PIN                              protected
PIN: 1.28        FIL_RUNOUT_PIN                         Input  = 1
PIN: 1.29        X_MIN_PIN                              protected
PIN: 1.30 (A 4)  BEEPER_PIN                             Output = 0
PIN: 1.31 (A 5)  <unused/unknown>   Analog in =     0   Input  = 1
PIN: 2.00        SERVO0_PIN                             Input  = 1
PIN: 2.01        X_ENABLE_PIN                           protected
PIN: 2.02        X_STEP_PIN                             protected
PIN: 2.03        FAN_PIN                                protected
PIN: 2.04        E0_AUTO_FAN_PIN                        protected
.                FAN1_PIN                               protected
PIN: 2.05        HEATER_BED_PIN                         protected
PIN: 2.06        X_DIR_PIN                              protected
PIN: 2.07        HEATER_0_PIN                           protected
PIN: 2.08        Y_ENABLE_PIN                           protected
PIN: 2.09        <unused/unknown>                       Input  = 0
PIN: 2.10        <unused/unknown>                       Input  = 1
PIN: 2.11        Z_DIR_PIN                              protected
PIN: 2.12        E0_ENABLE_PIN                          protected
PIN: 2.13        E0_STEP_PIN                            protected
PIN: 3.25        <unused/unknown>                       Input  = 1
PIN: 3.26        <unused/unknown>                       Input  = 1
PIN: 4.28        <unused/unknown>                       Output = 1
PIN: 4.29        X_SERIAL_TX_PIN                        Output = 1

I don't know why the SERVO0_PIN is declared as input. Also I measured a voltage of 2.2V and this is according the LPC1768 forum the voltage at a input pin.

M 43 S output is:
SENDING:M43 S
Servo probe test
.  using index:  0
.  deploy angle: 10
.  stow angle:   90
. probe uses Z_MIN pin: 57
. uses Z_MIN_ENDSTOP_INVERTING (ignores Z_MIN_PROBE_ENDSTOP_INVERTING)
. Z_MIN_ENDSTOP_INVERTING: false
. deploy & stow 4 times
please trigger probe
trigger not detected

The probe does not deploy or stow with this command.

After that I connected the probe to a Arduino nano with a very basic BLTOUCH sketch and there the probe works as aspected.

With the command M42 I P200 S255 I can measure 3.3V and with M42 I P200 S0 I measured 0V on the Servo Pin (also the the pin is than OUTPUT instead of INPUT). I think hardware should be fine.

Attachment Configuration File:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3121799/Marlin.zip)

jjansen85, 2019-04-26:
Just to be sure, how did you connect your BL-touch? 
I had to be carefull, because the SKR v1.3 has a Servos connection which is 5V, GND, P2_00 (from left to right). On the standard BL-touch cable the GND and 5V are 'twisted' compared to the SKR board; so GND, 5V, Signal.

For me it is working fine with the bugfix 2.0 version of a month ago.

Strahler85, 2019-04-26:
Hi yes I'm 100% sure that i connect the probe correctly

jjansen85, 2019-04-26:
I checked my situation (3Dtouch (clone)) with the latest version of bugfix 2.0. No problems found. 

In my config I have the following set to TRUE
Z_MIN_PROBE_ENDSTOP_INVERTING
Z_MIN_ENDSTOP_INVERTING

gloomyandy, 2019-04-26:
Are you sure that your probe is designed (and configured) to operate with 3.3V? Some probes need to be modified to have them work correctly.

SISLANGER, 2019-04-27:
@Strahler85 
You might need to rewire it going into the board that's how I got mine working on the SKR v1.3

Wiring is 
Orange - far right 
Brown - middle
Red - left nearest the mosfets for heaters etc

Strahler85, 2019-04-27:
@SISLANGER It's already wired like you mentiont.

SISLANGER, 2019-04-27:
I will copy my settings to secs

SISLANGER, 2019-04-27:
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_PIN P1_25 // Pin 32 is the RAMPS default I think this is your problem
 #define ENDSTOPPULLUP_ZMIN_PROBE //MNV 8 BLTOUCH
#define Z_MIN_PROBE_ENDSTOP_INVERTING true // set to true to invert the logic of the probe.
#define BLTOUCH_DELAY 100

SISLANGER, 2019-04-27:
@Strahler85 not in order but search looking again at your code I think the pin is set wrong try P1_25

Strahler85, 2019-04-27:
I used the Z_max_probe_pin for trigger and this is P1_25. But the trigger pin has nothing to do that the 3d_touch cannot be controlled with the board. Because the sensor works with an arduino nano.

But I will change it to your setting and try it out.

Strahler85, 2019-04-27:
No change, I cannot deloy or stow the pin of the sensor. Trigger the signal works, but without control of the  pin the sensor is useless.

SISLANGER, 2019-04-27:
strange mine was as mentioned the wiring going into the servo port try probing the port with multimeter

Strahler85, 2019-04-27:
multimeter show between GND and S pin 3.1V

gloomyandy, 2019-04-27:
Just because your probe works with an Arduino does not mean that it will work with the SKR board, the Arduino will be operating at 5V the SKR at 3.3V. Also some of the SKR pins will have additional hardware attached to them (so the Max/Min pins will have pull up resistors and a noise reducing capacitor), which may be causing problems.

Strahler85, 2019-04-27:
@gloomyandy that's correct I only wanted to check if the probe has an defect. I don't know what to do now. Maybe the SKR has an fault or my sensor needs more voltage on the servo pin.

zillarob, 2019-05-01:
I run a TL clone on one of my v1.3 and its fine.
My fw isnt the very latest though, prob 3wks or so old.

SISLANGER, 2019-05-01:
I wonder if it’s a bug in Marlin 
I will try and get previous Version

SISLANGER, 2019-05-01:
I have also ordered replacement screen just in case. If both older version of Marlin and new screen don’t fix then faulty board ??

zillarob, 2019-05-01:
Have mks mini and btt tft35 on mine. 
I did have to paste the extra code in for the mini to work on that one.
It works so havent messed with, but will update when I get a chance.

SISLANGER, 2019-05-01:
I also have a problem on the TFT when home an axis stops short. An ideas on that one?

Strahler85, 2019-05-15:
I think it is a hardware issue of the clone

closed 2019-05-15 by Strahler85