MarlinFirmware/Marlin #13830: Manual Mesh Bed Leveling is not being used

Revolt64, 2019-04-26:
- Marlin 1.1.9
- Printer: Ender 3
- Slicer: Cura
- Using automatic Restore Mesh Leveling define. Have also tried putting a M420 S1 in my start gcode to no avail.

I've enabled Manual Mesh bed leveling via:

`
#define MESH_BED_LEVELING
#define RESTORE_LEVELING_AFTER_G28
#define GRID_MAX_POINTS_X 5
#define LCD_BED_LEVELING
`

And define the points via the printers LCD. 

Response from M420 V1:

`Recv: Z offset: 0.00000
Recv: Measured points:
Recv:         0        1        2        3        4
Recv:  0 +0.09000 +0.11000 +0.20000 +0.25000 +0.30500
Recv:  1 +0.02500 +0.03500 +0.09000 +0.17500 +0.22000
Recv:  2 +0.01000 +0.04000 +0.06500 +0.09500 +0.18000
Recv:  3 +0.07500 +0.07500 +0.03500 +0.02000 +0.03500
Recv:  4 +0.19000 +0.09500 +0.06500 +0.09500 +0.08000
Recv: 
Recv: echo:Bed Leveling On
Recv: echo:Fade Height 1.00`

When the print starts, the nozzle is roughly 0.2mm away from the bed. Removing all custom start gcode doesn't resolve the issue either.

I've been fighting this problem, trying to find a solution now for over a week. I didn't want to ask here, but you're my last hope.

What am I doing wrong?
Thank you

reloxx13, 2019-04-26:
if u move down Z after homeing with the lcd to 0, is it on the bed?

Revolt64, 2019-04-26:
> if u move down Z after homeing with the lcd to 0, is it on the bed?

It's slightly below it. It homes z to -0.1mm.

I just used M42 I0 J0 (and so on) to move the extruder to to saved mesh points, to verify that they were valid, and on each one it moved to exactly where I positioned it earlier, when defining the mesh points.

As soon as a print starts though, it's like they never existed.

Here's the first section of the gcode cura is outputting for a raspberry pi case (so you can see more detail):

`;FLAVOR:Marlin
;TIME:8125
;Filament used: 5.80455m
;Layer height: 0.2
;Generated with Cura_SteamEngine 4.0.0
M140 S60
M105
M190 S60
M104 S210
M105
M109 S210
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
G28 ; Home all axes
M402 S1 ; restore bed level
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode
G92 E0
G1 F1500 E-6
;LAYER_COUNT:52
;LAYER:0
M107
G0 F3600 X76.335 Y68.345 Z0.2
;TYPE:SKIRT
G1 F1500 E0
G1 X76.691 Y67.95 E0.02122
G1 X77.175 Y67.454 E0.04888
G1 X77.713 Y66.959 E0.07806
G1 X78.29 Y66.535 E0.10664
G1 X79.172 Y65.956 E0.14875
G1 X79.843 Y65.569 E0.17967`

reloxx13, 2019-04-26:
`M402 S1 ; restore bed level`

its m420 s1

u wrote m402 s1 in ur startcode.

http://marlinfw.org/docs/gcode/M402.html

http://marlinfw.org/docs/gcode/M420.html

Revolt64, 2019-04-26:
> `M402 S1 ; restore bed level`
> 
> its m420 s1
> 
> u wrote m402 s1 in ur startcode.
> 
> > Stow the bed probe. The Z axis may raise up to make room for the probe to stow.
> 
> http://marlinfw.org/docs/gcode/M402.html
> 
> http://marlinfw.org/docs/gcode/M420.html

Stupid typo. I must have dyslexia. 
Updated the gist with the fixed gcode. It still exhibits the same issue. I've been trying this for days now (not with the typo, that was introduced whilst messing with the start gcode today it seems).

Any other ideas? I'm watching the Z position on the LCD screen whilst it's moving along the first layer, and there's no micro-stepping being performed (the z stepper isn't making micro adjustments either).

Revolt64, 2019-04-27:
Figured it out, at long last.

Changing the command to specify which mesh to use finally fixed the issue!

M420 L0 S1

closed 2019-04-27 by Revolt64