MarlinFirmware/Marlin #13832: [BUG] LCD goes Blank when changing filament

robbycandra, 2019-04-27:
LCD goes Blank for few seconds when changing the filament via LCD

### Steps to Reproduce

1. Turn On Printer
2. Menu -> Change FIlement
3. -> Pre Heat PLA
4. Screen show Nozzle Heating
5. At this Point... sometimes screen goes blank for few second.
6. Screen show Unload Filament

**Expected behavior:** [What you expect to happen]
No blank screen at point 5.

**Actual behavior:** [What actually happens]
Sometimes we have few seconds blank screen at point 5.

#### Additional Information

Sometimes when I print file from SD Card that have G28 at starting GCode Script. The Screen also goes blank after G28 and before start printing on the bed. I use Max Home Dir for Z axis. The blank screen happened when Z axis move from Max pos to first layer.

shitcreek, 2019-04-30:
What version of Marlin are you using?

robbycandra, 2019-04-30:
Latest Bugfix 2.0

robbycandra, 2019-05-07:
LCD Blank at 
pause.cpp
-> unload_filament(

```
  // Wait for filament to cool
  safe_delay(FILAMENT_UNLOAD_DELAY);

```

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@robbycandra please attach your 2 config files as the issue template says to do

and is the issue still present?

robbycandra, 2019-09-24:
It exist in many place, Last time I check, BL touch menu also makes LCD problem.

stefi01, 2019-10-03:
im using bugfix and use a smart discount lcd which works fine untill i start printing, then it goes blank untill the print finishes

robbycandra, 2019-10-05:
This problem has not fixed yet. I guess, it is because some LCD operation takes long time to finish but not call idle().

boelle, 2019-10-12:
@robbycandra please attach your 2 config files as the issue template says to do

boelle, 2019-10-22:
as there is still no config and there has been no activity i will close this one

also if you have a good practice fix you can submit a PR and if its not perfect Scott will tell you what is wrong or if its a small thing he will correct it

closed 2019-10-22 by boelle