MarlinFirmware/Marlin #13833: [FR] Option to bypass thermistor lookup tables with sensor specs

doggyfan, 2019-04-27:
Nice to have the option to specify thermistor parameters rather than have to use/create lookup tables.

So we could specify ..

**Beta value**
**Temp offset**
**Pull-up resistor value**

Plus any other required parameter that might be needed.

SJ-Innovation, 2019-04-27:
With modern STM32 processors this could be a valid approach due to the larger RAM and flash available, we could generate the table at runtime. For AVR and smaller controllers though a table saves on RAM usage as it is saved in flash by the compiler.

doggyfan, 2019-04-27:
No no, you just use the formula directly, no table needed, no ram for a table needed.

**#define SERIES_RESISTOR                 4700        // pull-up resistor
#define NOMINAL_TEMPERATURE   25
#define NOMINAL_RESISTANCE       100000   // 100k at 25C (NOMINAL_TEMPERATURE)
#define BETA                                    3950

int ADCvalue = averaged_adc_sample;
if (ADCvalue == 0)
   ADCvalue = 1;
float res = SERIES_RESISTOR / ((1023.0f / ADCvalue) - 1);

float steinhart = res / NOMINAL_RESISTANCE;   // (R/Ro)
steinhart = logf(steinhart);                                 // ln(R/Ro)
steinhart /= BETA;                                            // 1/B * ln(R/Ro)
steinhart += 1.0f / (NOMINAL_TEMPERATURE + 273.15f);  // + (1/To)
steinhart = 1.0f / steinhart;                                                  // Invert
float deg_c = steinhart - 273.15f;                                        // convert to C**

doggyfan, 2019-04-27:
For the PID temperature loop to maintain correct temperature, you don't need to operate in direct deg C units, you can optimize that by using raw values, so no need to go through all the calculation.

You only really need to do the full formula when you want too tell the user the temp in degrees C.

You'd get rid of all that CPU flash usage due to all the look-up tables, that's assuming you did away with the tables all together (no reason not too). Really not sure why you have the tables ??

psavva, 2019-04-27:
Why don't you do a pull request and see where it gets?

doggyfan, 2019-04-28:
@psavva The code needs additions for that, I need to go more indepth into the Marlin code for that, which I am doing bit by bit, so will take time.

But yes, I will once I'm able.

Idorobots, 2019-04-28:
Sounds like a nice new temp sensor type with some Configuration_adv.h setting for it.

AnHardt, 2019-04-29:
Please don't confuse the Steinhart-Hart equation with the calculation where beta is used.

PID temperature control will not work with a exponentially distorted temperature (raw) signal.

doggyfan, 2019-05-01:
Yes you're right of cause, the Steinhart-Hart equation is a little more involved (2 part).

Though the simpler log method is totally usable here.

doggyfan, 2019-05-02:
arrgghh git .. long time in life but not using git.

I use TortoiseGit in windows 7 and I'm wondering how best to go about doing a pull request for Marlin ?

I've updated my Marlin code files on my PC in the Marlin git directory I have, so just need to do what's desired by the Marlin team .. whatever that might be ?

gloomyandy, 2019-05-02:
You should create a new branch based on the very latest 2.0.x code, then make only the changes needed for the pull request to that branch (so do not include any config files that are specific to your machine). You then need to push those changes to github and after doing that create a pull request. 

See here: http://marlinfw.org/docs/development/getting_started_pull_requests.html for details.

doggyfan, 2019-05-02:
Thank you gloomyandy.

Tried to do a **push** but it says permission denied to me (after entering my username and password in TortoiseGit) ?

p3p, 2019-05-02:
You missed 2 very important steps and are trying to push changes directly into Marlin, so .. probably a good thing you get permission denied ;)

From the guide start with `Set up GitHub, Fork, and Clone` and then continue at `Make a feature branch`

doggyfan, 2019-05-02:
Sorry about that p3p. Think I've got it now, hopefully.

[wipes forehead}

doggyfan, 2019-05-02:
ssshhh..eee, the pull request doesn't look like it should look like ?

p3p, 2019-05-02:
nearly there, you targeted the wrong branch (1.1.x) you need to target bugfix-2.0.x, you can change the target with the edit button at the top.

doggyfan, 2019-05-02:
I blame the tablets ;)

you're a life saver p3p.

closed 2019-05-06 by doggyfan