MarlinFirmware/Marlin #13834: [FR] Buffer SLOWDOWN user threshold

reloxx13, 2019-04-27:
### Description

Currently the SLOWDOWN threshold is 50% of buffer. So it will slowdown if the buffer is half empty.

Now if i set a higher buffer, i have more commands which are still waiting in the buffer and i dont need to slowdown at 50%, rather at 25%.

Example1: Default
Buffer max 50 commands
Half 25 commands
=> Slowdown

Example2: Higher Buffer
Buffer max 100 commands
Half 50 commands, still enough
Now at quater 25 commands left
=> Slowdown

### Conclusion

It would be nice if the user can define the slowdown threshold himself, but set default at 50%.
The boards are also getting better and will have a higher buffer.

Config example:
```
// If defined the movements slow down when the look ahead buffer is under given percentage
define SLOWDOWN 25 //in percent (default 50)
```

vadsh, 2020-01-26:
I believe that would be a great feature. 50% with large buffer is not good. Custom threshold seems to be easy to implement and useful.

sjasonsmith, 2020-09-11:
This was added about 6 months ago:
https://github.com/MarlinFirmware/Marlin/pull/17171

closed 2020-09-11 by sjasonsmith