MarlinFirmware/Marlin #13835: [BUG] Can't browse SD card after enabling BLTouch

Zol-Tank, 2019-04-27:
### Description

When BLTouch is enabled and my Win10 computer is connected to the SKR v1.3 via USB, I am unable to see or browse the SD card, which makes it impossible to update the firmware without removing the SD from the SKR. Is BLTouch is disabled, then I can see and browse the SD card without issue.

### Steps to Reproduce

1. In Configuration.h uncomment #define BLTOUCH
2. Compile, upload, and reset the SKR.

**Expected behavior:** Enable BLTouch and still see and browse the SD card while connected to the controller via USB with my computer.

**Actual behavior:** Cannot see or browse the SD card while connected to the controller via USB with my computer.

#### Additional Information

I'm using an SKR v1.3 board, Windows 10, and Atom. I've downloaded the latest Marlin 2.0 bugfix as of yesterday. I've tried this on two separate computer using two separate SD cards and the behavior remained consistent. 

If I connect my computer to the SKR while BLTouch is enabled, my computer not only cannot see the SD card on the controller, but it also can't see an SD card plugged into its built in SD card slot.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3124279/Marlin.zip)

gloomyandy, 2019-04-27:
Have you made any changes to any other files (like for instance the pins file)?
Does your BLTouch work when you have it enabled?
What happens if you define the BLTouch in your configuration but unplug the BLTouch from the board can you see the SD card then?
How are you powering the board? Do you have it powered from the main power supply. What setting do you have the power jumper on (Int or USB), if USB try moving it to Int and make sure you use an external power supply.

I've seen a number of reports of people using a BLTouch along with the SD card working fine on this board, so something odd is going on.

dot-bob, 2019-04-27:
I wonder if this is related to the issue I was having with the BLTouch and SD card.  See #13804

The pins file wasn't defining the SD card pins correctly so it was defaulting to P1_23 which is used for servos on many of the LPC176X based boards.

Perhaps the defaults should be removed from the HAL thus requiring the pins files to define them correctly.

gloomyandy, 2019-04-27:
Servo pin on the SKR V1.3 is P2_00 so I don't think that should be conflicting with any of the SD card pins. As far as I can see there does not seem to be any sort of conflict with any of the USB pins either. My money is on some sort of voltage drop issue, but who knows!

Zol-Tank, 2019-04-27:
After working with it some more, the issue is happening when I'm using the Ender-3 example config files. The only think I'm changing in the Ender-3 configuration.h file is the board and enabling BLTouch. If I use the stock config files then the issue does not happen.

gloomyandy, 2019-04-27:
That is strange. I notice that you do not have the USB serial port enabled so you may want to try that...
#define SERIAL_PORT -1

Zol-Tank, 2019-04-28:
> That is strange. I notice that you do not have the USB serial port enabled so you may want to try that...
> #define SERIAL_PORT -1

I've tried that and it didn't seem to make any difference.

gloomyandy, 2019-04-28:
In which case you will need to work through the two files and try and identify which difference is causing the problem. But anyway you probably want to have #define SERIAL_PORT -1 so that you can use USB access to your printer.

Zol-Tank, 2019-04-28:
I'm in the process of doing that right now. It's just taking a while because I'm testing after every change.

I plan to change SERIAL_PORT to -1 once this is figured out, since I want to change as little as possible until then. 

Also, the SKR isn't in a printer or connected to any hardware yet.

Zol-Tank, 2019-04-28:
After further investigation, the issue occurs when both BLTouch and CR10_DISPLAY are enabled at the same time

gloomyandy, 2019-04-28:
As above do you have an external power supply or are you trying to run all of this from USB? It may be that you are drawing too much current from the USB supply.

Also just to be clear if you just enable the CR10_STOCKDISPLAY are things working OK? When you used the "stock config files" what display did you have defined if any?

Zol-Tank, 2019-04-28:
That's the other way around, if I enable CR10_STOCKDISPLAY while BLTouch is enabled I lose the SD card access. If I disable CR10_STOCKDISPLAY then everything is fine.

I'm running on USB but I don't have anything hooked up, not even the display. I'm still in the process of learning how to work with Marlin 2.0 and trying to get the initial config figured out. It's just a bare board right now without even stepper drivers installed.

gloomyandy, 2019-04-28:
So are you saying that you see these problems without either the BLTouch or the display connected to the board? Just enabling the options triggers the problem? How are you interacting with the board? Are you using any sort of PC software to talk to it?

 I really think you should attach the display and get that working. There is a very good chance that there is a message on the display telling you that there is some sort of error and that the firmware has been halted. You can't really just define things and not have the hardware attached.

Zol-Tank, 2019-04-28:
Yes, this happens without either the BLTouch or display connected to the board.

I don't currently have anything on my computer to talk to the SKR. When I've needed to do that so far I've connected it to a Pi running octoprint. It's not ideal, but I wasn't expecting a longer term need.

I pulled the display off one of my printers and hooked it up to the SKR and it remained dark. That said, the board was running on USB power at the time. Unfortunately, right now I don't really have a way to connect it to an external power supply without actually installing it in my printer, which I don't really want to do given this issue.

gloomyandy, 2019-04-28:
I am pretty sure that the firmware is detecting a problem and shutting down. When it does this USB access to the SD card will not be available and this is what you are seeing. I don't think this is a bug, but is probably expected. You really need to get a display working and to actually hook up the hardware if you want to test anything. You should be able to get the display working running off USB. You obviously need to make sure that the display type matches and that the cables are connected correctly (you sometimes need to rotate the connectors by 180 degrees).

Zol-Tank, 2019-04-28:
So, I finally got the display working (the fault was mostly my own stupidity). With only CR10_STOCKDISPLAY enabled the display works (though the contrast is off) and I can navigate around. However, once I enable BLTouch, as soon as it power on the display reads "BED THERMAL RUNAWAY - PRINTER HALTED - Please reset". It happens immediately upon startup. 

The BLTouch is not hooked up.

gloomyandy, 2019-04-28:
Well it would have saved a lot of puzzling if we had seen this from the start. But anyway. Make sure you have an up to date version of Marlin and you should probably follow this thread (and any associated ones)... https://github.com/MarlinFirmware/Marlin/issues/13831
But you may find that if you actually have a BLTouch connected, some of these other issues go away (as there will not be what may well be some sort of long timeout happening). What you are doing (running a board configured for non existent devices) is likely to be creating situation that have not really been tested very much, so you may well hit odd error conditions etc. So for instance you may find that the contrast changes when you use a different power supply.

Zol-Tank, 2019-05-07:
Increasing the THERMAL_PROTECTION_GRACE_PERIOD fix this issue for me.

Thank you everyone for the help!

closed 2019-05-07 by Zol-Tank