MarlinFirmware/Marlin #13836: [QUESTION] Marlin 2.0 CNC Mode Configuration

FNeo31, 2019-04-27:
### Description

Marlin 2.0 CNC Mode Configuration

### Explanation

To be successful compiled I need to comment this line:

#if ENABLED(SPINDLE_LASER_PWM)
        //MENU_ITEM_EDIT_CALLBACK(int3, MSG_LASER_POWER, &spindle_laser_power, SPEED_POWER_MIN, SPEED_POWER_MAX, update_spindle_laser_power);
      #endif

And need to have 1 Extruder and one temp sensor (when not make any sense for one CNC)

### Questions

Someone here it's using the Marlin in 3x CNC Machine?

Do you have some way to measure the tools or manage the tool change?

Do you can fix the coordinate system in offline mode (like select G54 and fix)?

It's normal from serial connection the machine send Unknown code?

###Please Help

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3124315/Configs.zip)

shitcreek, 2019-04-30:
To use G53 - G59, you need to uncomment this line in configuration_adv.h
//#define CNC_COORDINATE_SYSTEMS

thinkyhead, 2019-05-02:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-05-02 by thinkyhead

kfmut, 2020-05-26:
@FNeo31 

Hi, have you found your way for using Marlin with mill/router? I would like to try it for sake of testing benefits of backlash compensation, but I've found almost zero information about using Marlin as firmware for cutting machines.

FNeo31, 2020-05-26:
@kfmut yes I'm using Marlin. In my milling machine, I didn't test the backlash yet because didn't need (new ballscrews and the resolution of steppers is less than the ballscrews error)
My machine have 5axis but this feature at the moment is in Alfa tests. You can use my branch if you want.
I used SKR 1.3 has board and to have 5axis need to upgrade for SKR Pro v1.1
About the coordinates system you need to define G53 and after that all other G (54 to 59) will follow the G53 offset.
If you use my branch you have access to drilling cycles (not modal for now and just G81, G82 and G83)

kfmut, 2020-05-27:
@FNeo31 

Great news, thank you! And what about tool changes? Are they somehow handled via firmware or in G-code sender? Looks like you are using USB-flash for loading G-code? I'm currently running my router on UNO board clone with GRBL and bCNC( https://github.com/vlachoudis/bCNC ) as controller, and for the first glance there no G-code senders with Marlin support which is comparable to bCNC. I think most advanced is cncjs( https://cnc.js.org/ ) but looks like it's only has M6 commands for tool changes and that's it.

Also maybe you know is there any caveats for running Marlin on basic 32-bit development boards liker chinese STM32's? Buying specialized 3d-printer board for my testings would be bit odd. Or Arduino MEGA clone would be my trouble free bet?

FNeo31, 2020-05-27:
@kfmut

I don't know what kind of System you have but I will try to share my experience at the moment. Comparing directly with Industrial CNC Machine Controllers that I work everyday.
At first your questions:

**Tool Change** - in this topic exist some aspects that can be workarounded but I think it's very important.
**1-** Tool offsets:
Marlin at the moment doesn't have support for that, but this related directly to other aspects, one of them is the kind of tool change do you use.
In simple mode can be 3 types 

- manually directly on collet (and in this case you don't need the tool change option for nothing)
- manually with pre-monted tools in some kind of supports or cones ( and in this case was awesome to have all tool offsets)
- automatically (designed ATC "automatic tool change" and in this case you need to have a spindle or a system that get you the ATC)

**2-** Tool change code:
This kind of code is just necessary if you have a external ATC system that prepare next tool to fast tool change this  isn't necessary if the spindle change the tool directly from tool shop or a static holders on machine just need the tool command and depending of the system change Marlin for the kind of tool change that you have.
If you have a external ATC tool shop with Independent board you can use the same method for tool change used on MMU2.

**3-** Path compensation:
If you have all automatic system, without the offsets you don't have a possible feature to tool path compensation so this isn't supported by Marlin too.

Next topic
**G-code Sender on Marlin**
This is the kind of thing that you didn't need in my opinion. This was the reason that I change to Marlin. I want a off-line machine to work without carrying the PC with the machine. For this you have GRBL (just my opinion don't be mad).
But, yes it's possible to connect with a outside G code sender. The reality is that since I change the 12864LCD to the Bigtreetech tft35 v3.0 I started to work in g-code Sender mode because the TFT reads the code and send it by serial to the board so technically is a G-code Sender.

But with some gaps.
The communication with the board isn't in real-time when you are running a program the have a bad thing in my opinion, you can't feed hold the machine instantly, and resume. You can pause and resume but when you make a pause the machine will run all the commands until clean the buffer. This is completely normal if it's working in communication system that's why at the moment I'm trying to enable this feature that need to be a button connected directly on the board.

And last topic (if I didn't forget anything)

Yes, I'm using USB Pen Drive to put the programs on the machine.

The TFT gave me this possibility and another possibilitys like some macros and MDI direct commands (like the G-Code Sender but without a PC and without graphic simulation of tool path)

If you have some other doubts or if I forgot something please ask

FNeo31, 2020-05-27:
About the board like I said before I'm using a SKR Pro v1.1 board.
Is 32bits, faster than mega board (stm32F4) see specs

shitcreek, 2020-05-27:
@kfmut You may want to check out this PR for a patch of the latest laser/spindle code: https://github.com/MarlinFirmware/Marlin/pull/17661

kfmut, 2020-05-27:
@FNeo31 

Sorry, my bad. I should have been more specific but many thanks for such elaborate answer.

It's manual change with usual ER-collets but with some automation. bCNC has macro for calibrating current work Z-offset against known position, so that difference in Z-heights is used during tool changes to shift WCS in machine coordinates for each new tool. It's works something like this:
- operator finds work Z-origin with some probe tool;
- bCNC automatically tests probe tool with G38 against known position and saves difference in z-heights;
- when M6 met in G-code file bCNC stops g-code execution and substitutes it with little macro.
- operator is told to switch tools and new tool is tested by bCNC with G38 against known position;
- new tool overhang is automatically compenstaed by shifting current WCS with help of G10 L2;
- machining is resumed.
Such behavior is really handy.

Did I get right that pause command is not prioritized at all? It's quite useful when router is not particularly happy with cutting parameters and they are not that bad to hit emergency stop and possibly loose position. And what about emergency stop itself? Could it be directly connected to physical button with current Marlin firmware?
 
And yes I would like to know about soft limits(I think in Marlin they are called software endstops https://marlinfw.org/docs/gcode/M211.html ). Is it true that if move is exceeding defined physical limits it's just clipped without raising some kind of alarm? It could be quite dangerous if it's true.

Thank you for donating your time and knowledge!

kfmut, 2020-05-27:
@shitcreek 

Sorry it's bit early for me to go in such details I'm at early research stage right now but thanks :-)