MarlinFirmware/Marlin #13842: please, clean up double argument

Ludy87, 2019-04-28:
@Bob-the-Kuhn please, clean up all

```Configuration_adv.h```

```c++
  // Frivolous Game Options
  //#define MARLIN_BRICKOUT
  //#define MARLIN_INVADERS
  //#define MARLIN_SNAKE
```

https://github.com/MarlinFirmware/Marlin/blob/e43afb1a58c7272e5eb5c335ae129ee35e16f133/Marlin/Configuration_adv.h#L992-L1002

Roxy-3D, 2019-04-28:
Done...

Ludy87, 2019-04-29:
You forgot the other file 😅

Roxy-3D, 2019-04-29:
Is there another Configuration_adv.h file that has two sets of 'Frivolous Game Options' ?
If so...    I didn't find it.

Ludy87, 2019-04-29:
Yes

https://github.com/MarlinFirmware/Marlin/blob/142c8e2fed810c785fce1afadd0949025f178ad7/config/default/Configuration_adv.h#L995-L1003

https://github.com/MarlinFirmware/Marlin/blob/142c8e2fed810c785fce1afadd0949025f178ad7/config/examples/3DFabXYZ/Migbot/Configuration_adv.h#L995-L1003

and any files from config in the another directories

Roxy-3D, 2019-04-29:
Oh!   So all of the Configuration_adv.h files have this issue?    OK, I'll get that patched in the next few days.

Thank you for the heads up.

Ludy87, 2019-04-29:
I am creating a PR myself, if you want I can fix it right away.

Roxy-3D, 2019-04-29:
OK!   Great...    

Thank You!

Ludy87, 2019-04-29:
i close here, thanks for your work

closed 2019-04-29 by Ludy87