MarlinFirmware/Marlin #13843: [BUG] G53 is not using machine workspace

psychotik2k3, 2019-04-28:
### Description

G53 is not performing command in machine native workspace, movement are still in current workspace
tested with latest bugfixe that fixed G54-59 commands

### Steps to Reproduce

1. G1 Y15
2. G55
3. G92 Y0
4. G1 Y15
5. G53 Y0 or G53 then G1 Y0

**Expected behavior:** the Y should return to position 0 not 15

**Actual behavior:** the Y goes to the 55 space 0 which is 15 in machine workspace

#### Additional Information
you need to enable CNC_COORDINATE_SYSTEMS
https://www.youtube.com/watch?v=qE6nOG_HcOg

shitcreek, 2019-04-29:
Have you seen this? https://github.com/MarlinFirmware/Marlin/pull/13654

psychotik2k3, 2019-04-29:
Yes I applied this fix, this is why g54 and g55 are working. But not g53.

From what I saw in the code, the problem is that when using g54-59 we are
applying offsets. But we are not using zero offsets with g53, the problem
seems to be in the select_coordinate_system function that return false if
we pass active system workspace (sorry don't have computer access now)

I think it should return true and set an offsets of 0,0,0 and in the g53
function we should call it with the active_coordinate_system
And if we chain parser then we set, call parser then set with the current
system.

I hope I'm clear enough.
Sebastien

shitcreek, 2019-04-29:
'G53 Y0' gave me an 'unknown command', so it shouldn't work.
'G53 G1 Y0' and 'G53' then 'G1 Y0' both gave me the desired behavior.

What version of Marlin are you using? Could you post your configs?

psychotik2k3, 2019-04-29:
yes my mistake on the description of the bug i modified the description.
the version i have was the fork from mpcnc302 bugfix 2.0 and i applied the modification of the #13654 
with this modification i'm able to use G54 and G55 it works well.

here is the example of the commands i entered in pronterface (for info 60 correspond to a complete revolution), i put the position as "hour" equivalent position
>>>G1 Y15
SENDING:G1 Y15 (3 o'clock)
>>>G55
SENDING:G55
Select workspace 1
>>>G92 Y0
SENDING:G92 Y0
>>>G1 Y15
SENDING:G1 Y15 (6 o'clock)
>>>G53 G1 Y0
SENDING:G53 G1 Y0 (3 o'clock)

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3129262/config.zip)

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

psychotik2k3, 2019-07-21:
Sorry moved to duet meanwhile.

boelle, 2019-07-21:
no problemo

closed 2019-10-12 by boelle