MarlinFirmware/Marlin #13844: Print doesn't stop through menu command

cinealfa, 2019-04-29:
Print doesn't halt when stop is selected in the menu

1 Start print as normal through OCTOPRINT
2 Chose stop print from menu

**Expected behavior:
Print stops and display says print aborted

**Actual behavior:** 
display says print aborted but printer continues to happily print or make spaghetti as the case may be.
MKS Sbase Marlin 2.0 With BL touch
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3125738/Archive.zip)

ManuelMcLure, 2019-04-29:
When you start the print from OctoPrint are you starting a print from SD using OctoPrint or are you starting a normal print where OctoPrint send all the commands over USB?

If it's the latter, try enabling ``HOST_ACTION_COMMANDS``  in ``Configuration_adv.h``. This should allow Marlin to tell OctoPrint that the print was cancelled so that OctoPrint stops sending printing commands.

cinealfa, 2019-04-30:
@ManuelMcLure I will try that and report when I turn to my printer on Sat.

InsanityAutomation, 2019-05-16:
I see this is still open, if youve confirmed it working as expected please close this issue.

closed 2019-05-18 by cinealfa