MarlinFirmware/Marlin #13846: [BUG] ``HOST_PROMPT_SUPPORT`` + ``EMERGENCY_PARSER`` on LPC1768 fails to compile.

ManuelMcLure, 2019-04-29:
### Description
Compile error when enabling ``HOST_ACTION_COMMANDS``, ``HOST_PROMPT_SUPPORT`` and ``EMERGENCY_PARSER`` on LPC1768 platform
### Steps to Reproduce
Try to compile using the attached configuration file.
**Expected behavior:**
A clean build.
**Actual behavior:** [What actually happens]
The following compile errors:
```
In file included from d:\git\marlinrearm\marlin\src\hal\hal_lpc1768\hal.h:50,
from d:\git\marlinrearm\marlin\src\inc\marlinconfig.h:30,
from d:\git\marlinrearm\marlin\src\feature\host_actions.h:24,
from d:\git\marlinrearm\marlin\src\feature\emergency_parser.h:31,
from Marlin\src\HAL\HAL_LPC1768\usb_serial.cpp:5:
d:\git\marlinrearm\marlin\src\hal\hal_lpc1768\MarlinSerial.h:58:5: error: 'EmergencyParser' does not name a type
EmergencyParser::State emergency_state;
^~~~~~~~~~~~~~~
d:\git\marlinrearm\marlin\src\hal\hal_lpc1768\MarlinSerial.h: In constructor 'MarlinSerial::MarlinSerial(LPC_UART_TypeDef*)':
d:\git\marlinrearm\marlin\src\hal\hal_lpc1768\MarlinSerial.h:47:10: error: class 'MarlinSerial' does not have any field named 'emergency_state'
, emergency_state(EmergencyParser::State::EP_RESET)
^~~~~~~~~~~~~~~
d:\git\marlinrearm\marlin\src\hal\hal_lpc1768\MarlinSerial.h:47:26: error: 'EmergencyParser' has not been declared
, emergency_state(EmergencyParser::State::EP_RESET)
^~~~~~~~~~~~~~~
d:\git\marlinrearm\marlin\src\hal\hal_lpc1768\MarlinSerial.h: In member function 'virtual bool MarlinSerial::recv_callback(char)':
d:\git\marlinrearm\marlin\src\hal\hal_lpc1768\MarlinSerial.h:54:7: error: 'emergency_parser' was not declared in this scope
emergency_parser.update(emergency_state, c);
^~~~~~~~~~~~~~~~
d:\git\marlinrearm\marlin\src\hal\hal_lpc1768\MarlinSerial.h:54:31: error: 'emergency_state' was not declared in this scope
emergency_parser.update(emergency_state, c);
^~~~~~~~~~~~~~~
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\usb_serial.cpp.o] Error 1
In file included from Marlin\src\feature\../inc/../HAL/HAL_LPC1768/HAL.h:50,
from Marlin\src\feature\../inc/MarlinConfig.h:30,
from Marlin\src\feature\host_actions.h:24,
from Marlin\src\feature\emergency_parser.h:31,
from Marlin\src\feature\emergency_parser.cpp:31:
Marlin\src\feature\../inc/../HAL/HAL_LPC1768/MarlinSerial.h:58:5: error: 'EmergencyParser' does not name a type
EmergencyParser::State emergency_state;
^~~~~~~~~~~~~~~
Marlin\src\feature\../inc/../HAL/HAL_LPC1768/MarlinSerial.h: In constructor 'MarlinSerial::MarlinSerial(LPC_UART_TypeDef*)':
Marlin\src\feature\../inc/../HAL/HAL_LPC1768/MarlinSerial.h:47:10: error: class 'MarlinSerial' does not have any field named 'emergency_state'
, emergency_state(EmergencyParser::State::EP_RESET)
^~~~~~~~~~~~~~~
Marlin\src\feature\../inc/../HAL/HAL_LPC1768/MarlinSerial.h:47:26: error: 'EmergencyParser' has not been declared
, emergency_state(EmergencyParser::State::EP_RESET)
^~~~~~~~~~~~~~~
Marlin\src\feature\../inc/../HAL/HAL_LPC1768/MarlinSerial.h: In member function 'virtual bool MarlinSerial::recv_callback(char)':
Marlin\src\feature\../inc/../HAL/HAL_LPC1768/MarlinSerial.h:54:7: error: 'emergency_parser' was not declared in this scope
emergency_parser.update(emergency_state, c);
^~~~~~~~~~~~~~~~
Marlin\src\feature\../inc/../HAL/HAL_LPC1768/MarlinSerial.h:54:31: error: 'emergency_state' was not declared in this scope
emergency_parser.update(emergency_state, c);
^~~~~~~~~~~~~~~
*** [.pioenvs\LPC1768\src\src\feature\emergency_parser.cpp.o] Error 1
```
#### Additional Information
This seems to be because of the ``#include`` of ``MarlinConfiguration.h`` in ``host_actions.h`` - you end up with an include loop because ``MarlinSerial.h`` needs ``EmergencyParser`` but ``emergency_parser.h`` needs (eventually) ``MarlinSerial.h``.
I tried a fix where I changed ``host_actions.h`` to include ``MarlinConfigurationPre.h`` instead but then get an issue with ``NULL`` not being defined.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3125959/Configuration.zip)

ManuelMcLure, 2019-04-29:
It looks like a combination of ``stddef.h`` and ``MarlinConfigurationPre.h`` in ``host_actions.h`` fixes the problem. I've created a PR for this change.

InsanityAutomation, 2019-05-16:
Since this is merged, feel free to close the issue

PR #13847 (2019-05-05): Fix compile problem with HOST_PROMPT_SUPPORT
### Description
Avoid a #Include loop between ``MarlinSerial.h`` and ``emergency_parser.h`` when included from ``host_actions.h``.
### Benefits
Fix a compile error when both ``EMERGENCY_PARSER`` and ``HOST_PROMPT_SUPPORT`` are enabled on LPC1768
### Related Issues
Fixes #13846
closed 2019-05-16 by ManuelMcLure