MarlinFirmware/Marlin #13849: [FR] Power loss recovery for module

Simur, 2019-04-29:
Hi =) 
I need your help ! 
I've a module for power loss recovery (developed by a friend)
I need your help to change the behavior of marlin during the power outage.
There is a bank of capacitors to hold the current long enough to make some movements.
I need that when the pin detects the cut, marlin orders:

-M104 S0; (extrude heater off)
-M140 S0; (heated bed heater off)
-M125; (park extrude)

Currently, when detecting, Marlin kill the printer on the printed object.
Thank you =)

Simur, 2019-05-02:
J'ai une meilleure idée. 
Si détection power_loss : 
-M104 S0
-G27
-Attente jusqu'à plus de batterie.
-Si le courant revient avant la fin de la batterie, reprise automatique de l'impression.

Merci pour votre aide ! =) 
---------------
I have a better idea.
If power_loss detection:
-M104 S0
-G27
-Wait until more battery.
-If the power comes back before the end of the battery, resume printing automatically.

Thanks for your help ! =)

sjasonsmith, 2020-07-05:
It seems this has been partially implemented with the `BACKUP_POWER_SUPPLY` feature. It seems this currently only performs a Z Raise, but it could be modified to perform other actions as well.

sjasonsmith, 2020-09-11:
It is not very clear what this feature request is asking for. Additions have certainly been made to power-loss recovery in the 16 months since this was reported.

Since this is not very clear the the requester as not commented in 16 months, I am going to close this request. More specific bugs or feature requests can be reported based on Marlin 2.0's current capabilities.

closed 2020-09-11 by sjasonsmith