MarlinFirmware/Marlin #13854: [BUG] ReARM & MAX31855 not working

Grogyan, 2019-04-30:
### Description

MAX31855 for ReARM compiles, but then forces an error on the Display of E1 Max temp and ReARM is then bricked

### Steps to Reproduce

Board = Re-ARM
LCD =  RRD Full Graphic
Control Board =  RAMPS1.4
MAX31855 connected to SPI that is separate from the onboard SPI of the Re-ARM
Compile and upload
LCD says E1 Max Temp
ReARM is bricked, and requires external flashing of the Smoothieware firmware
Power is only going to the ReARM via USB, no external power is used, as this is a bench test set up before it replaces the MEGA 2560 currently in use on my printer

**Expected behavior:** 
Expect LCD to report room temperature from the MAX31855

**Actual behavior:** 
LCD reads E1 Max temp, and Re-ARM is bricked and unresponsive until external reflashing of Smoothieware firmware

#### Additional Information
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3129641/Configuration.zip)

Connections to the SPI is verified as correct and the same as my MEGA 2560 + RAMPS 1.4 + RRD Full Graphic display
MAX31855 CS pin is P1_28

Tried with SDSUPPORT enabled, and disabled
Tried with using onboard SD, or, LCD SD

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

Grogyan, 2019-07-21:
Actually
This isn't resolved.

I have not been able to move on because of this bug

Grogyan, 2019-07-21:
Requires review after #14698 is pushed to the main branch

Grogyan, 2019-08-27:
This should now be working, as I have tested this against the SKR 1.3.

Marking this as fixed

closed 2019-08-27 by Grogyan