MarlinFirmware/Marlin #13855: [BUG] TMC CONNECTION ERROR with TMC2660 and Azteeg X5GT (LPC1769)

andyman198, 2019-04-30:
### Description

Not sure if this is an issue with the SPI or the TMC library so close if its out of place. I am having issues with all the TMC2660 drivers in marlin. 

M122 Below -------

```
SENDING:M122
		X	Y	Z	E	E1
Enabled		true	true	true	true	true
Set current	800	800	800	800	800
RMS current	673	673	673	673	673
MAX current	949	949	949	949	949
Run current	23/31	23/31	23/31	23/31	23/31
Hold current
CS actual
PWM scale
vsense		1=.165	1=.165	1=.165	1=.165	1=.165
stealthChop
msteps		16	16	16	16	16
tstep
pwm
threshold
[mm/s]
OT prewarn
OT prewarn has
been triggered
off time		3	3	3	3	3
blank time	24	24	24	24	24
hysteresis
-end		-1	-1	-1	-1	-1
-start		1	1	1	1	1
Stallguard thrs	0	0	0	0	0
DRVSTATUS	X	Y	Z	E	E1
stst
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0x00:00:00:00	 Bad response!
		Z	0x00:00:00:00	 Bad response!
		E	0x00:00:00:00	 Bad response!
		E1	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW
Testing E1 connection... Error: All LOW
```

### Steps to Reproduce

   #define MOTHERBOARD BOARD_AZTEEG_X5_GT
   #define X_DRIVER_TYPE  TMC2660
   #define Y_DRIVER_TYPE  TMC2660
   #define Z_DRIVER_TYPE  TMC2660
   #define E0_DRIVER_TYPE TMC2660
   #define E1_DRIVER_TYPE TMC2660

**Expected behavior:** [What you expect to happen]
   Can see TMC drivers and configure

**Actual behavior:** [What actually happens]
  TMC CONNECTION ERROR on LCD and Bad Response! on M122

Ludy87, 2019-04-30:
Is your power for board connected to PSU or USB?

andyman198, 2019-04-30:
24V meanwell power. whats odd is that the motors do move, just wont respond to tmc commands.

andyman198, 2019-05-01:
After a process of elimination, this only seems to happen when the VIKI2 LCD is plugged in, without that it seems to work just fine. I wonder if that is causing some sort of SPI issue.

Ludy87, 2019-05-01:
Try the control over software SPI for the drivers.

kAdonis, 2019-05-01:
Dont know if it's relevant here, but PR #13712 looks suspiciuos to me
The choice which SD is used is only for LPC1768 not for LPC1769
and the pins file for the Azteeg X5 mini( LPC1769) is changed but not the file for the Azteeg X5 GT

Ludy87, 2019-05-01:
@kAdonis what has that got to do with this problem?

andyman198, 2019-05-01:
Might be. The X5 GT used to use 1768 now is 1769. I will try software SPI. I will also try half speed SPI and see if that helps.

Garyr14, 2019-05-24:
I'm running MKS GEN V1.4 with 2130's (V1.0), Bugfix 2.0x.  Been fine since June, 18.  Last update was in November.  Tried updating to the latest version a few days ago, but all I'm getting is TMC Connection Error.  
Tried comparing configuring.h and configuration_adv.h to working versions, but I can't figure out what's wrong.  I assume this has something to do with pin assignments and/or SPI?  What changed in the past few months that may be causing this?  Thanks for your help.

lukeslaboratory, 2019-05-27:
Hey @andyman198 I'm using a X5 GT,TMC2660, and a viki2, and I do not have issues with using the drivers or M122. 

As @Ludy87 suggested - make sure that SOFTWARE_DRIVER_ENABLE (line 1661 in Confifugartion_adv) is enabled, otherwise my TMC2660's are permanently engaged and I can't talk to them.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@andyman198 is the issue still present? and please attach your 2 config files

closed 2019-10-12 by boelle

MakerThinking, 2019-10-16:
> Lack of Activity
> This issue is being closed due to lack of activity. If you have solved the
> issue, please let us know how you solved it. If you haven't, please tell us
> what else you've tried in the meantime, and possibly this issue will be
> reopened.
Hello, I have the same problem, I have tried all the options but I still have a tmc connection error on the lcd I have skr board v1.3 and tmc2208 silent stepper
[image]
[image]
[image]