MarlinFirmware/Marlin #13858: [bugfix-1.1.x] BLtouch v3.0 won't deploy on G28, M280 work

rlesac, 2019-04-30:
### Description

The probe won't deploy on homing and Marlin runs into an error

### Steps to Reproduce

1. Issue G28

**Expected behavior:** 
The probe should deploy

**Actual behavior:** 
The probe doesn't deploy and Marlin goes into an error state

#### Additional Information
Using Ender 3 1.1.3 board, removed c7 cap.
Set up and flashed Marlin bugfix-1.1x (downloaded 2019-04-29).
LCD BLtouch commands work as do the M280.

rlesac, 2019-04-30:
Debug info:
>>> g28
SENDING:G28
>>> G28
Machine Type: Cartesian
Probe: BLTOUCH
Probe Offset X:44 Y:0 Z:0.00 (Right-Center & Same Z as Nozzle)
set_bltouch_deployed(0)
  current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
Raise Z (before homing) to 5.00
>>> do_blocking_move_to(0.00, 0.00, 5.00)
<<< do_blocking_move_to
  current_position=(0.00, 0.00, 5.00) : sync_plan_position
>>> do_blocking_move_to(-352.50, -352.50, 5.00)
<<< do_blocking_move_to
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, -352.50, [20.00])
  current_position=(0.00, 0.00, 5.00) : sync_plan_position
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

rlesac, 2019-04-30:
Seems to trigger just fine: 
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
>>> M280 P0 S60
SENDING:M280 P0 S60
>>> m119
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: TRIGGERED

FanDjango, 2019-04-30:
Try increasing the BLTOUCH_DELAY to 500ms maybe if it is not deploying from the code, but the M280 works.

rlesac, 2019-04-30:
@FanDjango Tried it, same results.

FanDjango, 2019-04-30:
Ok. Step by step.
It's not a problem of triggering, you say. So it's not really the current CReality - BLTouch V3 triggering problem. You did verify the trigger. Let's assume that for the moment. 

LCD deploy will exeucte a M280, so those two are the same and they both end up in MOVE_SERVO.

The deploy in the code also finally ends up in MOVE_SERVO. So it should work IF it gets executed.

```
 #if HOMING_Z_WITH_PROBE && ENABLED(BLTOUCH)
    // BLTOUCH needs to be deployed every time
    if (axis == Z_AXIS && set_bltouch_deployed(true)) return;
  #endif
```
Perhaps this is not happening? BLTOUCH is enabled I suppose. And HOMING_Z_WITH_PROBE?

```
#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN))
```
So post your configuration files here, let's check Z_MIN_PROBE_USES_ZMIN_ENDSTOP_PIN maybe?

rlesac, 2019-04-30:
Here are the config files.

[Marlin-bugfix-1.1.x-bltouchv3-ender.zip](https://github.com/MarlinFirmware/Marlin/files/3131389/Marlin-bugfix-1.1.x-bltouchv3-ender.zip)

sgsalman, 2019-04-30:
@FanDjango  I have similar issue. Sometimes, while working meshing (for ex:G29) probe stows and deploys very quickly (no touching the bed). I have a BLTOUCH_DELAY 100ms. Is it about this?

FanDjango, 2019-04-30:
@sgsalman Minimum BLTOUCH_DELAY recommended 375. Currently there is discussion that 750 would be better. probe stows and deploys very quickly is because of some probe error causing recovery.

FanDjango, 2019-04-30:
@rlesac sadly, the Z_MIN_PROBE_USES_ZMIN_ENDSTOP_PIN seems to be defined correctly

FanDjango, 2019-04-30:
Where did these 100ms come from originally?

phuxman, 2019-05-01:
I had the same issue on my "BLTouch smart 3" with a delay of 375ms. The problem of wrong mesh measurements was solved by setting BLTOUCH_DELAY  to 500ms. The probe still stows sometimes immediately after deploy but the delay is high enough to redeploy it before lowering.

rlesac, 2019-05-02:
@FanDjango The BLTOUCH_DELAY 100 comes from the TeachingTech video (and other places) https://www.youtube.com/watch?v=sOFxalLOZOI

Tried 375ms and 500ms, same results.

FanDjango, 2019-05-03:
What do I see here?
```
homeaxis(X)
Home 1 Fast:
do_homing_move(X, -352.50, [20.00])
current_position=(0.00, 0.00, 5.00) : sync_plan_position
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!
```
According the that sequence, he is failing on the X home before he even gets anywhere else.

FanDjango, 2019-05-03:
Ok, can you do the G28 in three separate steps and show the output?
G28 X
G28 Y
G28 Z

rlesac, 2019-05-03:
> What do I see here?
> 
> ```
> homeaxis(X)
> Home 1 Fast:
> do_homing_move(X, -352.50, [20.00])
> current_position=(0.00, 0.00, 5.00) : sync_plan_position
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> Error:Printer halted. kill() called!
> [ERROR] Error:Printer halted. kill() called!
> ```
> 
> According that sequence, he is failing directly after the X-home. I am missing some stuff for Y? before he even gets anywhere else. Looking the code again... but it's really weird.

OH! Thank you!
It seems that my X axis stepper has been disconnected *facepalm*. How I managed to miss that, I have no clue.
It's now homing properly, but I'll be doing proper testing to see if there are any other issues.
Thank you!

rlesac, 2019-05-03:
Working as a charm!

>>> m48 p5 x115 y115 v2
SENDING:M48 P5 X115 Y115 V2
M48 Z-Probe Repeatability Test
1 of 5: z: 0.043
2 of 5: z: 0.045
3 of 5: z: 0.045
4 of 5: z: 0.045
5 of 5: z: 0.045
Finished!
Mean: 0.044500 Min: 0.043 Max: 0.045 Range: 0.002
Standard Deviation: 0.001000

closed 2019-05-03 by rlesac

FanDjango, 2019-05-03:
I should have seen it earlier. Happy printing!

rlesac, 2019-05-03:
> I should have seen it earlier. Happy printing!

Thank you for your help!