MarlinFirmware/Marlin #13862: [BUG][bugfix-2.0.x] Stepper jitters when it should stay put

manutenfruits, 2019-04-30:
### Description
The stepper motor jitters when it should stay put. This causes irregular lines in the print. I have a CoreXY printer, and attempted to print something large. When printing the bottom layer, it prints lines at 45 degrees, which in a CoreXY printers means that one stepper moves while the other remains still. Mine jitters a little bit: [demo](https://photos.app.goo.gl/bCf8iwmNhJBCB13u9). You can see how there are quick movements to the left as it progresses from one line to the next, but in between there are small jitters in either direction, about one step.

I have an MKS Sbase set to 1/32 microsteps. This board uses DRV8825 drivers. This is something that started happening recently, and I believe it might be a firmware bug.

### Steps to Reproduce
1. Use cura to slice a print (the larger the easiest to see).
2. Wait until the bottom layer is infilled, or find a way to make your printer do 45 degree movements.
3. Check if the stepper motor jitters or stays put

**Expected behavior:** Stepper motor stays still when moving in 45 degrees

**Actual behavior:** Stepper motor jitters

#### Additional Information
Check out my config on https://github.com/manutenfruits/Marlin/tree/hypercube
Please let me know how can I best isolate the problem. I am planning on downgrading to a 6 month old version to see if it happens then and isolate the problem, but wanted to make sure if this was flagged.

gloomyandy, 2019-04-30:
You should really define the stepper type you have in your configuration file. At the moment they will be operating with some sort of default timing which may or not work with the drivers.

Oh and when you have done that you may want to try setting MINIMUM_STEPPER_PULSE to 4 and see if that helps. The DRV8825s are not happy with the default pulse you get with the LPC176x based boards.

manutenfruits, 2019-05-01:
I didn't know about that option! Unfortunately I set it and [it didn't do much](https://photos.app.goo.gl/2mhBvyMC81W7sZtF9). Is there a better way to isolate the problem?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

InsanityAutomation, 2019-07-23:
Just to add a late note, jittering on microstepping is usually a vref adjustment issue.

manutenfruits, 2019-07-23:
Sorry for not responding earlier, I wanted to test the fix on a larger part to make sure it was fixed. While the motor still kind of jitters, it is much less and it's not as noticeable on the parts.