MarlinFirmware/Marlin #13867: Tool ChangeServo Coding in Marlin

ubaidim, 2019-05-01:
Hi.

My plan is to use a single nozzle and single extruder motor setup and use a servo to clutch/declutch filament.
I have 0 experience with coding and need a lot of help here.
i need to know where and how in marlin do i configure?

what i need is simple.
i need to define servo positions for each of the 2 filaments.
0 ° and 180 °.
have a little sequence

retract T0, switch servo position to T1 and then purge in a purge bucket before getting back to the print.

shitcreek, 2019-05-01:
This feature already exists: http://marlinfw.org/docs/configuration/configuration.html#switching-nozzle. You add the tool changing code in your slicer program.

InsanityAutomation, 2019-05-01:
@shitcreek sounds like hes more describing Switching Extruder - i helped someone else through this recently. A sanity check actually just needs to be removed for it to work as expected with 2 nozzles on the single switching extruder though.

ubaidim, 2019-05-02:
@shitcreek is there anyway i can set the tool change sequence in the firmware? so all i need in the slicer is the tool change m command

shitcreek, 2019-05-02:
Look inside '/src/module/tool_change.cpp'

ubaidim, 2019-05-11:
> Look inside '/src/module/tool_change.cpp'

i can't find this file in IDE

shitcreek, 2019-05-13:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/tool_change.cpp

ubaidim, 2019-05-14:
> https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/tool_change.cpp

hi, apologies for so many questions. i am a noob with this.
how do i use the file?
download and paste in my marlin folder?

shitcreek, 2019-05-14:
It exists inside Marlin Bugfix 2.0. If you've downloaded bugfix 2.0, it should be in there already.

ubaidim, 2019-05-15:
Hi, thank you for the help.
i have the sequence working.
i cannot figure out how to have my sequence set in the firmware at the moment so i am testing with the tool change code in the slicer (slic3r)

My current tools change code is as follows:

T[previous_extruder]
G28 X0  ; home X axis
G91 ; relative positioning
G1 E22.5 F540 ; 9mm/s purge
G1 E-20 F3000 ; 50mm/s retract
G4 P2500 ; 2.5s cooling period
G1 E-140 F3000 ; 50mm/s long retract
T[next_extruder] ; select extruder to wipe
G1 E100 F3000 ; 50mm/s feed
G1 E50 F1500 ; 25mm/s feed
G1 E5 F300 ; 5mm/s feed
G90 ; absolute positioning

even though i have not mentioned anything about raising or lowering z height in the code,
the printer still raises z before tool change and stays there after tool change resulting in printing at the wrong height.
it stays at the wrong height until layer is complete then comes back down when the next layer starts. (i assume due to new z position being fed to the printer at layer change)
I have not set it to go up anywhere in the tool change code.

shitcreek, 2019-05-16:
Your slicer program is the proper place for your tool changing sequence.
Maybe adjust your offsets?
//#define HOTEND_OFFSET_Z {0.0, 0.00}  // (mm) relative Z-offset for each nozzle

ubaidim, 2019-05-17:
Ofsets aren't defined as i am using a single nozzle as well.
but i figured it out. think its a bug.
i do not use ABL and its not defined in my firmware either.
but still going and disabling

//#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
//#define Z_CLEARANCE_BETWEEN_PROBES 10 // Z Clearance between probe points

this solved the problem.

shitcreek, 2019-05-20:
Nice. Please close this issue now.

InsanityAutomation, 2019-05-20:
@shitcreek if that was the only change and it resulted in changing tool change return position something is really odd.... I'll test it on a standard machine with an un-modified branch to be sure.

ubaidim, 2019-05-21:
> @shitcreek if that was the only change and it resulted in changing tool change return position something is really odd.... I'll test it on a standard machine with an un-modified branch to be sure.

it didn't trigger a return position. it stopped the bed from being lowered.
i used to lower to 10mm height and not return.
now its stopped lowering.

ubaidim, 2019-05-21:
one more issue i am having,
my servo angles are 180 degrees apart.
while the T0 angle is always fine no matter what i set it to, 
it has a problem achieving the angle of T1.
are there any limits to the angles i can use?
i am using a tower pro 9g servo.

boelle, 2019-07-21:
will close this one as the feature is allready added

closed 2019-07-21 by boelle