MarlinFirmware/Marlin #13869: TMC2130 no sensorless homing

SSkull995, 2019-05-01:
Hi all! 
First of all, thanks for the support! I have a CreatBot printer, which I modified with a MKS Gen 1.4 and two TMC2130, plus 3 A4988 drivers. I use the TMCs for x and y axis only, and everything seems fine, except the sensorless homing. I tried to increase the sensitivity to -63, the maximum, but when I try to home one of the axis, the motor starts but never stops, I have to switch of the printer manually in order to avoid any damage. 
I have an LCD connected to the motherboard, I used the y_min and y_max pins, which would be used by the two y sensors, in order to keep the lcd and the SD working. 
Here is the debug, hope someone could help, thanks! 

< 16:20:31: X	Y
< 16:20:31: Enabled		false	false
< 16:20:31: Set current	800	800
< 16:20:31: RMS current	1436	1436
< 16:20:31: MAX current	2025	2025
< 16:20:31: Run current	25/31	25/31
< 16:20:31: Hold current	12/31	12/31
< 16:20:31: CS actual		0/31	0/31
< 16:20:31: PWM scale		0	0
< 16:20:31: vsense		0=.325	0=.325
< 16:20:31: stealthChop	false	false
< 16:20:31: msteps		256	256
< 16:20:31: tstep		1048575	1048575
< 16:20:31: pwm
< 16:20:31: threshold		0	0
< 16:20:31: [mm/s]		-	-
< 16:20:31: OT prewarn	false	false
< 16:20:31: OT prewarn has
< 16:20:31: been triggered	false	false
< 16:20:31: off time		0	0
< 16:20:31: blank time	16	16
< 16:20:31: hysteresis
< 16:20:31: -end		-3	-3
< 16:20:31: -start		1	1
< 16:20:31: Stallguard thrs	8	8
< 16:20:31: DRVSTATUS	X	Y
< 16:20:31: stallguard			
< 16:20:31: sg_result		0	0
< 16:20:31: fsactive			
< 16:20:31: stst		X	X
< 16:20:31: olb			
< 16:20:31: ola			
< 16:20:31: s2gb			
< 16:20:31: s2ga			
< 16:20:31: otpw			
< 16:20:31: ot			
< 16:20:31: Driver registers:	X = 0x80:00:00:00
< 16:20:31: Y = 0x80:00:00:00

SSkull995, 2019-05-01:
Ah one moment, I just saw the stealthChop false, I set it true on Marlin. I runt another time M122: 

< 16:22:05: X	Y
< 16:22:05: Enabled		false	false
< 16:22:05: Set current	800	800
< 16:22:05: RMS current	795	795
< 16:22:05: MAX current	1121	1121
< 16:22:05: Run current	25/31	25/31
< 16:22:05: Hold current	12/31	12/31
< 16:22:05: CS actual		12/31	12/31
< 16:22:05: PWM scale		0	0
< 16:22:05: vsense		1=.18	1=.18
< 16:22:05: stealthChop	true	true
< 16:22:05: msteps		16	16
< 16:22:05: tstep		1048575	1048575
< 16:22:05: pwm
< 16:22:05: threshold		0	0
< 16:22:05: [mm/s]		-	-
< 16:22:05: OT prewarn	false	false
< 16:22:05: OT prewarn has
< 16:22:05: been triggered	false	false
< 16:22:05: off time		5	5
< 16:22:05: blank time	24	24
< 16:22:05: hysteresis
< 16:22:05: -end		2	2
< 16:22:05: -start		3	3
< 16:22:05: Stallguard thrs	8	8
< 16:22:05: DRVSTATUS	X	Y
< 16:22:05: stallguard			
< 16:22:05: sg_result		0	0
< 16:22:05: fsactive			
< 16:22:05: stst		X	X
< 16:22:05: olb			
< 16:22:05: ola			
< 16:22:05: s2gb			
< 16:22:05: s2ga			
< 16:22:05: otpw			
< 16:22:05: ot			
< 16:22:05: Driver registers:	X = 0x80:0C:00:00
< 16:22:05: Y = 0x80:0C:00:00

tcm0116, 2019-05-01:
Please share your configuration files. Also, have you connected the diag pin on the TMC2130 to the X and Y min endstop inputs on your board?

teemuatlut, 2019-05-01:
Both your debug outputs show the sensitivity as 8.
There's also something weird going on with the first output, like the RMS current is far off from set current, CS actual is 0, msteps is 256, etc.

Try to first verify the SG threshold is set correctly after a command.

SSkull995, 2019-05-01:
Thank you for the answers! So, I will upload right away the config files. About the debug, I solved the mystery. As soon as I switch the printer on and connect it to Repetier, the output is always the second one, where the communication is ok between motherboard and drivers

SSkull995, 2019-05-01:
Ah I just used the 4 pins, didn't connect the drag pin to my motherboard @tcm0116

SSkull995, 2019-05-02:
Before sending the config files, I noticed something strange (to me, maybe it's normal). If I connect the printer to Repetier BEFORE switching the power on, the drivers don't work, motors are still no matter the GCode commands I send. If I switch the power on and AFTER I connect to Repetier, everything works fine. Is it normal? Thanks!

teemuatlut, 2019-05-02:
The drivers need VMOT power before any SPI / UART commands take hold.

SSkull995, 2019-05-02:
Perfect @teemuatlut, at least that part is working. I modified the pins for the signal, to D42 and D44 because I just wanted to see if the normal end stops would have worked with the same pins I used previously for the TMCs. They work fine, I will disconnect them to try the sensorless later

SSkull995, 2019-05-02:
Ok so this is the output with the power on before: 
< 11:23:40: X	Y
< 11:23:40: Enabled		false	false
< 11:23:40: Set current	800	800
< 11:23:40: RMS current	795	795
< 11:23:40: MAX current	1121	1121
< 11:23:40: Run current	25/31	25/31
< 11:23:40: Hold current	12/31	12/31
< 11:23:40: CS actual		12/31	12/31
< 11:23:40: PWM scale		2	2
< 11:23:40: vsense		1=.18	1=.18
< 11:23:40: stealthChop	true	true
< 11:23:40: msteps		16	16
< 11:23:40: tstep		1048575	1048575
< 11:23:40: pwm
< 11:23:40: threshold		0	0
< 11:23:40: [mm/s]		-	-
< 11:23:40: OT prewarn	false	false
< 11:23:40: OT prewarn has
< 11:23:40: been triggered	false	false
< 11:23:40: off time		5	5
< 11:23:40: blank time	24	24
< 11:23:40: hysteresis
< 11:23:40: -end		2	2
< 11:23:40: -start		3	3
< 11:23:40: Stallguard thrs	-64	-64
< 11:23:40: DRVSTATUS	X	Y
< 11:23:40: stallguard			
< 11:23:40: sg_result		0	0
< 11:23:40: fsactive			
< 11:23:40: stst		X	X
< 11:23:40: olb			
< 11:23:40: ola			
< 11:23:40: s2gb			
< 11:23:40: s2ga			
< 11:23:40: otpw			
< 11:23:40: ot			
< 11:23:40: Driver registers:	X = 0x80:0C:00:00
< 11:23:40: Y = 0x80:0C:00:00

SSkull995, 2019-05-02:
[Configuration_adv.pdf](https://github.com/MarlinFirmware/Marlin/files/3137411/Configuration_adv.pdf)
[Configuration.pdf](https://github.com/MarlinFirmware/Marlin/files/3137412/Configuration.pdf)
[pins_RAMPS.pdf](https://github.com/MarlinFirmware/Marlin/files/3137413/pins_RAMPS.pdf)

SSkull995, 2019-05-02:
Above there are the three files I managed to change for the TMCs setup. Don't ask me why, but I can only export them as pdf and not txt

SSkull995, 2019-05-03:
Do you have any idea of how I could solve the problem? Right now I am using the physical end stops, which fortunately work great

Patag, 2019-05-03:
Stallguard thrs	-64	-64: this line is poking my eyes but I'm not an expert.
My M122 reports 8 8.
May be have a look on lines
   #define SENSORLESS_HOMING // TMC2130 only
and
    _#define X_STALL_SENSITIVITY  8
    #define Y_STALL_SENSITIVITY  8
in your Configuration_adv.h file

wbarber69, 2019-05-03:
set your stall sensitivity to 2 and try again.

and in your pins file the xmin and xmax are using the same pins as well as the y-cs and e1-cs.

SSkull995, 2019-05-03:
@wbarber69 but in this way I will decrease the sensitivity, isn't it?

SSkull995, 2019-05-03:
@wbarber69 I am changing the pins right now. However, the two pins are not used at all in the printer. Well, won't be used at all one I will disconnect the x end stop. Will I have to change the e1 as well? I don't use it at all

wbarber69, 2019-05-03:
-64 is much lower than 2....

SSkull995, 2019-05-03:
exactly. The lower, the more precise

wbarber69, 2019-05-03:
Not exactly.  If I run mine above 10 then they stall out at the slightest movement.  If I run any lower than 0 then don’t stall out at all.  Higher sensitivity means more prone to stalling.  Lower sensitivity means the opposite.  Precision is in the calibration.  Setting them so low means they won’t stall out.  Lower numbers have nothing to with precision.  The correct number is the precise one.

teemuatlut, 2019-05-03:
Maybe don't try values lower than 0. It's quite possible the logic isn't correct in the library as the datasheet is a bit ambiguous about the coding.

wbarber69, 2019-05-03:
I think he just misunderstood how it works tee.  By increasing the value you increase the sensitivity while the sensitivity spectrum is quite vast, it’s set to 8 as default for a reason.  Because in most situations that’s the area you want to be orbiting your settings around.

SSkull995, 2019-05-03:
so wait, if I increase the value they will be more tent to stop at the slightest movement, perfect. I tried at 2, they crash against the end of the rod. I tried with 8 and they crashed as well. I might have to increase the number then, not decreasing it

SSkull995, 2019-05-03:
I am sorry if I reply later, but I don't receive the  emails

wbarber69, 2019-05-03:
And to be clear you do have the diag pin attached to to the x and y min  “S” pin?

SSkull995, 2019-05-03:
@wbarber69 nope, I don't have that pin used on my board. I just tried with 20 as the sensitivity, but still crash

SSkull995, 2019-05-03:
I bet I have to put it, otherwise they don't send the signal to the board

SSkull995, 2019-05-03:
Ok, I tried with stall 2, I soldered the pin and... TA DA! It worked. I feel a bit dumb, because I didn't realise I needed the x_min pin, really stupid move. 
Thank you @wbarber69  and @teemuatlut! I was going crazy ahah

wbarber69, 2019-05-03:
The onboard chip is able to understand the stall but without a way to relay it to the main board it’s all but useless.  The spi pins are for configuration and debugging only.

SSkull995, 2019-05-03:
Lol, dumb move then. Once again, thanks!

closed 2019-05-03 by SSkull995