MarlinFirmware/Marlin #13870: [BUG] Subsequent G29 results incrementally decreasing

Zol-Tank, 2019-05-01:
### Description

I use the BLTouch to physically level the bed. I first do G28, then repeated G29 while manually adjusting the bed between G29, until the four corners of the mesh are within a couple 100ths millimeters of each other. I've used this technique successfully with 1.1.9 for awhile now.

I've recently started using 2.0 and have tried to use the same procedure. The first G29 appears to return the correct results. However, subsequent G29 will increment all mesh values by the same amount. If I run a G28, the subsequent G29 will again return the correct values.

The amount the values increment with are the same within a set of repeated G29, but NOT consistent between different sets of G29.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3135446/Marlin.zip)

### Steps to Reproduce

1. G28
2. G29
3. Run G29 again without first running G28.

**Expected behavior:** 
I expect repeated G29 to return the same values.

**Actual behavior:** 
Repeated G29 return values which are all incremented by the same amount.

#### Additional Information

Here is an example:
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 -0.015 +0.010 +0.027
Recv:  1 -0.002 +0.012 +0.015
Recv:  2 +0.020 +0.000 -0.012
Recv: 
Recv: X:234.00 Y:228.00 Z:9.94 E:0.00 Count X:18720 Y:18240 Z:4000
Recv: ok

Recv: echo:busy: processing
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 -0.090 -0.070 -0.050
Recv:  1 -0.085 -0.070 -0.057
Recv:  2 -0.055 -0.075 -0.085
Recv: 
Recv: X:234.00 Y:228.00 Z:10.02 E:0.00 Count X:18720 Y:18240 Z:4000
Recv: ok

Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 -0.165 -0.145 -0.125
Recv:  1 -0.153 -0.142 -0.132
Recv:  2 -0.127 -0.153 -0.162
Recv: 
Recv: X:234.00 Y:228.00 Z:10.09 E:0.00 Count X:18720 Y:18240 Z:4000
Recv: ok

Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 -0.240 -0.225 -0.205
Recv:  1 -0.233 -0.220 -0.210
Recv:  2 -0.203 -0.230 -0.247
Recv: 
Recv: X:234.00 Y:228.00 Z:10.17 E:0.00 Count X:18720 Y:18240 Z:4000
Recv: ok

Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 -0.327 -0.317 -0.292
Recv:  1 -0.325 -0.312 -0.305
Recv:  2 -0.290 -0.320 -0.332
Recv: 
Recv: X:234.00 Y:228.00 Z:10.26 E:0.00 Count X:18720 Y:18240 Z:4000
Recv: ok

Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 -0.410 -0.390 -0.370
Recv:  1 -0.410 -0.392 -0.387
Recv:  2 -0.377 -0.397 -0.407
Recv: 
Recv: X:234.00 Y:228.00 Z:10.34 E:0.00 Count X:18720 Y:18240 Z:4000
Recv: ok

mgrapiniak, 2019-05-01:
That is very interesting. I’m having issues with bltouch but I need to check if I’m getting similar results.
My issue is related to Z offset. I have to change it every time I start a new print. Let say I do G28,G29 and print something with Z offset -1.600. Then printing the same thing (G28,G29) and the nozzle goes straight into the bed so I have to manually change the offset to a new value.

FanDjango, 2019-05-03:
I tried this on my machine:
```
First G29
Bilinear Leveling Grid:
      0      1      2      3      4      5      6
 0 +0.036 -0.033 -0.036 -0.034 -0.037 -0.047 -0.057
 1 +0.061 +0.041 +0.015 +0.013 +0.004 -0.005 -0.055
 2 +0.076 +0.000 +0.003 -0.007 -0.017 -0.026 -0.043
 3 +0.089 +0.080 +0.054 +0.045 +0.036 +0.029 -0.023
 4 +0.091 +0.034 +0.030 +0.020 +0.016 +0.013 -0.003
 5 +0.086 +0.084 +0.064 +0.061 +0.059 +0.056 +0.024
 6 +0.089 +0.036 +0.039 +0.035 +0.044 +0.029 +0.019
X:154.50 Y:193.00 Z:5.27 E:0.00 Count X:30900 Y:38600 Z:4240

Second G29:
Bilinear Leveling Grid:
      0      1      2      3      4      5      6
 0 +0.036 -0.027 -0.037 -0.035 -0.039 -0.043 -0.057
 1 +0.067 +0.046 +0.020 +0.011 +0.004 -0.011 -0.044
 2 +0.076 +0.006 +0.009 -0.007 -0.011 -0.021 -0.036
 3 +0.089 +0.074 +0.054 +0.045 +0.036 +0.029 -0.016
 4 +0.091 +0.034 +0.030 +0.025 +0.016 +0.013 +0.003
 5 +0.091 +0.084 +0.062 +0.060 +0.059 +0.056 +0.017
 6 +0.088 +0.036 +0.039 +0.041 +0.043 +0.034 +0.024
X:154.50 Y:193.00 Z:5.27 E:0.00 Count X:30900 Y:38600 Z:4240
```
So the real question is: What is different?

Zol-Tank, 2019-05-07:
> I tried this on my machine:
> 
> So the real question is: What is different?

I don't know? Are you using a BLTouch?

I'm not sure this needs to stay open, since I now just do a G28 before each G29.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle