MarlinFirmware/Marlin #13872: [BUG] SKR V1.3 heatbead PID brick board and no retraction

matikyle, 2019-05-01:
Hello,
at first after I enable PID heatbead in configuration, board doesn't start up and i can't connect to it via USB (pronterface) and can't upload new firmware. Need to manually connect SD card to pc and copy new firmware with PID heatbead disabled to fix it.

Second problem is that retraction doesn't work. When sending manually gcode or enabling retraction extruder extrudes more filament. I have tested it on tmc2130 and drv8825 - no difference. Manually checked traces on board. Connections are fine. Tried to swap pins in marlin with E1 but after that extruder doesn't work at all. 

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3135635/config.zip)

mbabic84, 2019-05-01:
Hello, I use skr 1.3 with tmc2208 (in uart mode) with practically no issues at all last couple of days. Heatbed pid working perfectly, retraction as well. There are few known problems with linear advance, but beside of that everything working great.

ManuelMcLure, 2019-05-01:
Try increasing ``THERMAL_PROTECTION_GRACE_PERIOD`` in ``src/HAL/HAL_LPC1768/HAL.h``. It should be 800 in the latest sources, but it looks like that may not be high enough. Try setting it to 5000 or even 10000.

matikyle, 2019-05-03:
This doesn't change anything, still when i enable pid board bricks. Actually i need more help with retraction becasue i can live without heatbed on pid and retraction is necessary to have good loking prints.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle