MarlinFirmware/Marlin #13875: [BUG] IDEX full control: Wrong nozzle z position after tool change

phuxman, 2019-05-01:
### Description

Wrong nozzle z position after a tool change in IDEX full control mode. The position shown on the display does not represent the real position.

Done on Vivendino TREX 3.

### Steps to Reproduce
Do by terminal (all movements absolute):
- Heat + Home everything
- Raise z (safe movement)

Then repeat for extruders T0/T1 alternating
- Switch to extruder
- Move to defined position
- Lower by Z0.1 steps until feeler gauge 0.1 was fitting good -> Read Z on display
- Raise z (safe movement)
- Move extruder T0/T1 away

**Expected behavior:** 

- Static value (ideally Z0.1)

**Actual behavior:** 

Increasing value after every tool change

T0->Z0.1 
T1->Z0.35
T0->Z0.4
T1->Z0.6
T0->Z0.65
T1->Z0.8
T0->Z0.8
T1->Z1.1
T0->Z1.4
T1->Z1.65
T0->Z1.7

#### Additional Information

Tool offsets:
- Probe (BLTouch 3) offset: -1.35mm
- Second tool offset: -0.05mm

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3135998/Configuration.zip)

phuxman, 2019-05-01:
Needless to say: This behavior leads to the nozzles grinding deeper and deeper in the bed/material during print after every tool change.

InsanityAutomation, 2019-05-25:
Can you test with #14117

phuxman, 2019-05-26:
Hi. I tested it with the method described and the issue seems solved. The z position stays correct after tool changes. 

Should I close the issue?

InsanityAutomation, 2019-05-26:
Go ahead, although technically it's not completely closed until @thinkyhead clicks the merge button

phuxman, 2019-05-26:
Thank you very much for your efforts.

phuxman, 2019-05-31:
I tried to print in idex full control mode and there is still some offset off after tool changes. It is now way smaller than before but after about 10 tool changes it becomes visible. I will provide further information on how to reproduce.

Is there something specific I can try/monitor to help identify the problem?

InsanityAutomation, 2019-05-31:
I'll go over your config files tonight and see what you have atm. Mobile can't pull in the zip... Are you using any bed leveling?

InsanityAutomation, 2019-05-31:
I know that config!!! Its mine! lol This issue was also reported some time ago by another user. Ive just been getting back around to looking into things with full control mode so hopefully ill find it soon. It appears to be an issue with adjustments applied by bed leveling as travel across the mesh is no longer linear.

phuxman, 2019-06-01:
I am using ubl bed leveling. 

I think it is a small offset error which accumulates during tool changes. Was also thinking about the bed correction mesh data.

As soon the current print is ready I will start a idex full control print with ubl disabled.

Yes its yours. :P

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle