MarlinFirmware/Marlin #13878: NEWEST BLTOUCH NOT WORKING

Aurawin, 2019-05-02:
### Description

Recently purchased newest authentic BLTouch from Amazon.  I had problems with halting error because Z-min was always triggered.  After much research, I found the M43 S tool which indicated that inversion was likely needed.

I inverted both z-min and probe-z-min invertion = true (previously always false) and eventually realized that the conditionals_lcd.h was overriding my explicit directives.  Which at least on this new BLTouch is THE PROBLEM.

### Steps to Reproduce

Issue G34 and error persists even if I follow suggested probe instructions to invert signal.

**Expected behavior:** [What you expect to happen]

Firmware should do as the configuration calls for and assume person editing these settings does so purposely.

**Actual behavior:** [What actually happens]

Conditionals_LCD.h overrides Configuration.h values for Z-MIN inverting

#### Additional Information

I quickly edited the BLTouch block in Conditionals_LCD.h / flashed / and M43 S returned with expected results.  i.e.) no more inversion suggestions.

// Always disable probe pin inverting for BLTouch
  #undef Z_MIN_PROBE_ENDSTOP_INVERTING
  #define Z_MIN_PROBE_ENDSTOP_INVERTING true
  #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
    #undef Z_MIN_ENDSTOP_INVERTING
    #define Z_MIN_ENDSTOP_INVERTING true
  #endif

Please remove this code.  It is going to be a BIG problem for ppl who just need Marlin to do what we programmed it to do.

Roxy-3D, 2019-05-02:
You might want to review some of the BL-Touch v3.0 threads:

https://github.com/MarlinFirmware/Marlin/issues/13345

https://github.com/MarlinFirmware/Marlin/pull/13841

https://github.com/MarlinFirmware/Marlin/pull/13814

InsanityAutomation, 2019-05-02:
Its forced with the BLTouch by design because thats what it requires to function correctly. As many miss that step when using the z probe on the z min input, that automatic set was added. The behavior is correct and I wouldnt agree with changing it unless a bltouch like clone with logic functionally different appears. It would just be one more variable we would get support requests about when somebody misses it....

Roxy-3D, 2019-05-02:
> It would just be one more variable we would get support requests about when somebody misses it....

Agreed.   There has to be a 'real' need for it before it will happen.

closed 2019-05-02 by Roxy-3D