MarlinFirmware/Marlin #13882: [BUG] (short description)IDEX X-Offset X-Offset can not be set on the LCD

MickyPf, 2019-05-02:
I own a Vivendino TREX 3.

With the current Marlin 2.0 I can not adjust the X-Offset for the second extruder via the LCD.

Here is my config
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3137363/Configuration.zip)

phuxman, 2019-05-02:
I had to change the menu data  type for the x offset to float51 on my trex3. Before the value always jumped to a fixed value 327.XX as soon as I changed it. The reason is the value of the x offset (~450) violates the limits of the original datatype.

But you will lose one decimal place due to the data type.

InsanityAutomation, 2019-05-02:
This is on my list to work on this weekend. I plan to look for a way to keep the precision. Im looking at allowing babystepping x and y to adjust these offsets as well which should give a better adjustment and ability to live-align.

MickyPf, 2019-05-02:
Thanks for the answers. Currently I enter the XOffset via M218.

InsanityAutomation, 2019-05-12:
Can you test with todays bugfix branch? A PR was just merged that may fix this. If not we may need to take a closer look. If this is now working, please go ahead and close the issue. Thanks.

MickyPf, 2019-05-13:
Great, it works

InsanityAutomation, 2019-05-13:
Sounds good, feel free to close this issue then!

closed 2019-05-13 by MickyPf

richialex, 2019-06-08:
> I had to change the menu data type for the x offset to float51 on my trex3. Before the value always jumped to a fixed value 327.XX as soon as I changed it. The reason is the value of the x offset [...]

I'm twisted. Can you tell me exactly where you made the changes? Since the problems still remain and when trying to regulate the offset from the LCD, the value is reset to fixed.