MarlinFirmware/Marlin #13883: [BUG] TMC5160 not showing rsense

tvm2487, 2019-05-02:
I have been trying to setup my skr v1.3 with the new TMC5160 drivers in SPI mode and sensorless homing I have been able to get all that working the issue I'm having is when I run M122 the report is not showing a rsense value. I am not sure if this is a firmware bug or if something is not configured correctly

kAdonis, 2019-05-02:
rsense is the value (in Ohm) of the current sense resistor on the stepstick PCB, it's impossible to read from software.
Watterott TMC5160 SilentStepsticks (and all other TMC5160 I have seen) are using 0,075Ω sense resistors.

Or do you mean vsense? Its a register in older Trinamic chips but has been replaced by Globalscaler in newer chips, so its not shown with M122

tvm2487, 2019-05-02:
Oh damn your right after looking at the image again its vsense thats not showing seen in the picture below 

https://drive.google.com/file/d/1hNrd5BQmnBqh5wiin64OFTqVXaK2sky7/view?usp=sharing

kAdonis, 2019-05-02:
I wouldn't call it a bug, but reporting vsense makes no sense for the TMC5160.
I made a PR ro remove the obsolete vsense from M122

tvm2487, 2019-05-02:
Ah ok cool thanks, there is not much information available for these drivers just yet kinda just trial and error

kAdonis, 2019-05-02:
It's all here:
https://learn.watterott.com/silentstepstick/

tvm2487, 2019-05-02:
But I cant find anywhere how to setup SPI mode on drivers theres a resistor across both the SPI and SD pads on the board i'm trying to figure out if thats right or now

kAdonis, 2019-05-02:
Which boards do you have, the Trinamic TMC5160-BOB?
SPI-mode and S/D-mode need to be set to VCC_IO

tvm2487, 2019-05-02:
I have these boards here and I'm using them on an skr v1.3 32 bit main board with sensor less homing 

https://www.amazon.com/TMC5160-Stepper-Stepstick-Printer-Control/dp/B07Q9Y98K9/ref=sr_1_11?crid=2PWKOY8CDUCEM&keywords=tmc5160&qid=1556801578&s=gateway&sprefix=tmc5%2Caps%2C130&sr=8-11

kAdonis, 2019-05-02:
Here is the schematic:

https://github.com/bigtreetech/BIGTREETECH-TMC5160-V1.0/blob/master/hardware/TMC5160-Schematic.pdf

The resistors are probably 0Ω and connecting SPI-mode and SD-mode to VCC_IO (W1 and W2 in the schematic) so thats fine
The CLK pin needs to be grounded, yes:
https://github.com/watterott/SilentStepStick/blob/master/hardware/SilentStepStick-TMC5160_v13.pdf

kAdonis, 2019-05-02:
Afaik, sensorless homing is only working in spreadcycle. Only "non homing" movements use stealthchop
All Trinamic drivers are quieter with 24V and you can tune `CHOPPER_TIMING` to get spreadcycle less loud

tvm2487, 2019-05-02:
Oh really that I didn't know, I'm guessing i can't set CHOPPER_TIMING to 24V if I'm not using a 24V psu correct. I guess tonight I will have to run a test print and see how everything does

kAdonis, 2019-05-02:
I think you got me wrong. The drivers are quieter when connected to 24V.
You can try different presets for `CHOPPER_TIMING` but ` CHOPPER_DEFAULT_12V` is probably better than others when using 12V. You can fine tune `CHOPPER_TIMING`
```
   * Define you own with
   * { <off_time[1..15]>, <hysteresis_end[-3..12]>, hysteresis_start[1..8] }
```
but this is an advanced task

tvm2487, 2019-05-02:
Oh okay and I also another question should I be setting anything for MINIMUM_STEPPER_PULSE and MAXIMUM_STEPPER_RATE both are commented out in my current config file

kAdonis, 2019-05-02:
There is an ongoing discussion about `MINIMUM_STEPPER_PULSE`
I would try to leave it commented out, If you have problems with Linear Advance or you have layer shifts with a coreXY you may try  `MINIMUM_STEPPER_PULSE` 1 to 4

tvm2487, 2019-05-02:
Okay cool thanks so much for all the help and information, I'm going to try to run a test print tonight and see how everything comes out. I'm glad I was able to get everything setup correctly the wireless SPI setup on the new TMC5160 drivers and the skr v1.3 board is pretty awesome

flthymcnsty, 2019-05-27:
Hey tvm2487, I have a similar set up to yours (skr v1.3 with TMC5160s). When you wired the CLK pin to ground did you clip off the pin below it going into the socket or is it OK to keep it?

davidcgu, 2019-06-08:
I'm trying to get sensorless working for TMC5160 guys, if I understood well should I bridge CLK pin with ground?  can I keep the original pin on the socked anyway or should I remove it?  Thanks in advance

kAdonis, 2019-06-08:
The CLK pin on the driver board needs to be removed, because most mainboards connect this pin to the neighboring pin, which is, in case of the TMC5160, the MISO pin. With the CLK pin connected to ground the MISO pin gets also grounded and SPI communication fails

davidcgu, 2019-06-08:
Thanks for your reply, I'm trying to make it run on a core XY on an SKR v1.3 board, I just tried as you mentioned, removed the CLK pin and jump the connection with GNG from same driver however still not working.  I welded a pin on Diag1 and put the jumper on X&Y+ at right of the board, I should be missing something.... any idea?

Patag, 2019-06-08:
With the same configuration as yours but on a Cartesian printer, I had to set sensitivity to 1 on X axis (Y not tested yet).

davidcgu, 2019-06-08:
Thanks for your reply, I have the bump but does not work for me, what do you have on config pullups/downs and inverting the logic?

Patag, 2019-06-08:
Probably a silly remark: 
`#define SENSORLESS_HOMING // TMC2130 only
` in configuration_adv.h
Endstops in configuration.h
`#define USE_XMIN_PLUG`
`#define USE_YMIN_PLUG`
`#define ENDSTOPPULLUPS`
`#define X_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.`
`#define Y_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
`

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle