MarlinFirmware/Marlin #13886: Second extruder doesn't move

SSkull995, 2019-05-02:
Hi! 
First of all, thanks in advance to anybody who will answer me. Secondly, I have a MKS Gen 1.4 board with marlin 1.1.9; I am having a problem with my second extruder, which is correctly connected to the board in the E1 slot. When I try to run G1 E10 T1 on repetier, not only the second extruder's motor doesn't move, but the first extruder's one moves counterclockwise, like it is retracting. 
It is very weird actually, because all the other features of the second extruder work fine, temps and probe. However, as soon as I send the GCode, only the E0 works. Ah, with G1 E10 T0 the first extruder moves correctly like it is loading the filament. Another strange thing is that on Repetier, the second extruder doesn't come up, I have to manually set the temp with the M109 command and then it works correctly. Any help will be highly appreciated, thanks!

shitcreek, 2019-05-02:
Welcome to the Marlin Github.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) 

For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

That said, I am willing to guess you haven't configured the second extruder.

SSkull995, 2019-05-02:
Sorry @shitcreek, didn't want to do anything wrong. Will ask to someone on the reprap forum then. The extruders are configured btw, on Cura there are no problems

closed 2019-05-02 by SSkull995