MarlinFirmware/Marlin #13889: [BUG] Changing EXTRUDERS changes layout of EEPROM

marcio-ao, 2019-05-02:
At the moment, it looks like the number of EXTRUDERS changes the layout of the EEPROM. This causes problems with printers such as our with interchangable toolheads.

Would it make sense to reserve an amount of space independent of number of extruders? At the least, the stuff related to EXTRUDERS would need to be moved to the end of the EEPROM so it doesn't cause everything else to shift.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@marcio-ao is the issue still there?

marcio-ao, 2019-09-24:
@boelle: Yes.

boelle, 2019-10-14:
> Would it make sense to reserve an amount of space independent of number of extruders?

to a non programmer like me the answer would be yes

boelle, 2019-11-06:
@thinkyhead @shitcreek time to have a look here before the issue gets lost and not fixed

boelle, 2019-12-07:
i guess this one still needs to be worked on?

boelle, 2019-12-24:
what can we do to fix this one?

closed 2020-01-07 by boelle