MarlinFirmware/Marlin #13891: [BUG] ULTI_CONTROLLER does not turn on in Marlin 2, but works in 1.1.9

DavidBjerreBjoerklund, 2019-05-02:
### Description
Ultimaker 2 display remains black, even though ULTI_CONTROLLER has been defined in config.
This only happens in Marlin 2.0.x, if the same configs are set in 1.1.9 LCD works without a problem!

### Steps to Reproduce
Download Bugfix-Marlin 2.0.x
Define MOTHERBOARD BOARD_ULTIMAIN_2
Define ULTI_CONTROLLER
Upload to Ultimaker 2 board. Display goes black.

Do the same thing on Marlin 1.1.9 - No problems, display works like a charm!
Uploaded both config files:

[Marlin_1.1.9_Ulti_controller_WORKS.zip](https://github.com/MarlinFirmware/Marlin/files/3139792/Marlin_1.1.9_Ulti_controller_WORKS.zip)
[Marlin_2_Display_Is_Black.zip](https://github.com/MarlinFirmware/Marlin/files/3139795/Marlin_2_Display_Is_Black.zip)

thinkyhead, 2019-05-03:
Do you have an oscilloscope? It would be helpful to compare the SPI output of both 1.1.9 and 2.0.x to see how they differ. We can't find any obvious differences in the code on first pass, and it's unlikely that we'll be jumping in to do a slew of troubleshooting. So, any diagnosis on your end will be invaluable.

DavidBjerreBjoerklund, 2019-05-03:
Yes I do, but only two channels I think! Happy to help as much as I can.
Could an Arduino be helpful to read SPI output?

DavidBjerreBjoerklund, 2019-05-04:
Ok I have tried today to get a reading using my oscilloscope, actually a picoscope. I have been unsuccessful in getting any data dump. But one thing I did notice was that there was almost no signal like activity when Marlin 2 was loaded. But when marlin 1 is loaded there is a lot of binary activity, looking like square signals.

It looks like the ssd1309 is contructed as a I2C display.

Also It looked like commands for updating the display (I did turn on I2C protocol analysis). When I changed things in the menu, I got a lot of data in Marlin 1.
I think I need a better scope to get further.

soimy, 2019-05-09:
Ulticontroller use I2C bus, I can confirm it working with RAMPS-FD through https://github.com/soimy/UltiAdapter-FD

boelle, 2019-07-21:
I just used bugfix-1.1.x and the problem disappeared

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@DavidBjerreBjoerklund do you still have a problem?

DavidBjerreBjoerklund, 2019-09-25:
I have stopped trying. I did not have either skills nor equipment to come
any further.

boelle, 2019-10-24:
@shitcreek any idea what we can do here?

boelle, 2019-11-03:
will close this one as @soimy says its working

we can reopen if time skills etc make it possible to test it

closed 2019-11-03 by boelle