MarlinFirmware/Marlin #13893: [FR] (feature request title)

gouache, 2019-05-03:
### Description

new gcode for auto Steps (X/Y/Z/E) Calibration

### Steps to Reproduce
GXXX [X<distance>] [Y<distance>] [Z<distance>] [E<distance>]

1. GXXX X100
2. laser range finder or digital calipers measure moving distance
3. calculate new x step using difference 100 mm with real moving distance
4. laser finder or digital calipers can be used for X/Y/Z step
5. color sensor can be used for E step
  5-1. color sensor detects white or black marking on filament

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

- range finder example : https://www.youtube.com/watch?v=Q_TQ68qFiJc&feature=youtu.be&t=4m35s
- color sensor example : https://www.youtube.com/watch?v=ok5L-zcKd8g&feature=youtu.be&t=4m52s
- calipers example : https://3.bp.blogspot.com/-krFgCjx_n0Q/XIMIQswKLRI/AAAAAAABhE8/w8IMjwJWeNwaTiRrqT0ZvRy6XlgWFMnFwCLcBGAs/s640/20190309_084121.jpg

Roxy-3D, 2019-05-05:
How does #2 get into the system for the Marlin firmware to use?
Is it a floating point number?   How accurate is the number it reports?
How repeatable is the number it gives us?    It almost sounds like we should
get a parameter added to M48 to do Standard Deviation on how accurate the movement is.
```
2- laser range finder or digital calipers measure moving distance
```

jmerc77, 2023-02-27:
another way for e calibration I thought of is if using a tachometer based runout sensor, compare motor movement to mm/tick when going from one tick to the next.