MarlinFirmware/Marlin #13897: Dual X Carriage with 4 nozzles

ssaid1, 2019-05-03:
# NO SUPPORT REQUESTS PLEASE

Support Requests posted here will be automatically closed!

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. See https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question.

For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org Marlin Forum http://forums.reprap.org/list.php?415
- Tom's 3D Forums https://discuss.toms3d.org/
- Facebook Group "Marlin Firmware" https://www.facebook.com/groups/1049718498464482/
- Facebook Group "Marlin Firmware for 3D Printers" https://www.facebook.com/groups/3Dtechtalk/
- Marlin Configuration https://www.youtube.com/results?search_query=marlin+configuration on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a Bug Report at https://github.com/MarlinFirmware/Marlin/issues/new/choose).

ssaid1, 2019-05-03:
Hi,
  I am building 3d printer with dual x carriage. And each x extruder have 2 nozzles. Totally, there are two x axises and four nozzles. I could not do this in Marlin firmware ? Marlin firmware is introduce, there is just one nozzle in first x axis and other 3 of them are in second one. This should not be. Are there anyone who can help me about this ?

Roxy-3D, 2019-05-03:
That configuration is not currently supported.    But you could add the support in a fairly straight forward manner.    If I was doing this, I would use the IDEX code in Auto Park Mode and make it so each carriage had two tools on it.

This isn't currently supported.   But there are several of us developers that have an open and on-going offer to any printer manufacturer.    We want to by-pass what you need.   We want to evolve the firmware to support 3-color mixing extruders on IDEX machines.      

But so far...   no manufacturer has stepped up and said they want a product like that.

ssaid1, 2019-05-03:
I could not tell. I already want to each carriage has two nozzles. But I could not provide this feature. I am trying to use like this. For example, when the gcode work with second extruder, it gets warm, it's extruder step motor is working. But Second X carriage is working instead of first x  carriage. I could not solve this problem. 
[image]

InsanityAutomation, 2019-05-03:
In my opinion that defeats most of the advantages of idex needing to align the nozzles to each other, and fighting crossing between them. Bowden y tubes may be more doable here.  Roxy already mentioned the other concept we've been discussing. I've got a mechanical concept of 4 independent heads with x and y dual carriages. To support those, we've been seeking a board to drive 10 steppers.

rlesac, 2019-05-03:
> In my opinion that defeats most of the advantages of idex needing to align the nozzles to each other, and fighting crossing between them. Bowden y tubes may be more doable here. Roxy already [...]

Sounds like you're looking for a Duet with an expansion board. It supports 5 steppers on the main board and 5 more on the expansion board.

InsanityAutomation, 2019-05-03:
I'm not a fan of fixed drivers especially TMC. They're too sensitive and especially during development work likelihood of something getting fried is pretty high. I would very much prefer to not have to do surface component soldering when a single driver gets burned up. Cost of the board is also higher than what most manufacturers would be willing to spend building a machine so then I don't expect traction to be gained for the kinematic styles we're discussing.

ssaid1, 2019-05-04:
I drawed my board myself referencing ramps board. There are more than 10 stepper on the board and something more that I can need.  I configured by using ramps 1.4 as a mother board in software. So I think, there should be something in software that I can fix this problem. So can you say anything about this ? My problem is; when I sent the gcode to the machine, second extruder is moving but first extruder getting warm. I hope, I could tell the problem to you.

Roxy-3D, 2019-05-04:
For IDEX machines...   That makes sense.    The first extruder is assumed to be on the first carriage, and other (the 2nd!!!) extruder is assumed to be on the second carriage.     

You need to modify the firmware to get your machine working the way you want it to work.   If you want somebody else to modify the firmware for you, that is possible, but you will need to send the developer a machine so they have a test platform to develop the code.   

You probably are best off just using supported configurations for now.    We are trying to get the changes you need into the code base.   But we don't have machines to test against right now.    (And as stated earlier...   We are hoping to get an IDEX platform with color mixing extruders to do the work.)

InsanityAutomation, 2019-05-04:
Though a quick pointer, changes would be needed in stepper indirection, tool change, sanity check, and m605 for sure, probably more in stepper.cpp and a few other things as well would be found as someone digs in.

incrediblesolutions, 2019-05-05:
this is overly complicated.  there are triple heads available.  then one heater gets access to all tyhe colours or mixes of you wish to layer for strength.  then two heads are not needed for the movement and three steppers can be driven with a calculation for the time and amount of filament per step needed during the feeder to nozzle change, one heater nozzle three colours.