MarlinFirmware/Marlin #13906: [BUG] Marlin 2.x - Issue with coordinate definitions, buildplate size and workspace offsets

bipsendk, 2019-05-04:
### Description

It would be desirable to be able to pre-define workspace offsets (M206), as there are some functions that doesn't work well without them - especially if having relatively large workspace offsets.
It seems like validation of coordinates in sanitychecks and bed-leveling-calculations (and probably more functions) doesn't take the workspace offset into consideration.

### Steps to Reproduce

Define a printer with buldplate of 300x300 mm
Workspace offsets are X -40 Y -50
Probe offset on X = 27 (no Y offset)
Travel limits needs to be set to:
```
#define X_MAX_POS X_BED_SIZE+40
#define Y_MAX_POS Y_BED_SIZE+50

```as the workspace offset not is being considered during compilation.
When doing bilinear bed leveling with MIN_PROBE_EDGE 10, probing will initially start outside of the builplate.
Also, if using these workspace offsets, it is not possible to specify 
```
  #define LEFT_PROBE_BED_POSITION 40
  #define RIGHT_PROBE_BED_POSITION 310
  #define FRONT_PROBE_BED_POSITION 50
  #define BACK_PROBE_BED_POSITION 350

```as the sanity check will tell you, that these coordinates are outside the probe region.

#### Additional Information

I suspect, that there might be a general issue when having larger offsets. In my case the offsets are large, as the printer has been built, so that I can upgrade it with a larger buildplate in the future. But these large workspace offsets seems to generate challenges for safe z homing (which not will take place at the center of the buildplate due to the offset), bed leveling, probably also Z-stepper alignment.. All because the workspace offsets apparently not can be pre-defined, nor is being used in the calculations of coordinates afterwards (where I would expect 150,150 to be on the center of my buildplate after homing).

[Config-large_m206.zip](https://github.com/MarlinFirmware/Marlin/files/3144621/Config-large_m206.zip)

gloomyandy, 2019-05-04:
If my understanding of your setup is correct it may make more sense to change the way you are defining the base coordinate system. Have you tried setting X_MIN_POS to -40 and Y_MIN_POS to -50?

FanDjango, 2019-05-04:
@gloomyandy Beat me to it. 
What we are reading into your information, is that afer homing into the endstops, your workspace is somewhere further in, and beyond the workspace there is more room even. So the X_MIN_POS and the Y_MIN_POS brings your (0,0) to the workspace corner, and X_MAX_POS, Y_MAX_POS as well as Bed Size can be set as is. Your probe positions may than be truthfull bedsize/2 to get to the middle.

bipsendk, 2019-05-04:
I will try to modify the X_MIN_POS and Y_MIN_POS to see if that corrects the issues. Will be back with an update.

gloomyandy, 2019-05-04:
Just to be clear, with the above changes you should not need M206. There could still be a problem with M206, I don't think many people use it.

bipsendk, 2019-05-04:
When uploadning the new firmware, M206 will be set to 0,0 - just to be sure...

bipsendk, 2019-05-04:
Setting the X/Y_MIN_POS did the trick...

closed 2019-05-04 by bipsendk