MarlinFirmware/Marlin #13910: [BUG Marlin 2.0.x] Z offset issue

mgrapiniak, 2019-05-04:
### Description

Setting up Z-probe offset works for the the first print but next time I print the same gcode nozzle hits into the bed and offset value has to be amended again (usually by the order of magnitude). Issue is reproducible on demand.

- Ender 3
- 3Dtouch clone by trianglelabs
- SKR 1.1 board
- TMC 2208 on all axis

### Steps to Reproduce

1. Compile and flash Marlin 2.0.x firmware into SKR 1.1 
2. M502 and M500
3. Start printing using attached gcode (before the print G29 is issued from withing gcode file itself)
4. Double click the knob for babysteps shortcut for Z-offset settings
5. Offset changed to -4.150 to get a first layer correctly.
6. Printing ends.
7. Store settings in EEPROM via Menu->Configuration (or M500)
8. Power cycle the printer
9. Opening the same gcode from SD card and starting the print
10. After G29 finish nozzle hit the bed so have to compensate by manually adjusting Z-offset again from babysteps doubleclick or from the menu
11. New value of Z-offset -1.150

**Expected behavior:** [What you expect to happen]
Not to set Z-offset every time the print is started

**Actual behavior:** [What actually happens]
New Z-offset has to be set manually every time a new print is started 

#### Additional Information
* Issue reproducible on demand.
* Configuration.h and Configuration_adv.h attached.
[z-offset-issue.zip](https://github.com/MarlinFirmware/Marlin/files/3144925/z-offset-issue.zip)

phuxman, 2019-05-05:
I think I experienced the same during setup. Does this only happen if you adjust the offset during print?

Aurawin, 2019-05-05:
I have experienced this as well.  I would need to readjust z-offset and the space between the nozzle and the bed kept getting closer and closer.  I had ABL bilinear which is a nightmare at this point. I disabled ABL and manually adjusted my bed.  Now I have no problems.

mgrapiniak, 2019-05-05:
> I think I experienced the same during setup. Does this only happen if you adjust the offset during print?

Not really. During the print you can set up Z-offset as well but eventually it should end up in EEPROM (I think)

mgrapiniak, 2019-05-05:
> I have experienced this as well. I would need to readjust z-offset and the space between the nozzle and the bed kept getting closer and closer. I had ABL bilinear which is a nightmare at this point. I disabled ABL and manually adjusted my bed. Now I have no problems.

Yes. I agree that using ABL would make no sense with such an issue. However since I have already mounted probe sensor it would be nice to take advantage of this and hence reported it here.

I've been struggling with this issue for weeks now. Perhaps I have missed something simple in configuration but someone smarter then me would have to verify.

CRCinAU, 2019-05-05:
If you try this, what happens:
```
M502
M500
M851 Z-4.150
G28
G1 Z0 F300
```

How far is it off the bed when it thinks its Z == 0?

mgrapiniak, 2019-05-05:
> If you try this, what happens:
> 
> ```
> G90
> M502
> M500
> M851 Z-4.150
> G28
> G1 Z0 F300
> ```
> 
> How far is it off the bed when it thinks its Z == 0?

I stopped here as the nozzle crashed into bed surface

CRCinAU, 2019-05-05:
did it hit the bed before of after the G28 (homing)?

I don't know what mount you have, but -4.15 is a lot... Mine is on the stock Creality mount - and I use -2.70 (which is a fraction too much).

mgrapiniak, 2019-05-05:
> did it hit the bed before of after the G28 (homing)?
> 
> I don't know what mount you have, but -4.15 is a lot... Mine is on the stock Creality mount - and I use -2.70 (which is a fraction too much).

I execute gcode line by line in terminal. G28 works fine. Probe get to the middle of the bed, and after successfull homing lift about 10mm and stop. Then I execute `G1 Z0 F300` and it crashes into bed surface :/

CRCinAU, 2019-05-05:
I'd suggest your offset is too much.... Try this:

```
M502
M500
M851 Z0
G90
G28
G1 Z0 F150
```

That's pretty slow, so it should crash much - see how high it is off the bed with a zero offset.

mgrapiniak, 2019-05-05:
> I'd suggest your offset is too much.... Try this:
> 
> ```
> M502
> M500
> M851 Z0
> G90
> G28
> G1 Z0 F150
> ```
> 
> That's pretty slow, so it should crash much - see how high it is off the bed with a zero offset.
> 

Around 1.5 mm above the bed

> > But why it works fine when I tune Z-offset to this value during the print ?
> 
> You can also set the offset in the slicer software. I'm guessing you have different values set on the printer vs the slicer.

I have attached gcode in question to description. I will examine the gcode later on.

CRCinAU, 2019-05-05:
Ok, retest now with:

```
M502
M500
M851 Z-1.5
G90
G28
G1 Z0 F150
```

That should get you close to the bed. Try it with a bit of paper and tune it that way.

Don't set a Z offset in your slicer (ie use 0.00) and see how that behaves.

mgrapiniak, 2019-05-05:
> Ok, retest now with:
> 
> ```
> M502
> M500
> M851 Z-1.5
> G90
> G28
> G1 Z0 F150
> ```
> 
> That should get you close to the bed. Try it with a bit of paper and tune it that way.
> 

Appreciate your advice. I will try it for sure but shouldn't we have a better (more user friendly) method of setting the Z-offset ?

> Don't set a Z offset in your slicer (ie use 0.00) and see how that behaves.
>

I checked Cura (using 3.6). No reference to Z offset whatsoever.
 
> EDIT: You can tweak the value of M851 as required to get the right offset. Then store it (M500) and forget it :)
I'll give it a try.

phuxman, 2019-05-05:
I dialed my z offset in by entering a default value (-1.5mm) which makes sure the nozzle does not crash into the bed. Then I move the extruder down by 0.1mm steps until a sheet of paper fits tightly. The final value is added to the z-offset afterwards.

Example: 

Initial offset: -1.5mm
3 times 0.1mm down to fit sheet of paper
Final offset = -1.5 + 3 * 0.1 = -1.2 

As soon as I tried to change the offset(s) during print and save it afterwards there was strange behavior (but I can not recall is by detail).

mgrapiniak, 2019-05-05:
> I dialed my z offset in by entering a default value (-1.5mm) which makes sure the nozzle does not crash into the bed. Then I move the extruder down by 0.1mm steps until a sheet of paper fits [...]

That doesn't make sense for me. Are you sure your math is correct ?

I mean if you set up -1.5 safe value that means your nozzle doesn't touch the bed. Then you increase the value by 0.1 which is fine as you want to make it tight. Shouldn't the result be:
Final offset = -1.5 + 3 * (-0.1) = - 1.8 

Please correct me if I'm wrong

CRCinAU, 2019-05-05:
Yeah - the higher the number, the further away. A negative number is below the probe, a positive number is above it.

mgrapiniak, 2019-05-05:
I will make multiple prints of the same model now to verify if the result is consistent. Thanks !

phuxman, 2019-05-05:
Yeah, maybe with default offsets my nozzle was already touching the bed. 

But I wanted to show a general principle to dial in z offset without printing and much hassle:

-> Enter default z probe offset for your printer
-> Home
-> Adjust z by stepping until sheet of paper fits
-> Read z position and add it to the z probe offset
-> Store
-> Home
-> Verify result

mgrapiniak, 2019-05-05:
> Yeah, maybe with default offsets my nozzle was already touching the bed.
>
> But I wanted to show a general principle to dial in z offset without printing and much hassle:
>
> -> Enter default z probe [...]

I have made few tests and results are consistent nevertheless setting Z-offset during the print produce chaotic results and definitely the method you mentioned above would be recommended instead of setting Z-offset during the print. 
My gut feeling is there may be some nasty bug hidden in Marlin 2.0.x

Anyways thanks for your help.

closed 2019-05-05 by mgrapiniak

alex-stefan, 2020-02-05:
I have the same issue. If I set the Z-Offset through the Configuration menu and then store it, it will be forgotten after a power cycle. Running on SKR Mini E3 V1.2

Send: N7 M115*33
Recv: FIRMWARE_NAME:Marlin bugfix-2.0.x (GitHub) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 EXTR

thinkyhead, 2020-02-07:
@alex-stefan — Please file a new issue and be sure to fill out the entire template.

alex-stefan, 2020-02-07:
Thanks for replying! Located another similar issue which also confirmed the fix, so will just compile the firmware and I should be good to go!

TheGadge, 2020-02-07:
> Thanks for replying! Located another similar issue which also confirmed the fix, so will just compile the firmware and I should be good to go!

Where? @ alex

AnakinSpain, 2020-04-16:
I have the same problem and I don't know how to solve that.

avolmensky, 2020-05-06:
I also have the same problem. When adjusting z-offset during a bed level test print it will work fine for that print, next print the head will crash into the bed. Before I upgraded Marlin, I used to get about 3mm, but when tuning during the print it's over 4mm now.

alex-stefan, 2020-05-06:
> > Thanks for replying! Located another similar issue which also confirmed the fix, so will just compile the firmware and I should be good to go!
> 
> Where? @ alex

I have an SKR Mini E2  and just got the latest pre-compiled firmware from their repository and the issue is no longer there.

thinkyhead, 2020-05-11:
The default delay for Trinamic steppers will now be 60ns, so let's see how that plays out over the coming weeks…

randellhodges, 2020-05-11:
I've been running 60ns on my 2209s for over a week now (0.9 degree steppers all around, with a bmg extruder) and I haven't seen a problem that I can attribute to this. (But I am printing slow, low acceleration and little jerk, classic)

`#define DEFAULT_AXIS_STEPS_PER_UNIT   { 160.0, 160.0, 400, 830 }` (Using 1/4 stepping on my dual z) and using LA.

I have seen an increase in VFA at some point this week but I cannot say that this is the cause. I think I have a hardware issue. I only bring it up as something to keep an eye out for.