MarlinFirmware/Marlin #13911: [BUG] ABL should redo points after a failure

Keichide, 2019-05-05:
I have a BLTouch V3 installed and configured to do 4 Probes per Point to average the result i'm using a 4x4 Grid with Auto_Bed_Level_Linear since i have a flat Glassbed on the Printer. The Issue is, that often 1 of the 4 Probeings fails but that one point is still calculated into the average height, which drops of the one level and gives me spikes which should'be there. 

The Strange thing is, that no one ever mentioned it or fixed it, since i had it in Marlin 2.0 where i also enabled the reprobe and recover option and now in Marlin 1.1.9 Bugfix, since the 2.0 branch just was'nt smooth enough on an Ender 3. 

I think the whole probing behavior should be changed to check if the Probe failed or not and redo that single point or at least ignore it. 

Attached is one example, where 1 of the 4 Probings failed (and i know that the level is way of, i have to print a few washers for the mod that i'm using to get the right side up a bit because i can't lower the left side anymore)

[image]

FanDjango, 2019-05-05:
I totally agree with this. I will think about adding that to my PR #13841

Keichide, 2019-05-05:
I don't know if it is just an issue with the v3 or if the v2 also had this, but i think many others still face this bug, but just don't know what todo about it. But anyway since it's really a hassle todo the G29 5-10 times to get one single G29 output where the probe never failed, i'm happy to help you with it, and test it. 

Also this is a bit OT but i think the Descriptions on the Configuration.h should be a bit expanded because i still don't know what the open drain option does, and just found out that i had to enable the 5V Mode because of the Teaching Tech Video on Youtube.

FanDjango, 2019-05-05:
This is the version I was proposing in my PR #13481. Dunno if any of the text will find it's way in to Marlin, but if it helps you, here it is:

```
/**
 * The BLTouch probe uses a Hall effect sensor and emulates a servo.
 */
#define BLTOUCH
#if ENABLED(BLTOUCH)
  /**
   * For all BLTouch probes and clones:
   */
  #define BLTOUCH_DEBUG_MSGS  // Produce some debugging messages when commands are issued to the probe
  #define BLTOUCH_DELAY 200   // (ms) Allow the servo signal to settle for this long before the next command

  /**
   * For BLTouch CLASSIC V1.2, V1.3, SMART 1.0, 2.0, 2.2, 3.0 and newer probes which support SW mode. 
   * Some clones support this too.
   * These support the full integration into the Marlin probing architecture.
   */
  /**
   * Do not allow SW mode to be used for status querying, because the probe does not support it.
   * Either because it is a strange clone or because it is a very old BLTouch CLASSIC 1.0 or whatever.
   * This takes the probe out of the full Marlin probing architecture.
   * Also use this if your Z-AXIS is set to such slow movement, that you need separate STOW-DEPLOY
   * sequences on probing.
   */
  //#defined BLTOUCH_SW_MODE_UNUSABLE
  /**
   * Set the probe into SW mode after deploy, producing a longer pulse when triggered, useful when 
   * there is noise or high capacitance on the signal pin.
   * If your probe works fine without this, don't turn it on, as it slows down probing
   */
  //#define BLTOUCH_FORCE_SW_MODE
  
  /**
   * For BLTouch V3.0 and newer probes:
   * If you want the new functions of the BLTouch V3 or need 5V mode (see below), enable this.
   * This adds the special V3 commands (5V, OD, SW modes) to the LCD configuration menu for BLTouch
   */
  #define BLTOUCH_V3
  #if ENABLED(BLTOUCH_V3)  
    /**
     * These probes default to OPEN DRAIN mode. All other probes are in 5V mode per default.
     * If you have a BLTouch V3.0 or newer and explicitely want 5V mode, use this define -  this 
     * makes it more compatible with previous probe versions. 
     * Warning: Make sure your controller boards input pin is 5V tolerant.
     */
    #define BLTOUCH_FORCE_5V_MODE
    /**
     * Advanced users who fervently believe in the reliability of their BLTouch probes can 
     * activate the HIGH SPEED mode for their unit. This will reduce the timing so far that 
     * DEPLOY and STOW failures are only captured on the NEXT such command to be issued.
     * The result is faster probing but much less surety of catching and identifying errors.
     * NOT RECOMMENDED for novice users and in unstable environments. As homing and probing are
     * not done so often, the speed gain is nice but not really life changing.
     */
    #define BLTOUCH_HS_MODE
  #endif

#endif
```

FanDjango, 2019-05-05:
After looking at the code, I can see some problems / questions:

In my PR, the code would retry if a deploy or stow fails. If the retry fails the probe is probably blocked, the process will stop. If it succeeds the probe point should be ok.
Only two other possible fails, is a) probe does not trigger before hitting bed and b) probe triggers too early (due to noise). In the case of a) I would advise to check a few things before continuing. And for b), one would need to define how big the deviance should be for this to be considered an error, and also **one should endevour to eliminate the cause of the false triggers**.

In the current code, success of deploy and stow is not checked - a failure is cleared at the beginning of the probe for the NEXT point and the point where the probe failed is added to the mesh, which is the mistake.

So if you are getting many bad probes on a 4x4, after a little thought, I would do the following:
1. Fix the probe so it works ALL the time. At least try. I can help you with that. V3? CRealtity machine?
2. If you still need to and you don't want to wait, you can then use my improved version to get around the problem.
3. And you can wait and see what the future versions (bugfix-2.0.x) will bring in terms of improvement.

Keichide, 2019-05-05:
I don't get that many bed probes, often just 1 on the whole 4x4 Grid which is enough for the printer to raise Z to high. I also think that the Bugfix should be put into the 1.1.9 Branch, since the Ender 3 is so damn cheap that many people are buying it. And since the Marlin 2.0 is really more for 32bit Boards and was'nt running that smooth on my Ender 3 i downgraded him to 1.1.9 Bugfix. And i think most people will stick with the 1.1.9 until the buy another board, which of course would probablay be a MKS Gen L which also benefits more from the 1.1.9 Branch. 

Anyway i think the simpelst soultion would be to check if there was an ALARM from the BLTouch or compare all of the values if multiple probing points is on and check if there is a huge gap. 

Also i saw some log where it got debug output from the bltoch in it, how do i enable it?

FanDjango, 2019-05-05:
>And since the Marlin 2.0 is really more for 32bit Boards and was'nt running that smooth on my Ender 3

That would interest the dev-guys, because the first half of that statement is not true and they would be interested to know what the problem was. 8 bit AND 32 bit support. 2.0 is running fine on one of my AVR 2560 8 bit boards. I don't think 1.1.9 will get new stuff anymore. Perhaps you should make that a separate issue and present the facts of why you prefer 1.1.9 to the team?

> Also i saw some log where it got debug output from the bltoch in it, how do i enable it?

Depends on what you saw. One thing is enabling "DEBUG_LEVELLING" for example.

Keichide, 2019-05-05:
> 2.0 is running fine on one of my AVR 2560 8 bit boards. 

Yeah.. AVR 2560... The 1284 that is build into the Creality Printers is slower and has less Memory. Which is why i use 96% of Memory with 1.1.9 and 98% with 2.0 and still have to disable stuff that i can enable in 1.1.9. Also it feels like the whole menu reacts much slower compared to 1.1.9 so i think it has much more load on the CPU. But since its still an Beta it's probaly just not fine tuned yet. 

Anyway it`s the same problem in 1.1.9 and 2.0 so i think we are still good with the bug report.

FanDjango, 2019-05-05:
Re. the 1284 I agree especially the memory.

There are persons who would advocate putting advanced features on advanced hardware, and there are those for whom it is sport to tickle that last bit of performance out of outdated equipment. 

The current handling of the BLTouch is "sparse", every error message and debug output, every piece of error recovery and some more coding to get around quirks of the probe cost additional code memory. To counter this, something else must be disabled. On a 1284.

So to a certain degree your wish to have "more sophistication" in the handling of bed levelling and probing is in direct contradiction to your choice of a printer with a minimum price of ???$ and inferior hardware.

I remember my first printer with a 1284p and I disabled SD card support to have some other stuff I needed space for. So I can understand your situation.

But you are actually victim of a second problem: The V3 BLTouch, combined with a sub-par pin configuration on CR machines (the ominous C7 capacitor), which makes your probe "unreliable". It (the V3) worked absolutely perfectly on my old Anet A8 1284p.

I am assuming you have an Ender 3. Please look at that following video (courtesy of CRCinAU): [G28 G29 Ender 3](https://www.youtube.com/watch?v=jjG0n0-D2Lg). Proves with the right firmware, the probe will perform, therefore no real need for code-byte eating addtional program. Of course, on a 2560 or a 32 bit controller, there is no discussion, it should be done.

But for the little ones (1284 etc.) there is a limit for what can be done. So I repeat: Solve the probe reliability problem. a.) Connection b.) Low noise c.) Suitable firmware

Keichide, 2019-05-06:
So you are telling me right now that it's actually my fault that the Probe is unreliabel and not the code? Okay... Thats just a low excuse, for a problem that still persits in the Firmware. 

No matter if it is a noise or connection problem, Marlin still uses that one probe point in his maths which is **wrong**. And can also happen if i solve all the other problems or even build in a 2560 board or 32bit board. 

The point is i only saw that behaviour because i hooked the printer up to a PC to see the output and later put it onto a Raspi for Octopi. If i did'nt had done that, i still would'nt even know that it's not working like i except it, which is the whole point of an ABL to level the little bit left on an unleveled bed.

dot-bob, 2019-05-06:
> I totally agree with this. I will think about adding that to my PR #13841.

@FanDjango It may be best for a separate pull request.  It is best to keep the pull requests simple and limited in scope.  If not it may not get merged any time soon.  Large pull requests take a lot of work and time to fully vet.

As far as addressing topic of this issue a simple work around would be to sample three or more points,  throw out the high and low and average the rest.

FanDjango, 2019-05-06:
@dot-bob Yes, I understand.
If the BLTOUCH stuff is not recoded to sensibly report probe failures, which it currently does not, the only way is to pass outlier analysis on the collected points, like the TO is suggesting and replace any ones exceeding a specific deviation limit by a value deemed "a satisfactory replacement". See [this, for example](https://www.rapidinsight.com/handle-outliers/). So let's drop the outliers, it would seem to be the easiest? Looking at the bilinear code, it would be easier to replace any outlier by the average of all values that are not outliers... kludge code to correct the failings of other kludgecode. On a controller that is bursting at the seams and the developers are trying to save single bytes of code, of ram. 

Maybe if the BLTouch code would please fail correctly, it is better to just abort the G29 so the user knows he can't use the data. And if that happens too often, address the real reason why the probe is failing.

@Keichide I am certain that "So you are telling me right now that it's actually my fault that the Probe is unreliabel and not the code? " is a misunderstanding. **The code is at fault for just using the incorrect probe point.** What to do in case of a probe fail is a big point for discussion, see above. Some things seem so easy and then when you get down to the nitty gritty, it gets ugly.

InsanityAutomation, 2019-05-06:
A sane comparison maximum deviation has been discussed once or twice before on some of this, between g28, g29, and m48. It will probably get done. Not first on my list. Making the probe as reliable as possible in and of itself is the first item. Filtering for when problems still exist will come back around at some point.

InsanityAutomation, 2019-05-06:
> Yeah.. AVR 2560... The 1284 that is build into the Creality Printers is slower and has less Memory. Which is why i use 96% of Memory with 1.1.9 and 98% with 2.0 and still have to disable stuff that [...]

My builds are kept at 94% for the atmega1284p and seem to run fine. Youre welcome to try one for comparison to how yours is doing on Marlin 2.0, I run it on 20 machines atm with a couple of them 1284p.

https://github.com/InsanityAutomation/Marlin/tree/Creality_2.0_Devel

boelle, 2019-07-21:
I just used bugfix-1.1.x and the problem disappeared

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle