MarlinFirmware/Marlin #13913: [BUG] Marlin won't compile with LCD disabled.

Strat00s, 2019-05-05:
### Description

Marlin won't compile, because LCD is disabled

### Steps to Reproduce

1. Use /example_configurations/delta/FLSUN/auto_calibrate configuration.
2. Disable RepRapDiscount Smart Controller.
3. Try to compile it.

**Expected behavior:**
It would compile no problems and I can upload it to the Arduino.

**Actual behavior:**

C:\Users\xxxxx\AppData\Local\Temp\cc6r8ITe.ltrans2.ltrans.o: In function `gcode_G33()':

<artificial>:(.text+0x161a): undefined reference to `lcd_probe_pt(float const&, float const&)'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

#### Additional Information

I guess that https://github.com/MarlinFirmware/Marlin/issues/11513 is exactly the same problem, but I have no idea how to get rid of it.

Ludy87, 2019-05-06:
I can not confirm the error - tested with 1.1.x and bugfix 1.1.x.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle