MarlinFirmware/Marlin #13919: [FR][bugfix-2.0.x]MKS Gen 1.3 Mesh Bed Leveling disabled on startup

IgorMilavec, 2019-05-05:
Using c791d2a I have enabled MESH_BED_LEVELING and RESTORE_LEVELING_AFTER_G28, however the controller shows bed leveling as disabled on startup:
```
Mesh Bed Leveling OFF
3x3 mesh. Z offset: 0.00000
Measured points:
        0        1        2
 0 +0.00000 -0.07500 -0.00000
 1 -0.00000 -0.10000 -0.00000
 2 -0.00000 -0.12500 -0.00000
```
Is it possible to enable bed leveling on startup to make it "transparent" and not require the user to change slicer settings? I have tried "M420 S1" followed by "M500", which did not help.

If this is currently not possible, I propose to add ENABLE_LEVELING_ON_STARTUP to have this feature permanently enabled.

reloxx13, 2019-05-06:
its first restored after g28, cuz g28 z0 is the reference. 

if u dont home after startup, u dont have reference 0 to apply the mesh to.

dont save (m500) after m420 s1. m420 loads the saved mesh.

IgorMilavec, 2019-05-06:
Even if I do a G28 after startup, the Mesh Bed Leveling is still Off.

RESTORE_LEVELING_AFTER_G28 only enables leveling after G28 if it was enabled before, it does not enable it:
```
/**
 * Normally G28 leaves leveling disabled on completion. Enable
 * this option to have G28 restore the prior leveling state.
 */
```

boelle, 2019-07-21:
so this is more a bug than its a feature request?

reloxx13, 2020-07-02:
Closing is not fixing