MarlinFirmware/Marlin #13920: [BUG] PROBING_FANS_OFF compile error

Alissonverd, 2019-05-05:
### Description

Compiler produce an error when PROBING_FANS_OFF is enabled.

### Steps to Reproduce

1. Download the last code from github (2.0x)
2. Compile with and without PROBING_FANS_OFF.

#### Additional Information

> c:\[...]\sketch\src\core\macros.h:188:36: note: in definition of macro 'ZERO'
>  #define ZERO(a)             memset(a,0,sizeof(a))
> 
> c:\[...]\src\gcode\control\M80_M81.cpp:100:27: error: 'class Temperature' has no member named 'paused_fan_speed'
>        ZERO(thermalManager.paused_fan_speed);
> 
> c:\[...]\src\core\macros.h:188:47: note: in definition of macro 'ZERO'
>  #define ZERO(a)             memset(a,0,sizeof(a))

No need modification on configuration files, only enable PROBING_FANS_OFF.

An older version (date 2018.12.30) dont appear this issue.

Arduino IDE 1.9
Windows 10 Pro

Vertabreak, 2019-05-05:
In file included from /tmp/arduino_build_373670/sketch/src/gcode/control/../../inc/MarlinConfigPre.h:32:0,
                 from /tmp/arduino_build_373670/sketch/src/gcode/control/../../inc/MarlinConfig.h:28,
                 from /tmp/arduino_build_373670/sketch/src/gcode/control/../gcode.h:267,
                 from /tmp/arduino_build_373670/sketch/src/gcode/control/M80_M81.cpp:23:
/tmp/arduino_build_373670/sketch/src/gcode/control/M80_M81.cpp: In static member function 'static void GcodeSuite::M81()':
/tmp/arduino_build_373670/sketch/src/gcode/control/M80_M81.cpp:100:27: error: 'class Temperature' has no member named 'paused_fan_speed'
       ZERO(thermalManager.paused_fan_speed);
                           ^
/tmp/arduino_build_373670/sketch/src/gcode/control/../../inc/../core/macros.h:188:36: note: in definition of macro 'ZERO'
 #define ZERO(a)             memset(a,0,sizeof(a))
                                    ^
/tmp/arduino_build_373670/sketch/src/gcode/control/M80_M81.cpp:100:27: error: 'class Temperature' has no member named 'paused_fan_speed'
       ZERO(thermalManager.paused_fan_speed);
                           ^
/tmp/arduino_build_373670/sketch/src/gcode/control/../../inc/../core/macros.h:188:47: note: in definition of macro 'ZERO'
 #define ZERO(a)             memset(a,0,sizeof(a))
                                               ^

Vertabreak, 2019-05-05:
this compile error started after https://github.com/MarlinFirmware/Marlin/commit/3fcab8dbf1f0746a6d3825a9ec099fedbd958943 commit.

Vertabreak, 2019-05-05:
#define ADVANCED_PAUSE_FANS_PAUSE being enabled by it self also produces this error.

/tmp/arduino_build_75750/sketch/src/module/temperature.cpp: In member function 'void Temperature::set_fans_paused(bool)':
/tmp/arduino_build_75750/sketch/src/module/temperature.cpp:185:16: error: 'fans_paused' was not declared in this scope
       if (p != fans_paused) {
                ^

Ludy87, 2019-05-06:
> #define ADVANCED_PAUSE_FANS_PAUSE being enabled by it self also produces this error.
>
> /tmp/arduino_build_75750/sketch/src/module/temperature.cpp: In member function 'void [...]

@thinkyhead 

does it make sense to change that?

https://github.com/MarlinFirmware/Marlin/blob/9853331b34ef6fd214a8f1fbc315e722c90b7341/Marlin/src/module/temperature.h#L382

to 

```c++
#if EITHER(PROBING_FANS_OFF, ADVANCED_PAUSE_FANS_PAUSE)
```

https://github.com/MarlinFirmware/Marlin/blob/9853331b34ef6fd214a8f1fbc315e722c90b7341/Marlin/src/module/temperature.cpp#L141

to

```c++
#if EITHER(PROBING_FANS_OFF, ADVANCED_PAUSE_FANS_PAUSE)
```

Ludy87, 2019-05-06:
https://github.com/MarlinFirmware/Marlin/blob/9853331b34ef6fd214a8f1fbc315e722c90b7341/Marlin/src/gcode/control/M80_M81.cpp#L100

to
```c++
      ZERO(thermalManager.saved_fan_speed);
```
---

https://github.com/MarlinFirmware/Marlin/blob/9853331b34ef6fd214a8f1fbc315e722c90b7341/Marlin/src/module/temperature.cpp#L141-L143

to

```c++
  #if EITHER(PROBING_FANS_OFF, ADVANCED_PAUSE_FANS_PAUSE)
    bool Temperature::fans_paused; // = false;
    uint8_t Temperature::saved_fan_speed[FAN_COUNT]; // = { 0 }
  #endif
```
----

delete line
https://github.com/MarlinFirmware/Marlin/blob/9853331b34ef6fd214a8f1fbc315e722c90b7341/Marlin/src/module/temperature.cpp#L184

---

https://github.com/MarlinFirmware/Marlin/blob/9853331b34ef6fd214a8f1fbc315e722c90b7341/Marlin/src/module/temperature.h#L382-L384

to

```c++
      #if EITHER(PROBING_FANS_OFF, ADVANCED_PAUSE_FANS_PAUSE)
        static bool fans_paused;
        static uint8_t saved_fan_speed[FAN_COUNT];
      #endif
```

closed 2019-05-07 by thinkyhead