MarlinFirmware/Marlin #13922: [FR] (Gcode for bypass Z home)

devildant, 2019-05-06:
### Description

Hello,

I am coding a plugin for octoprint to handle print recovery after a power failure (with UPS).
Basically, I will rely on the reading position in the file, then edit the source gcode to delete the instructions already read (except temperature, fan speed, etc.).
I made some tests of modification of the file gcode on hand, it works but, one thing bothers me.
before the recovery, I created a house X and Y so that the printer can position the hotend.
but obviously I can not do that on the Z.
on Z it's a theoretical position (the Z alternates between value and?)
except that in this configuration, the park nozzle no longer works.
I should be able to get the exact Z position before cutting, so I would like to be able to create or use an existing gcode (but I did not find it) to pretend that the printer had made a home Z.
basically i would like that the printer considers that it has made a home Z, even if it is not the case.
I had hope with the command G92 Zpos, but the absence of Z home is problematic.

I will need your help, if there is no existing solution to do what want, to know in which .cpp (Marlin_main.cpp?) it would be better to make my change, and especially or look to see how to stabilize the Z without doing a Z home.
I also thought to modify the G92 order processing directly, what do you think? (with a specific #define of course)

### Steps to Reproduce

#### Additional Information

marlin 1.1.9

xC0000005, 2019-05-06:
If you look at https://github.com/xC0000005/Marlin/tree/ZBasedRecoveryWork (something I put together to let me restart prints before Creality's PLR was open source), it's a similar idea, and was working fine on my ANET A8 and MP Mini V1. It's clunky, and I should have had it accept a z height parameter for the start printing, but it does something similar - passing each operation through a function which decides to execute it, drop it and continue recovery, or drop it and start printing. You'd want to modify homing instructions to not home Z. There's also code int hat branch to home and get the resulting original Z position, if your build plate is large enough. Of course, on a delta, it would work because it would home upward.

devildant, 2019-05-06:
Hello @xC0000005 thx for your reply, I looked at your commit, I had trouble seeing the operation, but it seems intrusive :)
on my side, I would just make the printer believe that she did are "home Z" with the G92 command

devildant, 2019-05-06:
with a little luck it's just a variable to set to true, but I do not find

FanDjango, 2019-05-07:
Follow the variable NO_MOTION_BEFORE_HOMING maybe where it is used?

devildant, 2019-05-07:
@FanDjango good idea, I had not thought about it,
I'm going to look, well it's been over 8 years since I did not touch c ++, it's not win :p

devildant, 2019-05-07:
i found this method : set_axis_is_at_home :tada: 

it's promising, now I have to experiment

devildant, 2019-05-07:
@FanDjango, you unlocked my brain :p, I searched for the implementation of the command G28 :) after the NO_MOTION